libusb/libusb #1104: WinUSB SuperSpeed Isochronous Transfer issue

mcuee, 2022-03-30:
Continuation of #759 and #749, specific to SuperSpeed USB isoc transfer.
* #759 
* #749

mcuee, 2022-03-30:
From https://github.com/libusb/libusb/pull/749#issuecomment-1082565952

I am encounter issues with SuperSpeed transfer with WinUSB, libusbk seems to perform better. I am not so sure if this is address in this pull request or not.
https://github.com/pyusb/pyusb/issues/380#issuecomment-1082556530

FW is Cypress FX3 SDK Isoc Loopback.

Code:
```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x00f0)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

dev.set_interface_altsetting(interface = 0, alternate_setting = 1)
intf = cfg[(0,1)]

print(intf)

ep = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x03)
        
ep2 = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x83)

assert ep is not None
assert ep2 is not None

#print ('ep:',ep)
#print ('ep2:',ep2)

chunkr = 2**14
chunkw = 2**14

print ("Starting CYUSB FX3 ISOC Loopbacktest")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(0x83, chunkr, 1000)
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Transfered %d kB/s" % (bps/(1024)) )
            trans = 0
[... 59 lines cut ...]
2022-03-30 10:38:43,316 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0, 0, 0)
2022-03-30 10:38:43,316 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0)
2022-03-30 10:38:43,316 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0, 1, 0)
2022-03-30 10:38:43,316 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0, 0, 1, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0, 1, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 1, 0, 1, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0, 1, 0)
2022-03-30 10:38:43,317 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000298D2615DF0>, 0)
2022-03-30 10:38:43,319 DEBUG:usb.backend.libusb1:_LibUSB.iso_read(<usb.backend.libusb1._DeviceHandle object at 0x00000298D2557D00>, 131, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 [... cut ...]  65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
...

```

mcuee, 2022-03-30:
Debug log for libusbk:
```
2022-03-30 10:41:22,912 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7fff891a0000 at 0x1a5472c9f70>)
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 10:41:22,921 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 10:41:22,922 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 10:41:22,960 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445B80>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445D30>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445BE0>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445D90>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445C70>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001A547445DF0>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,964 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x000001A547387D00>, 0)
2022-03-30 10:41:22,964 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x000001A547387D00>, 0, 1)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 1, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 1, 0, 1, 0)
2022-03-30 10:41:22,968 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
[... cut ...]  65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)
...
```

mcuee, 2022-03-30:
Using the Cypress isoc sourcesink FW and the issue seems to be similar. libusbk seems to be okay, WinUSB seems to have issues.

```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x00f1)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

dev.set_interface_altsetting(interface = 0, alternate_setting = 1)
intf = cfg[(0,1)]

print(intf)

ep = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x03)
        
ep2 = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x83)

assert ep is not None
assert ep2 is not None

#print ('ep:',ep)
#print ('ep2:',ep2)

chunkr = 24576*8
chunkw = 24576*8

print ("Starting CYUSB FX3 ISOC Loopbacktest")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(0x83, chunkr, 1000)
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Transfered %d kB/s" % (bps/(1024)) )
            trans = 0
            start = time.time()

def write_loop():
    dat = "A"*chunkw
    while True:
        try:
            ep.write(dat)
        except:
            pass

_thread.start_new_thread( write_loop, () )

read_loop()

```
[winusb_isoc_srcsink_pyusb_debug.zip](https://github.com/libusb/libusb/files/8377233/winusb_isoc_srcsink_pyusb_debug.zip)
[libusbk_isoc_srcsink_pyusb_debug.zip](https://github.com/libusb/libusb/files/8377235/libusbk_isoc_srcsink_pyusb_debug.zip)

mcuee, 2022-03-30:
Full debug log with LIBUSB_DEBUG=4
[libusbk_isoc_srcsink_pyusb_full_debug.zip](https://github.com/libusb/libusb/files/8377263/libusbk_isoc_srcsink_pyusb_full_debug.zip)
[winusb_isoc_srcsink_pyusb_full_debug.zip](https://github.com/libusb/libusb/files/8377272/winusb_isoc_srcsink_pyusb_full_debug.zip)

mcuee, 2022-03-30:
The Cypress FX3 isoc sourcesink FW Device USB Descriptor as from the Cypress Control Center app.
```
<DEVICE>
	FriendlyName="Cypress FX3 USB StreamerExample Device"
	Manufacturer="Cypress"
	Product="FX3"
	SerialNumber=""
	Configurations="1"
	MaxPacketSize="512"
	VendorID="04 B4"
	ProductID="00 F1"
	Class="00h"
	SubClass="00h"
	Protocol="00h"
	BcdDevice="00 00"
	BcdUSB="03 10"
	<BOS>
		NumberOfDeviceCapability="02h"
		DescriptorType="15"
		DescriptorLength="5"
		TotalLength="22"
		<USB20 Device Extension>
			DescriptorLength="7"
			DescriptorType="16"
			DeviceCapabilityType="2"
			bmAttribute="641Eh"
		</USB20 Device Extension>
		<SUPERSPEED USB>
			DescriptorLength="10"
			DescriptorType="16"
			DeviceCapabilityType="3"
			FunctionalitySupporte="3"
			bmAttribute="00h"
			U1Device Exit Latency="0"
			U2Device Exit Latency="00h"
		</SUPERSPEED USB>
	</BOS>
	<CONFIGURATION>
		Configuration="0"
		ConfigurationValue="1"
		Attributes="80h"
		Interfaces="1"
		DescriptorType="2"
		DescriptorLength="9"
		TotalLength="53"
		MaxPower="50"
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="0"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="0"
			DescriptorType="4"
			DescriptorLength="9"
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="1"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="2"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="ISOC"
				Direction="OUT"
				Address="03h"
				Attributes="01h"
				MaxPktSize="24576"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			<SUPER SPEED ENDPOINT COMPANION>
				Type="SUPERSPEED_USB_ENDPOINT_COMPANION"
				MaxBurst="7"
				Attributes="02h"
				BytesPerInterval="6000h"
			</ENDPOINT>
			<ENDPOINT>
				Type="ISOC"
				Direction="IN"
				Address="83h"
				Attributes="01h"
				MaxPktSize="24576"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			<SUPER SPEED ENDPOINT COMPANION>
				Type="SUPERSPEED_USB_ENDPOINT_COMPANION"
				MaxBurst="7"
				Attributes="02h"
				BytesPerInterval="6000h"
			</ENDPOINT>
		</INTERFACE>
	</CONFIGURATION>
</DEVICE>
```

tormodvolden, 2022-03-30:
Can you try applying the patch in #749 (the part that is not already included in git master)? It is meant to increase stability and thus throughput if I understand correctly.

mcuee, 2022-03-30:
I have to patch libusb manually. Hopefully it is correct because when I tried to apply the following patch with "git am .\749.patch" and it failed.
https://patch-diff.githubusercontent.com/raw/libusb/libusb/pull/749.patch

```
diff --git a/libusb/os/windows_winusb.c b/libusb/os/windows_winusb.c
index 6c0db2a..dde1335 100644
--- a/libusb/os/windows_winusb.c
+++ b/libusb/os/windows_winusb.c
@@ -2873,8 +2873,41 @@ static void WINAPI winusbx_native_iso_transfer_continue_stream_callback(struct l
 	} else {
 		// If the transfer wasn't successful we reschedule the transfer while forcing it
 		// not to continue the stream. This might results in a 5-ms delay.
-		transfer_priv->iso_break_stream = TRUE;
-		libusb_submit_transfer(transfer);
+		libusb_transfer_cb_fn postSubmissionHandler = NULL;
+		bool resubmitTransfer= TRUE;
+
+		if (transfer->callback) {
+			// Notify the user callback that a resubmission is about to take place
+			// This gives it an option to interfer or schedule submission of other transfers after this one
+			transfer->status = LIBUSB_TRANSFER_STALL;
+			transfer->callback(transfer);
+			if (transfer->callback != transfer_priv->iso_user_callback) {
+				// Callback temporarily used as signal by user callback
+				if (transfer->callback == NULL) {
+					// Signal to not resubmit with ContinueStream=False
+					resubmitTransfer = FALSE;
+				} else {
+					// Otherwise handler to call after resubmission
+					postSubmissionHandler = transfer->callback;
+				}
+				// Restore the user callback
+				transfer->callback = transfer_priv->iso_user_callback;
+			}
+		}
+
+		if (resubmitTransfer) {
+			usbi_info(TRANSFER_CTX(transfer), "Resubmitting with ContinueStream=False!");
+			transfer_priv->iso_break_stream = TRUE;
+			libusb_submit_transfer(transfer);
+		} else {
+			usbi_info(TRANSFER_CTX(transfer), "User callback supressed resubmission with ContinueStream=False!");
+		}
+
+		if (postSubmissionHandler) {
+			// User-supplied handler to call AFTER resubmission
+			// Probably submitting other transfers to follow up on the last one to continue the stream
+			postSubmissionHandler(transfer);
+		}
 	}
 }
 static int winusbx_submit_iso_transfer(int sub_api, struct usbi_transfer *itransfer)
@@ -3038,6 +3071,13 @@ static int winusbx_submit_iso_transfer(int sub_api, struct usbi_transfer *itrans
 		// - Transfers are first scheduled with ContinueStream = TRUE and with winusbx_iso_transfer_continue_stream_callback as user callback.
 		// - If the transfer succeeds, winusbx_iso_transfer_continue_stream_callback restore the user callback and calls its.
 		// - If the transfer fails, winusbx_iso_transfer_continue_stream_callback reschedule the transfer and force ContinueStream = FALSE.
+		// However, since this behaviour interferes with the standard practice to submit multiple transfers at once (see #747), additionally
+		// the user callback is called with the STALL event, allowing it signal additional information. The user callback can signal to
+		// suppress this behaviour (by setting transfer->callback to NULL) or alternatively set transfer->callback to a custom callback
+		// which get's executed AFTER the resubmission with ContinueStream = False, allowing the user callback to submit further transfers
+		// to continue the stream only after one transfer has established a stream. This does NOT require extra work in the user callback,
+		// if it properly ignores stall events for isochronous transfers, however it adds more control. libusb can and should not operate
+		// on just speculation on such a matter. Implementation example of a working isochronous stream in #747
 		if (!transfer_priv->iso_break_stream) {
 			transfer_priv->iso_user_callback = transfer->callback;
 			transfer->callback = winusbx_native_iso_transfer_continue_stream_callback;
diff --git a/libusb/version_nano.h b/libusb/version_nano.h
index 46b5e01..243a660 100644
--- a/libusb/version_nano.h
+++ b/libusb/version_nano.h
@@ -1 +1 @@
-#define LIBUSB_NANO 11717
+#define LIBUSB_NANO 999999

```

Now it crashed with both drivers.
[winusb_749.zip](https://github.com/libusb/libusb/files/8378988/winusb_749.zip)
[libusbk_749.zip](https://github.com/libusb/libusb/files/8378989/libusbk_749.zip)
.

mcuee, 2022-03-30:
libusbk seems to be okay whereas WinUSB will hang.
[winusb_debug_749.zip](https://github.com/libusb/libusb/files/8379165/winusb_debug_749.zip)
[libusbk_debug_749.zip](https://github.com/libusb/libusb/files/8379166/libusbk_debug_749.zip)

mcuee, 2022-03-30:
But it may be better to use C based codes to test. So I think we do not need to worry too much about this for now. We can always wait after1.0.26 if there is no quick fix.

tormodvolden, 2022-03-30:
Yes, since we know there are issues with pyusb isochronous we should try testing it with C if possible. I have no device/software for this either. Your findings here is good input for #749 but this is not critical for 1.0.26. I think overall isoch on Windows is in better shape than a long time, or ever.

mcuee, 2022-03-30:
> Yes, since we know there are issues with pyusb isochronous we should try testing it with C if possible. I have no device/software for this either. Your findings here is good input for #749 but this [...]

Totally agree that now it is in a much better shape than before.

mcuee, 2022-03-31:
Isochronous transfer is hard.
The relevant Microsoft documentation is here.
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-set-up-to-use-windows-devices-usb

Another potential reference is libusbK WinUSB code.
https://github.com/mcuee/libusbk/blob/master/libusbK/src/lusbk_bknd_winusb.c

mcuee, 2022-03-31:
I think I can reproduce the issue with C based application, a quick hack of the libusb example.
https://github.com/pyusb/pyusb/issues/380#issuecomment-1084547337

```
/*
 * libusb example program to measure Atmel SAM3U isochronous performance
 * Copyright (C) 2012 Harald Welte <laforge@gnumonks.org>
 *
 * Copied with the author's permission under LGPL-2.1 from
 * http://git.gnumonks.org/cgi-bin/gitweb.cgi?p=sam3u-tests.git;a=blob;f=usb-benchmark-project/host/benchmark.c;h=74959f7ee88f1597286cd435f312a8ff52c56b7e
 *
 * An Atmel SAM3U test firmware is also available in the above repository.
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
 */

#include <config.h>

#include <errno.h>
#include <signal.h>
#include <stdio.h>
#include <stdlib.h>
#ifdef HAVE_SYS_TIME_H
#include <sys/time.h>
#endif
#include <time.h>

#include "libusb.h"

#define EP_ISO_IN	0x82

static volatile sig_atomic_t do_exit = 0;
static struct libusb_device_handle *devh = NULL;

static unsigned long num_bytes = 0, num_xfer = 0;
static struct timeval tv_start;

static void get_timestamp(struct timeval *tv)
{
#if defined(PLATFORM_WINDOWS)
	static LARGE_INTEGER frequency;
	LARGE_INTEGER counter;

	if (!frequency.QuadPart)
		QueryPerformanceFrequency(&frequency);

	QueryPerformanceCounter(&counter);
	counter.QuadPart *= 1000000;
	counter.QuadPart /= frequency.QuadPart;

	tv->tv_sec = (long)(counter.QuadPart / 1000000ULL);
	tv->tv_usec = (long)(counter.QuadPart % 1000000ULL);
#elif defined(HAVE_CLOCK_GETTIME)
	struct timespec ts;

	(void)clock_gettime(CLOCK_MONOTONIC, &ts);
	tv->tv_sec = ts.tv_sec;
	tv->tv_usec = (int)(ts.tv_nsec / 1000L);
[... 145 lines cut ...]

	benchmark_in(EP_ISO_IN);

	while (!do_exit) {
		rc = libusb_handle_events(NULL);
		if (rc != LIBUSB_SUCCESS)
			break;
	}

	/* Measurement has already been done by the signal handler. */

	libusb_release_interface(devh, 2);
out:
	if (devh)
		libusb_close(devh);
	libusb_exit(NULL);
	return rc;
}

```

The following is with Cystream FW on the Cyusb FX2LP (high speed USB)
Debug Log with LIBUSB_DEBUG=4, WinUSB got one time error 87. libusbk got many missing packets but this may be due to the code.
[libusbk_full_debug.zip](https://github.com/pyusb/pyusb/files/8389850/libusbk_full_debug.zip)
[winusb_full_debug.zip](https://github.com/pyusb/pyusb/files/8389851/winusb_full_debug.zip)

mcuee, 2022-04-01:
Same codes seem to kind of working for the packet but not so sure why transfer size shows as zero. So probably the code is still a bit wrong.

```
pack0 length:1024, actual_length:1024
pack1 length:1024, actual_length:1024
pack2 length:1024, actual_length:1024
pack3 length:1024, actual_length:1024
pack4 length:1024, actual_length:1024
pack5 length:1024, actual_length:1024
pack6 length:1024, actual_length:1024
pack7 length:1024, actual_length:1024
pack8 length:1024, actual_length:1024
pack9 length:1024, actual_length:1024
pack10 length:1024, actual_length:1024
pack11 length:1024, actual_length:1024
pack12 length:1024, actual_length:1024
pack13 length:1024, actual_length:1024
pack14 length:1024, actual_length:1024
pack15 length:1024, actual_length:1024
length:16384, actual_length:0

```

Full debug log under Ubuntu 20.04 Linux.
[linux_libusb_cystream_isoc_in_debug.zip](https://github.com/libusb/libusb/files/8394707/linux_libusb_cystream_isoc_in_debug.zip)

mcuee, 2022-04-01:
Maybe this one is a better C example. It may need some minor fixes for Windows though.
https://github.com/walac/isochronous

Minor mod to test Cystream FW.
```
#include <libusb.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <assert.h>
#include <unistd.h>
#include <libgen.h>

#define LIBUSB_CHECK(expr) libusb_check(#expr, expr, __LINE__, __FILE__)

#define IDVENDOR 0x04b4
#define IDPRODUCT 0x1003
#define EPOUT 0x06
#define EPIN 0x82

static int
libusb_check(const char *expr, int ec, size_t line, const char *filepath)
{
    if (ec != LIBUSB_SUCCESS) {
        const size_t expr_size = strlen(expr) + 1;
        char fn[expr_size];

        memcpy(fn, expr, expr_size * sizeof(char));
        char *p = strchr(fn, '(');

        if (p)
            *p = '\0';

        const size_t filepath_size = strlen(filepath) + 1;
        char file[filepath_size];

        memcpy(file, filepath, filepath_size * sizeof(char));

        fprintf(stderr, "%s(%zu): %s\n", basename(file), line, libusb_error_name(ec));
        exit(EXIT_FAILURE);
    }

    return ec;
}

static void
onexit(void)
{
    libusb_exit(NULL);
}

static void *
create_dummy_data(size_t size)
{
    uint8_t *data = calloc(size, sizeof(uint8_t));
    assert(data);

    uint8_t ch = 0;
    for (size_t i = 0; i < size; ++i)
        data[i] = ch++;

    return data;
}

static int
get_packet_length(libusb_device *dev, unsigned char ep)
{
    assert(dev);

    struct libusb_config_descriptor *config;
[... 196 lines cut ...]

It seems to work under Ubuntu Linux 20.04.
```
mcuee@UbuntuSwift3:~/build/isochronous$ ./iso-test -n 8192
n= 8192 .
packet size is 1024 .
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
packet size is 1024 .
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.

```

mcuee, 2022-04-01:
WinUSB output under Windows 11. OUT endpoint -- totally not working; IN endpoint -- somewhat working.
```
$ ./iso-test.exe -n 8192
packet size is 1024 .
n= 8192 .
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 0 bytes.
packet size is 1024 .
Packet 0 requested to transfer 0 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 2048 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 3072 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 4096 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 5120 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 6144 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 7168 bytes, transfered 1024 bytes.
```

Debug log:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000024] [00001088] libusb: debug [libusb_init] created default context
[ 0.000032] [00001088] libusb: debug [libusb_init] libusb v1.0.25.11692
[ 0.000037] [00001088] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000050] [00001088] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000050] [00001088] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000056] [00001088] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.000057] [00001088] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001455] [00001088] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.002970] [00001088] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.004755] [00001088] libusb: info [windows_init] UsbDk backend is not available
[ 0.004797] [00001088] libusb: debug [libusb_get_device_list]  
[ 0.004929] [0000469c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.010670] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.010702] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.011045] [00001088] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.011067] [00001088] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.011072] [00001088] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.011072] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.011178] [00001088] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.011200] [00001088] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.011203] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.011274] [00001088] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.011290] [00001088] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.011290] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [37]
[ 0.011436] [00001088] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.011453] [00001088] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.011454] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.011520] [00001088] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.011535] [00001088] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.011536] [00001088] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.011676] [00001088] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.011697] [00001088] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.012203] [00001088] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.012252] [00001088] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.012260] [00001088] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.012267] [00001088] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.012307] [00001088] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.012311] [00001088] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.012338] [00001088] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.012352] [00001088] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.012356] [00001088] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.012399] [00001088] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[... 145 lines cut ...]
[ 0.028238] [00001088] libusb: debug [libusb_release_interface] interface 0
[... cut ...]
Packet 0 requested to transfer 0 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 2048 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 3072 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 4096 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 5120 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 6144 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 7168 bytes, transfered 1024 bytes.
 destroy device 1.0

```

mcuee, 2022-04-01:
libusbk -- not working for EP IN and EP OUT.
```
$ ./iso-test.exe -n 8192
packet size is 1024 .
n= 8192 .
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 0 bytes.
packet size is 1024 .
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 0 bytes.
```

Debug log.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000015] [00003754] libusb: debug [libusb_init] created default context
[ 0.000023] [00003754] libusb: debug [libusb_init] libusb v1.0.25.11692
[ 0.000046] [00003754] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000094 events 0
[ 0.000051] [00003754] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000052] [00003754] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000098 events 0
[ 0.000061] [00003754] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.000062] [00003754] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001753] [00003754] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.003341] [00003754] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.004925] [00003754] libusb: info [windows_init] UsbDk backend is not available
[ 0.004966] [00003754] libusb: debug [libusb_get_device_list]  
[ 0.005183] [00003a2c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.010758] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.010785] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.011102] [00003754] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.011122] [00003754] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.011127] [00003754] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.011127] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.011198] [00003754] libusb: debug [get_api_type] driver(s): libusbK
[ 0.011213] [00003754] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.011214] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [25]
[ 0.011299] [00003754] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.011319] [00003754] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.011321] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [35]
[ 0.011456] [00003754] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.011470] [00003754] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.011471] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.011530] [00003754] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.011544] [00003754] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.011544] [00003754] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.011677] [00003754] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.011696] [00003754] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.012166] [00003754] libusb: debug [winusb_get_device_list] found existing device for session [35]
[ 0.012209] [00003754] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.012216] [00003754] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.012223] [00003754] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.012260] [00003754] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.012263] [00003754] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.012288] [00003754] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.012305] [00003754] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.012309] [00003754] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.012349] [00003754] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[... 149 lines cut ...]
packet size is 1024 .
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 0 bytes.
[... cut ...]
[ 0.027289] [00003754] libusb: debug [usbi_handle_transfer_completion] transfer 00000156358907b8 has callback 00007ff744201b79
[ 0.027301] [00003754] libusb: debug [libusb_free_transfer] transfer 00000156358907b8
[ 0.027304] [00003754] libusb: debug [libusb_release_interface] interface 0
[ 0.027310] [00003754] libusb: debug [libusb_close]  
[ 0.027323] [00003754] libusb: debug [libusb_unref_device] destroy device 1.8
[ 0.027324] [00003754] libusb: debug [libusb_unref_device] destroy device 1.0

```

mcuee, 2022-04-01:
I also manged to port Cyusb FX3 SDK Linux libcyusb (based on libusb) to Windows. And then the test results using the 09_cyusb_performance example are similar, bulk transfer works flawlessly under Windows. Isoc transfer does not work with either WinUSB (data rate is zero) or libusbk (hangs).

Example run log:
WinUSB:
```
$ ./09_cystream_isoc_performance.exe -e 6 -s 16 -q 16 -d 1
C:\work\cypress\cy_fx3_sdk\cyusb_linux_1.0.5\src\09_cystream_isoc_performance.exe: Found endpoint 0x6 in interface 0, setting 6
C:\work\cypress\cy_fx3_sdk\cyusb_linux_1.0.5\src\09_cystream_isoc_performance.exe: Starting test with the following parameters
        Request size     : 0x10
        Queue depth      : 0x10
        Test duration    : 0x1
        Endpoint to test : 0x6

        Endpoint type    : 0x1
        Max packet size  : 0x200
Transfer Counts: 16 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 32 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 48 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 64 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 80 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 96 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 112 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 128 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 144 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 160 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 176 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 192 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 208 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 224 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 240 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 256 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 272 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 288 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 304 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 320 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 336 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 352 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 368 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 384 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 400 pass 0 fail
Data rate: 0.000000 KBps

Transfer Counts: 416 pass 0 fail
Data rate: 0.000000 KBps

C:\work\cypress\cy_fx3_sdk\cyusb_linux_1.0.5\src\09_cystream_isoc_performance.exe: Test duration is complete. Stopping transfers
16 requests are pending
14 requests are pending
Transfer Counts: 432 pass 0 fail
Data rate: 0.000000 KBps
```

libusbk:
```
$ ./09_cystream_isoc_performance.exe -e 6 -s 16 -q 16 -d 1
C:\work\cypress\cy_fx3_sdk\cyusb_linux_1.0.5\src\09_cystream_isoc_performance.exe: Found endpoint 0x6 in interface 0, setting 6
C:\work\cypress\cy_fx3_sdk\cyusb_linux_1.0.5\src\09_cystream_isoc_performance.exe: Starting test with the following parameters
        Request size     : 0x10
        Queue depth      : 0x10
        Test duration    : 0x1
        Endpoint to test : 0x6

        Endpoint type    : 0x1
        Max packet size  : 0x200
        <hang>
```

mcuee, 2022-04-01:
The libcyusb port is here (including the libcyusb and all the examples source and 64bit binaries compiled under MSYS2 MinGW-64).
[libcyusb_port_to_windows.zip](https://github.com/libusb/libusb/files/8396726/libcyusb_port_to_windows.zip)

tormodvolden, 2022-04-03:
OK, so the "transfered 0 bytes." for OUT packets seen with walac's iso-test is not much of an issue (#1105) and is also seen on Full-Speed.

The "length:16384, actual_length:0" you saw with sam3u_benchmark.c is because the program wrongly uses the /transfer/ actual_size, which it shouldn't do in case of isochronous transfers. It should rather add up the /packet/ actual_size's - which are broken for OUT on WinUSB/libusbK.

I have reworked sam3u_benchmark so much that I think I will rather submit it as a new test-tool, see attached. I can also work these changes into the sam3u_benchmark file but since I cannot test it on the original device, it is more tricky. Maybe you want to try it. I use it for my custom stm32 firmware but simply changing the #define's at the top should do for other devices.
[stm32_isoc.c.txt](https://github.com/libusb/libusb/files/8404707/stm32_isoc.c.txt)

tormodvolden, 2022-04-03:
As mentioned in #1105 I get the error 87 with my FS device, so I think the issue also doesn't belong to this HS/SS ticket. iso-test works fine once after replugging the device, then IN fails (all zeroes received) in the second run. Do you see this pattern also?

mcuee, 2022-04-03:
My #1105 test is using a Full Speed USB device, so yes it also happens with FS. I have changed the title.

As for the replugging -- that means the USB device is in a bad state after the run (eg: Data Toggle error for bulk transfer, but isoc does not have data toggling so probably not the issue here). I think I have not seen this. But I will try all my FS/HS/SS testing again with pull request #1107.

tormodvolden, 2022-04-04:
Actually if I reset the microcontroller (USB lines stay pulled up so no reenumeration) it remains the same, so I wonder if the bad state is in the host controller or driver.

mcuee, 2022-04-04:
> Actually if I reset the microcontroller (USB lines stay pulled up so no reenumeration) it remains the same, so I wonder if the bad state is in the host controller or driver.

My test shows no error 87 if using libusbk. And replug does not help to sort out error 87 when using WinUSB. Please refer to the debug log in #1105 (with #1107 applied).
https://github.com/libusb/libusb/issues/1105#issuecomment-1087053412

* #1105
* #1107

mcuee, 2022-04-04:
With #1107 pull request applied, there is no issue with CyStream FW on the High Speed FX2LP device, if using libusbk. 

For WinUSB, it has some limitation for the transfer size, if not there will be an error message like the following. 
```
libusb: error [winusbx_submit_iso_transfer] length of isoch buffer must be a multiple of the MaximumBytesPerInterval * 8 / Interval
```
And there will be error 87 for both Isoc IN and Isoc OUT.

WinUSB run log.
```
PS C:\work\libusb\isochronous\iso_cyusb> .\iso-test.exe -n 8192
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
packet size is 1024 .
n= 8192 .
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
packet size is 1024 .
libusb: error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
```

libusbk has the similar limitation of transfer size (different error message) but it does not have the error 87.

```
PS C:\work\libusb\isochronous\iso_cyusb> .\iso-test.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
packet size is 1024 .
n= 1024 .
libusb: error [windows_handle_transfer_completion] detected I/O error 1: [1] Incorrect function.
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.
packet size is 1024 .
libusb: error [windows_handle_transfer_completion] detected I/O error 1: [1] Incorrect function.
Packet 0 requested to transfer 1024 bytes, transfered 0 bytes.

PS C:\work\libusb\isochronous\iso_cyusb> .\iso-test.exe -n 8192
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
packet size is 1024 .
n= 8192 .
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
packet size is 1024 .
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.

```

tormodvolden, 2022-04-04:
> With #1107 pull request applied, there is no issue with CyStream FW on the High Speed FX2LP device, if using libusbk.

Do you mean #1107 changed something or just that everything works like before?

> 
> For WinUSB, it has some limitation for the transfer size, if not there will be an error message like the following. And there will be error 87 for both Isoc IN and Isoc OUT.

Same question, is this something new after #1107, or are you just clarifying the current status with all this applied?

tormodvolden, 2022-04-04:
#1109 (libusbK only) will affect applications that actually check the packet status. E.g. sam32u_benchmark or stm32_isoc might terminate earlier, because a packet error is no longer ignored.

mcuee, 2022-04-04:
I will see how I can test #1109. The CyStream FW does not do loop back on Alt Setting 6, it is rather a source/sink combination. But I can probably use Cypress FX3 isoc-loopback example and stm32_isoc.c.

tormodvolden, 2022-04-04:
In any case, I tested libfreenect successfully on both WinUSB and libusbK with #1109 applied. There are no errors in the log, and the application is not complaining about lost packets much.

The status for my own test device:
WinUSB: IN and OUT works, but I only get half of the expected bandwidth (measured by stm32_isoc) and the log shows error 87 (previously also seen error 31) for every second transfer.
libusbK: iso-test OUT works, IN works on first run then always error 87, stm32_isoc works and shows almost full bandwidth IN and OUT.

I had initially problems with stm32_isoc always failing on first transfer with libusbK but that was because I had the transfer size too large (5632) but adjusting this (2046) fixed that. WinUSB seems fine with larger sizes.

tormodvolden, 2022-04-04:
BTW, I have been frustrated by the very time-consuming switching of drivers with Zadig (it takes forever, although not the 5 minutes it says it can take) and I need to switch for both interfaces on my device to not step into the composite driver issue (#965). Now the advantage of a $2 test device, just throw in another one and have each assigned a different driver :)

mcuee, 2022-04-04:
The Cypress FX3 SDK libcyusb example 09 seems to work fine as well for both libusbk and WinUSB for isoc out.

```
C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_bulk_performance.exe: Found endpoint 0x6 in interface 0, setting 2
C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_bulk_performance.exe: Starting test with the following parameters
	Request size     : 0x10
	Queue depth      : 0x10
	Test duration    : 0x1
	Endpoint to test : 0x6

	Endpoint type    : 0x2
	Max packet size  : 0x200
Transfer Counts: 16 pass 0 fail
Data rate: 20685.197156 KBps

Transfer Counts: 32 pass 0 fail
Data rate: 23664.263265 KBps

Transfer Counts: 48 pass 0 fail
Data rate: 26987.138942 KBps

Transfer Counts: 64 pass 0 fail
Data rate: 24965.866979 KBps

Transfer Counts: 80 pass 0 fail
Data rate: 23743.275830 KBps

Transfer Counts: 96 pass 0 fail
Data rate: 24192.024192 KBps

Transfer Counts: 112 pass 0 fail
Data rate: 25728.643216 KBps

Transfer Counts: 128 pass 0 fail
Data rate: 25427.095749 KBps

Transfer Counts: 144 pass 0 fail
Data rate: 24530.471445 KBps

Transfer Counts: 160 pass 0 fail
Data rate: 24811.009886 KBps

Transfer Counts: 176 pass 0 fail
Data rate: 26299.568523 KBps

Transfer Counts: 192 pass 0 fail
Data rate: 24019.515857 KBps

Transfer Counts: 208 pass 0 fail
Data rate: 25559.105431 KBps

Transfer Counts: 224 pass 0 fail
Data rate: 23664.263265 KBps

Transfer Counts: 240 pass 0 fail
Data rate: 23920.762474 KBps

Transfer Counts: 256 pass 0 fail
Data rate: 25858.585859 KBps

Transfer Counts: 272 pass 0 fail
Data rate: 23302.384853 KBps

Transfer Counts: 288 pass 0 fail
Data rate: 24137.280784 KBps

Transfer Counts: 304 pass 0 fail
Data rate: 24893.037729 KBps

Transfer Counts: 320 pass 0 fail
Data rate: 24006.001500 KBps

Transfer Counts: 336 pass 0 fail
Data rate: 24897.879790 KBps

Transfer Counts: 352 pass 0 fail
Data rate: 24219.489120 KBps

Transfer Counts: 368 pass 0 fail
Data rate: 25785.656728 KBps

Transfer Counts: 384 pass 0 fail
Data rate: 24478.867852 KBps

Transfer Counts: 400 pass 0 fail
Data rate: 23332.118119 KBps

Transfer Counts: 416 pass 0 fail
Data rate: 25422.045680 KBps

Transfer Counts: 432 pass 0 fail
Data rate: 24376.309274 KBps

Transfer Counts: 448 pass 0 fail
Data rate: 25754.527163 KBps

Transfer Counts: 464 pass 0 fail
Data rate: 23268.496637 KBps

Transfer Counts: 480 pass 0 fail
Data rate: 24610.651798 KBps

Transfer Counts: 496 pass 0 fail
Data rate: 26143.790850 KBps

Transfer Counts: 512 pass 0 fail
Data rate: 23171.614772 KBps

Transfer Counts: 528 pass 0 fail
Data rate: 23845.007452 KBps

Transfer Counts: 544 pass 0 fail
Data rate: 23836.126629 KBps

Transfer Counts: 560 pass 0 fail
Data rate: 23867.238486 KBps

Transfer Counts: 576 pass 0 fail
Data rate: 19695.337744 KBps

Transfer Counts: 592 pass 0 fail
Data rate: 20668.496690 KBps

Transfer Counts: 608 pass 0 fail
Data rate: 24150.943396 KBps

Transfer Counts: 624 pass 0 fail
Data rate: 26218.762802 KBps

Transfer Counts: 640 pass 0 fail
Data rate: 24907.569566 KBps

Transfer Counts: 656 pass 0 fail
Data rate: 26981.450253 KBps

Transfer Counts: 672 pass 0 fail
Data rate: 23725.671918 KBps

Transfer Counts: 688 pass 0 fail
Data rate: 23607.524899 KBps

Transfer Counts: 704 pass 0 fail
Data rate: 24777.390631 KBps

Transfer Counts: 720 pass 0 fail
Data rate: 26750.261233 KBps

Transfer Counts: 736 pass 0 fail
Data rate: 25432.147824 KBps

Transfer Counts: 752 pass 0 fail
Data rate: 28545.941124 KBps

Transfer Counts: 768 pass 0 fail
Data rate: 27959.807776 KBps

C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_bulk_performance.exe: Test duration is complete. Stopping transfers
16 requests are pending
15 requests are pending
Transfer Counts: 784 pass 0 fail
Data rate: 127.263526 KBps

C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_bulk_performance.exe: Transfers completed
C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_bulk_performance.exe: Test completed

```

mcuee, 2022-04-04:
> #1109 (libusbK only) will affect applications that actually check the packet status. E.g. sam32u_benchmark or stm32_isoc might terminate earlier, because a packet error is no longer ignored.

I will try out sam32u_benchmark or stm32_isoc  for my devices tomorrow.

mcuee, 2022-04-05:
Pull request #1109 seems to work for libusbk for Isoc OUT but not Isoc IN.

Isoc IN is woring for altf=6 (EP82 and EP06, packet size=1024)
```
PS C:\work\libusb\libusb_test\isoc_stm32> .\stm32_isoc.exe -n8192 -v -o
num_iso_pack = 8
request endpoint:06 length:8192, xfr length (bulk):8192, total (isoch):8192
request endpoint:06 length:8192, xfr length (bulk):0, total (isoch):8192
request endpoint:06 length:8192, xfr length (bulk):8192, total (isoch):8192
request endpoint:06 length:8192, xfr length (bulk):0, total (isoch):8192
...
request endpoint:06 length:8192, xfr length (bulk):0, total (isoch):8192
request endpoint:06 length:8192, xfr length (bulk):0, total (isoch):8192
516 transfers (total 4227072 bytes) in 1066 milliseconds => 3965358 bytes/sec
request endpoint:06 length:8192, xfr length (bulk):0, total (isoch):8192

```

Isoc OUT is not working since it shows actual length as zero.

```
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
...
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
55 transfers (total 8192 bytes) in 692 milliseconds => 11838 bytes/sec
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
```

tormodvolden, 2022-04-05:
stm32_isoc.exe -o does OUT (endpoint 06), otherwise IN (endpoint 82). Is the actual_length different without #11109?

mcuee, 2022-04-05:
> stm32_isoc.exe -o does OUT (endpoint 06), otherwise IN (endpoint 82). Is the actual_length different without #1109? The last commit there is just for testing, you may want to remove it just in case.

Sorry I have a typo in the comments. Isoc IN is not working whereas Isoc OUT is working.
Are you saying that https://github.com/tormodvolden/libusb/commit/a447a44c2e32183d8b8fcba1e8fb3fa1729544f0 should be removed in your tree?

tormodvolden, 2022-04-05:
The commit should not be needed in normal cases. It would fix and warn about an issue a saw earlier (maybe while using too large transfer sizes for libusbK, I am not sure) where I saw reported negative packet lengths (printed as e.g.  4294966252). But just in case I changed something inadvertently there, just revert it. The principal change in #1109 (without this) is to report the (libusbK) packet status back to the application. It would like only affect non-successful statuses that previously were taken as successful. Please add a debug statement to print the status before and after the new assignment. Maybe libusbK always reports failure (which was previously ignored) but now the passed-on failed status causes the upper layer to discount the package?

mcuee, 2022-04-05:
> In your IN log above the symptom is that the packet actual_length is 0. The status is still COMPLETED, otherwise stm32_isoc would have bailed out. I am surprised that #1109 can affect the packet actual_length since it only changes the status. Is #1109 the only change from the run in [#1104 (comment)](https://github.com/libusb/libusb/issues/1104#issuecomment-1087062875) ?

I belive so. But let me redo the testing with or without #1109 for Isoc IN transfer of the CyStream FW (04b4:1003) on Cypress FX2LP (High Speed USB).

mcuee, 2022-04-05:
Looks like the results with or without #1109 is the same if using stm32_isoc test program. Both will show actual_length=0.
[libusbk_no_p1109_debug_verbose.zip](https://github.com/libusb/libusb/files/8422439/libusbk_no_p1109_debug_verbose.zip)
[libusbk_p1109_debug_verbose.zip](https://github.com/libusb/libusb/files/8422440/libusbk_p1109_debug_verbose.zip)

mcuee, 2022-04-05:
WinUSB seems to behave the same with or without #1109 (#1109 does not affect WinUSB). The main problem is still Error 87.
[winusb_no_p1109_debug_verbose.zip](https://github.com/libusb/libusb/files/8422459/winusb_no_p1109_debug_verbose.zip)
[winusb_p1109_debug_verbose.zip](https://github.com/libusb/libusb/files/8422460/winusb_p1109_debug_verbose.zip)

mcuee, 2022-04-06:
@tormodvolden Somehow I can not get your version of isochronous to work.
PS C:\work\libusb\libusb_test\isochronous\isochronous_new> .\iso-test.exe -d 04b4:1003 -i 0 -a 6 -e 82 -n 8192 -v
Assertion failed: packet_size, file iso.c, line 87

The original version is working with hard-coded configurations.
```
PS C:\work\libusb\libusb_test\isochronous\isochronous_me> .\iso-test.exe -n8192
Successfully found pyusb benchmark firmware device.
Successfully claimed Interface 0.
Successfully claimed Interface 0 alt setting 6.
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 0 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 1 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 2 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 3 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 4 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 5 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 6 requested to transfer 1024 bytes, transfered 1024 bytes.
Packet 7 requested to transfer 1024 bytes, transfered 1024 bytes.
```

Code:
```
#include <libusb.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <assert.h>
#include <unistd.h>
#include <libgen.h>

#define LIBUSB_CHECK(expr) libusb_check(#expr, expr, __LINE__, __FILE__)

#define IDVENDOR 0x04b4
#define IDPRODUCT 0x1003
#define EPOUT 0x06
#define EPIN 0x82

static int
libusb_check(const char *expr, int ec, size_t line, const char *filepath)
{
    if (ec != LIBUSB_SUCCESS) {
        const size_t expr_size = strlen(expr) + 1;
        char fn[expr_size];

        memcpy(fn, expr, expr_size * sizeof(char));
        char *p = strchr(fn, '(');

        if (p)
            *p = '\0';

        const size_t filepath_size = strlen(filepath) + 1;
        char file[filepath_size];

        memcpy(file, filepath, filepath_size * sizeof(char));

        fprintf(stderr, "%s(%zu): %s\n", basename(file), line, libusb_error_name(ec));
        exit(EXIT_FAILURE);
    }

    return ec;
}

[... 193 lines cut ...]
    int opt;

    while ((opt = getopt(argc, argv, optstr)) != -1)
        switch (opt)
        {
            case 'n':
                n = atoi(optarg);
                break;

            default:
                exit(EXIT_FAILURE);
        }

    LIBUSB_CHECK(libusb_init(NULL));
//    atexit(onexit);

    libusb_device_handle *dev = setup_bm_device();
    assert(dev);

    write_data(dev, n);
    read_data(dev, n);
    close_device(dev);
}

```

mcuee, 2022-04-06:
If I hard code your version, it also works. Maybe my command line option is not correct.
```
diff --git a/iso.c b/iso.c
index 547acaa..ee4f9b3 100644
--- a/iso.c
+++ b/iso.c
@@ -8,12 +8,12 @@

 #define LIBUSB_CHECK(expr) libusb_check(#expr, expr, __LINE__, __FILE__)

-#define IDVENDOR 0x04d8
-#define IDPRODUCT 0xfa2e
+#define IDVENDOR 0x04b4
+#define IDPRODUCT 0x1003
 #define IFACE 0
-#define ALTS 2
-#define EPOUT 0x03
-#define EPIN 0x83
+#define ALTS 6
+#define EPOUT 0x06
+#define EPIN 0x82

 static int verbose;

PS C:\work\libusb\libusb_test\isochronous\isochronous_new> .\iso-test.exe -n 8192
Packet 0 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 0 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.

```

mcuee, 2022-04-06:
@tormodvolden Not so sure why your version of isochronous always give packet_size =0. I added a few debug print but I could not figure out the problem somehow.

```
S C:\work\libusb\libusb_test\isochronous\isochronous_new> C:\libusbK-dev-kit\kList.exe

Loading USB ID's maintained by Stephen J. Gowdy <linux.usb.ids@gmail.com>..

 1. Benchmark Device (Microchip Technology, Inc.) [Connected]
    Service              : libusbK
    ClassGUID            : {ECFB0CFD-74C4-4F52-BBF7-343461CD72AC}
    DeviceID             : USB\VID_04D8&PID_FA2E\LUSBW1
    DeviceInterfaceGUID  : {1920C2F3-91A4-DB54-83C3-891298961B7D}
    SymbolicLink         : \\?\usb#vid_04d8&pid_fa2e#lusbw1#{1920c2f3-91a4-db54-83c3-891298961b7d}
    DevicePath           : \\?\usb#vid_04d8&pid_fa2e#lusbw1#{1920c2f3-91a4-db54-83c3-891298961b7d}
    SerialNumber         : LUSBW1
    BusNumber            : 0
    DeviceAddress        : 1

Select device (1-1) :1

Loading driver api..
Getting descriptors..

-Device:
  bLength             :18
  bDescriptorType     :0x01
  bcdUSB              :0x0200
  bDeviceClass        :0x00 (Defined at Interface level)
  bDeviceSubClass     :0x00
  bDeviceProtocol     :0x00
  bMaxPacketSize0     :8
  idVendor            :0x04D8 (Microchip Technology, Inc.)
  idProduct           :0xFA2E
  bcdDevice           :0x0001
  iManufacturer       :1 (Travis Robinson)
  iProduct            :2 (Benchmark Device)
  iSerialNumber       :3 (LUSBW1)
  bNumConfigurations  :1
!End Device
-Configuration:
  bLength             :9
  bDescriptorType     :0x02
  wTotalLength        :55
  bNumInterfaces      :1
  bConfigurationValue :0x01
  iConfiguration      :0
  bmAttributes        :0xC0
  MaxPower            :50 (100ma)
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x00
    bNumEndpoints       :2
    bInterfaceClass     :0x00 (Defined at Interface level)
    bInterfaceSubClass  :0x00
    bInterfaceProtocol  :0x00
    iInterface          :0
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x01
      bmAttributes        :0x05 (Isochronous, Asynchronous, Data Endpoint)
      wMaxPacketSize      :0
      bInterval           :0x01
   !End Endpoint
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x81
      bmAttributes        :0x05 (Isochronous, Asynchronous, Data Endpoint)
      wMaxPacketSize      :0
      bInterval           :0x01
   !End Endpoint
 !End Interface
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x01
    bNumEndpoints       :2
    bInterfaceClass     :0x00 (Defined at Interface level)
    bInterfaceSubClass  :0x00
    bInterfaceProtocol  :0x00
    iInterface          :0
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x01
      bmAttributes        :0x05 (Isochronous, Asynchronous, Data Endpoint)
      wMaxPacketSize      :64
      bInterval           :0x01
   !End Endpoint
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x81
      bmAttributes        :0x05 (Isochronous, Asynchronous, Data Endpoint)
      wMaxPacketSize      :64
      bInterval           :0x01
   !End Interface
 !End Configuration

diff --git a/iso.c b/iso.c
index 547acaa..65a9ce1 100644
--- a/iso.c
+++ b/iso.c
@@ -69,7 +69,7 @@ get_packet_length(libusb_device *dev, unsigned char ep, int iface, int alts)
     struct libusb_config_descriptor *config;

     LIBUSB_CHECK(libusb_get_active_config_descriptor(dev, &config));
-
+       printf("number of alt interface is %d \n", config->interface[iface].num_altsetting);
     assert(alts < config->interface[iface].num_altsetting);

     const struct libusb_interface_descriptor *intf = config->interface[iface].altsetting + alts;
@@ -79,11 +79,13 @@ get_packet_length(libusb_device *dev, unsigned char ep, int iface, int alts)
     for (int i = 0; i < intf->bNumEndpoints; ++i) {
         const struct libusb_endpoint_descriptor *endpoint = intf->endpoint + i;
         if (endpoint->bEndpointAddress == ep) {
+                       printf("found EP = %d \n", ep);
             packet_size = endpoint->wMaxPacketSize;
+                       printf("packet_size = %u \n", packet_size);
             break;
         }
     }
-
+       printf("packet_size = %u \n", packet_size);
     assert(packet_size);
     return packet_size;
 }

PS C:\work\libusb\libusb_test\isochronous\isochronous_new> .\iso-test.exe -d 04d8:fa2e -i 0 -a 1 -e 1 -n 8192 -v
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: info [windows_init] UsbDk backend is not available
number of alt interface is 2
packet_size = 0
Assertion failed: packet_size, file iso.c, line 89
```

mcuee, 2022-04-06:
In any case, stm32_isoc and sam3u_isoc will fail for my USB devices, be it FS, HS or SS, with or without #1109.
Symptom: with libusbK -- actual length =0.
Symptom: with WinUSB -- error 87.

tormodvolden, 2022-04-06:
> Somehow I can not get your version of isochronous to work.
It is because you forgot to specify the -u OUT endpoint. Otherwise it defaults to an endpoint not existing on your device, and you get the "Assertion failed: packet_size" error. It would still work if you were only doing IN transfers with the "in" argument.

tormodvolden, 2022-04-06:
OK, thanks for all the testing of #1109. I will only apply it after 1.0.26 (since it doesn't solve any known problems).

I get libusbK IN errors with my test device and test programs, but not with libfreenect and the XBox camera, so maybe the test program (or device firmware) is to blame.

mcuee, 2022-04-06:
> Somehow I can not get your version of isochronous to work.
> It is because you forgot to specify the -u OUT endpoint. Otherwise it defaults to an endpoint not existing on your device, and you get the [...]

I see. Maybe a "-h" option is good to have.

libusbk and WinUSB have the same output, a bit strange for the IN transfer but may be possible.

```
C:\work\libusb\libusb_test\isochronous [adapt ≡ +2 ~1 -0 !]> .\iso-test.exe -d 04d8:fa2e -i 0 -a 1 -e 81 -u 1 -n 512
Packet 0 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 1 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 2 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 3 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 4 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 5 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 6 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 7 endpoint 01 requested transfer 64 bytes, transferred 64 bytes.
Packet 0 endpoint 81 requested transfer 64 bytes, transferred 64 bytes.
Packet 1 endpoint 81 requested transfer 64 bytes, transferred 64 bytes.
Packet 2 endpoint 81 requested transfer 64 bytes, transferred 64 bytes.
Packet 3 endpoint 81 requested transfer 64 bytes, transferred 64 bytes.
Packet 4 endpoint 81 requested transfer 64 bytes, transferred 0 bytes.
Packet 5 endpoint 81 requested transfer 64 bytes, transferred 0 bytes.
Packet 6 endpoint 81 requested transfer 64 bytes, transferred 0 bytes.
Packet 7 endpoint 81 requested transfer 64 bytes, transferred 0 bytes.

```
[libusbk_debug_benchmark_pic.zip](https://github.com/libusb/libusb/files/8425830/libusbk_debug_benchmark_pic.zip)
[winusb_debug_benchmark_pic.zip](https://github.com/libusb/libusb/files/8425831/winusb_debug_benchmark_pic.zip)

mcuee, 2022-04-06:
> OK, thanks for all the testing of #1109. I will only apply it after 1.0.26 (since it doesn't solve any known problems).
> 
> I get libusbK IN errors with my test device and test programs, but not with libfreenect and the XBox camera, so maybe the test program (or device firmware) is to blame.

Yes that can be the case. In the case of Cystream or Cypress FX3 isoc_loopback or isoc_sourcesink FW, I am pretty sure the firmware is good as they are from Cypress and I have tested them with Cypress tools or libusbK. So I was thinking the test program is to blame.

tormodvolden, 2022-04-06:
> Symptom: with libusbK -- actual length =0.

Is this still a seen, and in which case? Apart from the reported value being 0, is there any way you can check if the transfer actually works and the data is received by the device? This only affects OUT, right? I've lost a bit track now because of all the different combinations of hardware/driver/program/parameters.

I also cannot get the XBox camera to work with iso-test but that is probably because the camera needs setting registers via a HID interface to start streaming - it doesn't have the standard alternate interfaces for streaming on/off.

mcuee, 2022-04-06:
> Is this still a seen, and in which case? Apart from the reported value being 0, is there any way you can check if the transfer actually works and the data is received by the device? This only [...]

It only affects IN. 
[
](https://github.com/libusb/libusb/issues/1104#issuecomment-1088164738)

mcuee, 2022-04-06:
Let me try it out under Linux first. And I will concentrate on the stm32-isoc with CyStream FW on the FX2LP (04b4:1003) high speed device, it is a more complicated device with multiple alt interfaces.

```
~/build/cyusb/cyusb_linux$ ./bin/cyusb 
Found device 04b4 8613 (Default Cypress USB device)
No of devices of interest found = 1
Found Hex format file
Releasing FX2 CPU from reset
Signal 10 (=SIGUSR1) received !
Found device 04b4 1003 (FX2LP CY-Stream Device)
No of devices of interest found = 1

~/build/cyusb/cyusb_linux$ lsusb -v -d 04b4:1003

Bus 003 Device 008: ID 04b4:1003 Cypress Semiconductor Corp. CY-Stream
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04b4 Cypress Semiconductor Corp.
  idProduct          0x1003 
  bcdDevice            0.00
  iManufacturer           1 Cypress
  iProduct                2 CY-Stream
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0087
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
[... 121 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

```

mcuee, 2022-04-06:
stm32_isoc modification.
```
#define VID 0x04b4
#define PID 0x1003
#define IFACE 0
#define ALTS 6
#define EP_ISO_IN  0x82
#define EP_ISO_OUT 0x06
#define PACKET_LEN 1024
```

mcuee, 2022-04-07:
Typical output of stm32_isoc.

```
PS C:\work\libusb\libusb_test\isoc_stm32> .\stm32_isoc.exe -n8192
request endpoint:82 length:8192, xfr length (bulk):8192, total (isoch):8192
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
...
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
134 transfers (total 16384 bytes) in 1637 milliseconds => 10008 bytes/sec
pack 0 endpoint:82 length:1024, actual_length:0
pack 1 endpoint:82 length:1024, actual_length:0
pack 2 endpoint:82 length:1024, actual_length:0
pack 3 endpoint:82 length:1024, actual_length:0
pack 4 endpoint:82 length:1024, actual_length:0
pack 5 endpoint:82 length:1024, actual_length:0
pack 6 endpoint:82 length:1024, actual_length:0
pack 7 endpoint:82 length:1024, actual_length:0
request endpoint:82 length:8192, xfr length (bulk):0, total (isoch):0
...

```

mcuee, 2022-04-07:
@tormodvolden I noticed some typo with my older comments. IN becomes OUT and OUT becomes IN. So I have corrected them. Sorry for the confusions caused. Looks like IN has never been working for libusbk with the Cypress devices. This has been the case with all different libusb based application (sam3u_isoc, stm32_isoc, isochronous, or libcyusb example 09) under Windows.

The FW and host application like stm32_isoc, isochronous or libcyusb example 09 are all good. 
1) FW has been cross-verified by Cypress tools (under Windows, with Cypress driver) or libusbk kbench (using libusbk or WinUSB driver under Windows).
2) stm32_isoc, isochronous or libcyusb example 09 are good as well since I have tested them under Linux.

So #1105 is good as it fixed Isoc OUT. Not so sure if we need something similar for Isoc IN.

mcuee, 2022-04-07:
libcystream 09 example seems to work for bulk IN/OUT, and Isoc OUT but not Isoc IN.

```
C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_isoc_performance.exe: Starting test with the following parameters
        Request size     : 0x10
        Queue depth      : 0x10
        Test duration    : 0x1
        Endpoint to test : 0x2

        Endpoint type    : 0x1
        Max packet size  : 0x200
Transfer Counts: 16 pass 0 fail
Data rate: 3510.504086 KBps

Transfer Counts: 32 pass 0 fail
Data rate: 3993.635144 KBps

Transfer Counts: 48 pass 0 fail
Data rate: 4005.883642 KBps

C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_isoc_performance.exe: Starting test with the following parameters
        Request size     : 0x10
        Queue depth      : 0x10
        Test duration    : 0x1
        Endpoint to test : 0x82

        Endpoint type    : 0x1
        Max packet size  : 0x200
Transfer Counts: 0 pass 16 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 32 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 48 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 64 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 80 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 96 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 112 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 128 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 144 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 160 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 176 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 192 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 208 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 224 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 240 fail
Data rate: 0.000000 KBps

Transfer Counts: 0 pass 256 fail
Data rate: 0.000000 KBps

C:\work\libusb\libusb_test\libcyusb_port_to_windows\src\09_cystream_isoc_performance.exe: Test duration is complete. Stopping transfers
16 requests are pending
Transfer Counts: 0 pass 272 fail
Data rate: 0.000000 KBps

13 requests are pending

```

mcuee, 2022-04-07:
> The only exception seems to be iso-test which seems to work for both IN and OUT.

```
C:\work\libusb\libusb_test\isochronous [adapt ≡ +8 ~1 -0 !]> .\iso-test.exe -d 04b4:1003 -i 0 -a 6 -e 82 -u 6 -n 8192
Packet 0 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 0 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.

```

mcuee, 2022-04-07:
> The only exception seems to be iso-test which seems to work for OUT and then work for IN for Cystream FW and partially for FS/SS USB (first 4-packets for FS USB, first 8 packets for SS USB)

I have tried Cystream and it seems to work fine with libusbk driver, for both IN and OUT. And I have tried different transfer size: 8192, 16384, 32768, 65536, 655360.

```
C:\work\libusb\libusb_test\isochronous [adapt ≡ +13 ~1 -0 !]> .\iso-test.exe -d 04b4:1003 -i 0 -a 6 -e 82 -u 6 -n 16384
Packet 0 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 8 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 9 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 10 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 11 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 12 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 13 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 14 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 15 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 0 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 8 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 9 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 10 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 11 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 12 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 13 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 14 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 15 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
```

mcuee, 2022-04-07:
BTW, iso-test also works well with WinUSB driver despite the error 87.

```
C:\work\libusb\libusb_test\isochronous [adapt ≡ +15 ~1 -0 !]> .\iso-test.exe -d 04b4:1003 -i 0 -a 6 -e 82 -u 6 -n 16384
Packet 0 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 8 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 9 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 10 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 11 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 12 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 13 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 14 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 15 endpoint 06 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 0 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 1 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 2 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 3 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 4 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 5 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 6 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 7 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 8 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 9 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 10 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 11 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 12 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 13 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 14 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.
Packet 15 endpoint 82 requested transfer 1024 bytes, transferred 1024 bytes.

C:\work\libusb\libusb_test\isochronous [adapt ≡ +16 ~1 -0 !]> $Env:LIBUSB_DEBUG=4
C:\work\libusb\libusb_test\isochronous [adapt ≡ +16 ~1 -0 !]> .\iso-test.exe -d 04b4:1003 -i 0 -a 6 -e 82 -u 6 -n 16384 -v >winusb_isotest_working_cystream_verbose_debug.log 2>&1
```

tormodvolden, 2022-04-07:
I would be careful judging the different actual_length and sizes as "works" or not. E.g. #1107 doesn't really fix anything in the transfer, it just "fakes" a better reported value than the 0 value (that the driver probably doesn't even try to set). Of course, a more correct value might have repercussions in the application, which otherwise would judge it as failed. However we should try to determine if data actually is transferred correctly. If the test device has an echo functionality that's easier, or if the device has some other means (like the analog voltage in my first tests) to see if data transfer was successful.

OTOH, when we see the actual_length change in the middle of a transfer there is something going on.

mcuee, 2022-04-07:
> I would be careful judging the different actual_length and sizes as "works" or not.

Good point.

> If the test device has an echo functionality that's easier, or if the device has some other means (like the analog voltage in my first tests) to see if data transfer was successful.

Two of the Isoc FW devices I have do support loopback (IN will echo back OUT) function. It is just I need to figure out the way to enable the function, probably your extension to the isoc_test is good enough for testing with the verbose option to look at the data (but need to be modified to enter the loopback test mode).

1) Benchmark FW for PIC MCU and Cypress FX3 -- use control transfer to select the test mode (write, read or loopback)
2) CyStream FW for FX2LP (need to check whether it is possible to enter loopback mode).
3) Cypress Isoc-loopback -- already in the loopback mode.
4) Cypress Isoc-sourcesink -- (need to check whether it is possible to enter loopback mode)

mcuee, 2022-04-08:
The pyusb bmfw is a fork of the PIC Benchmark FW and it has the loopback enable already. It is actually the corresponding FW for the iso-test app.
https://github.com/walac/bmfw

Here is the sample output: first 64 bytes transfer is a success but not the later.
```
PS C:\work\libusb\libusb_test\isochronous\isochronous_new> .\iso-test.exe -n 128 -v
number of alt interface is 3
found EP = 3
packet_size = 32
packet_size = 32
Packet 0 endpoint 03 requested transfer 32 bytes, transferred 32 bytes.
Packet 1 endpoint 03 requested transfer 32 bytes, transferred 32 bytes.
Packet 2 endpoint 03 requested transfer 32 bytes, transferred 32 bytes.
Packet 3 endpoint 03 requested transfer 32 bytes, transferred 32 bytes.
number of alt interface is 3
found EP = 131
packet_size = 32
packet_size = 32
Packet 0 endpoint 83 requested transfer 32 bytes, transferred 32 bytes.
Packet 1 endpoint 83 requested transfer 32 bytes, transferred 32 bytes.
Packet 2 endpoint 83 requested transfer 32 bytes, transferred 0 bytes.
Packet 3 endpoint 83 requested transfer 32 bytes, transferred 0 bytes.
 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

[28 more replies]