libusb/libusb #1105: WinUSB isoch OUT doesn't report packet actual_length

tormodvolden, 2022-04-03:
Running isochronous OUT transfers on WinUSB, the log shows
[windows_handle_transfer_completion] handling transfer 00000000009B9D70 completion with errcode 0, length 0
and the application receives 0 in the `libusb_iso_packet_descriptor` `actual_length` member.

Interestingly, the winusbx_copy_transfer_data() WinUSB code updates actual_length for IN but not for OUT.

[iso-test-winusb.log](https://github.com/libusb/libusb/files/8404454/iso-test-winusb.log)
[iso-test-linux.log](https://github.com/libusb/libusb/files/8404455/iso-test-linux.log)

mcuee, 2022-04-03:
Same here. This time I am using USB PIC Benchmark FW from libusbk project. Isoc transfer will report transfer of 0 bytes, for both libusbk and winusb driver.
Host program: https://github.com/walac/isochronous with some debug printf.

```
Successfully found pyusb benchmark firmware device.
Successfully set configuration 1.
Successfully claimed Interface 0.
Successfully claimed Interface 0 alt setting 1.
packet size is 64 .
n= 64 .
Packet 0 requested to transfer 64 bytes, transfered 0 bytes.
packet size is 64 .
Packet 0 requested to transfer 64 bytes, transfered 64 bytes.
```
[winusb_debug.zip](https://github.com/libusb/libusb/files/8404640/winusb_debug.zip)
[libusbk_debug.zip](https://github.com/libusb/libusb/files/8404641/libusbk_debug.zip)

tormodvolden, 2022-04-03:
Thanks for confirming it happens on other devices and for libusbK as well.

It has been like this in winusbx_copy_transfer_data() since WinUSB isoch was introduced in #284. I think it is because Windows usbd doesn't report the actual length for OUT transfers (it is maybe pointless anyway). However Linux and macOS seem to fill in these values, although I don't know if they have any significance or are just copied from the requested length. I started preparing a PR for this (WinUSB only) in #1107, it would be great if you can test it. But it needs extending to libusbK before it should be merged.

tormodvolden, 2022-04-03:
When I tested the WinUSB fix in #1107, I also tested on libusbK, and there the situation is worse. OUT transfers are reaching the device, but the returned actual_length values are zero, even if the current libusbK code section seems to update the value. The first IN transfer looks good, with correct actual_length values. However, when rerunning "iso-test", which first does OUT transfer, then IN transfer, the data received on the second IN was all zeroed, and actual_length zero here also.
[iso-test-libusbk-first.log](https://github.com/libusb/libusb/files/8405560/iso-test-libusbk-first.log)
[iso-test-libusbk-second.log](https://github.com/libusb/libusb/files/8405561/iso-test-libusbk-second.log)

tormodvolden, 2022-04-03:
I added the same fix for libusbK  in #1107, and now the OUT test is fine.

For IN, I get the same error 87 as mentioned in #1104 (appararently this part is not limited to High/SuperSpeed). This seems related to state in the driver or host controller, because when I replug it, the first run works again, but if I reset the microcontroller (without reenumeration) nothing changes.

mcuee, 2022-04-04:
Great. Now it seems to work for my USB PIC Benchmark FW, with pull request #1107 applied. I still see the error #87 though for Isoc IN. Unplug and replug do not help in my case.

```
PS C:\work\libusb\isochronous> .\iso-test.exe
Successfully found pyusb benchmark firmware device.
Successfully set configuration 1.
Successfully claimed Interface 0.
Successfully claimed Interface 0 alt setting 1.
packet size is 64 .
n= 64 .
Packet 0 requested to transfer 64 bytes, transfered 64 bytes.
packet size is 64 .
Packet 0 requested to transfer 64 bytes, transfered 64 bytes.
```
[winusb_debug_USB_FS.zip](https://github.com/libusb/libusb/files/8406344/winusb_debug_USB_FS.zip)
[libusbk_debug_USB_FS.zip](https://github.com/libusb/libusb/files/8406345/libusbk_debug_USB_FS.zip)

mcuee, 2022-04-04:
#1107 seems to be good to me.

tormodvolden, 2022-04-04:
Yes, #1107 only changes what is reported back to the application, it will likely not affect deeper issues like the error 87.

commit bfbef179 (2022-04-04): windows: Update isochronous OUT packet actual length

On Windows, unlike other platforms, the isochronous packet actual_length
value is not set on completion of OUT transfers. However, our API
requires the user to check this value for isochronous transfers instead
of the transfer actual_length, if the transferred length is of interest.

The usbd Length field is not used for isochronous OUT transfers:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usb/ns-usb-_usbd_iso_packet_descriptor

To make it consistent with other platforms, just return the requested
length.

Fixes #1105
Clo [... cut]
closed 2022-04-04 by tormodvolden