libusb/libusb #1106: Coverity report for core.c

mcuee, 2022-04-03:
1 new defect(s) introduced to libusb/libusb found with Coverity Scan.
1 defect(s), reported by Coverity Scan earlier, were marked fixed in the recent build analyzed by Coverity Scan.

New defect(s) Reported-by: Coverity Scan
Showing 1 of 1 defect(s)

** CID 351373:  Memory - corruptions  (OVERRUN)
/libusb/core.c: 2646 in log_v()

________________________________________________________________________________________________________
*** CID 351373:  Memory - corruptions  (OVERRUN)
/libusb/core.c: 2646 in log_v()
2640                    text_len = (int)sizeof(buf) - header_len;
2641            }
2642            if (header_len + text_len + (int)sizeof(USBI_LOG_LINE_END) >= (int)sizeof(buf)) {
2643                    /* Need to truncate the text slightly to fit on the terminator. */
2644                    text_len -= (header_len + text_len + (int)sizeof(USBI_LOG_LINE_END)) - (int)sizeof(buf);
2645            }
>>>     CID 351373:  Memory - corruptions  (OVERRUN)
>>>     "strcpy" will overrun its first argument "buf + header_len + text_len" which can accommodate 1 bytes.  The length of the second argument ""\n"" is 2 bytes, including the terminating null.
2646            strcpy(buf + header_len + text_len, USBI_LOG_LINE_END);
2647     
2648            log_str(level, buf);
2649     
2650            /* Per-context log handler */
2651     #ifndef ENABLE_DEBUG_LOGGING

benzea, 2022-04-03:
I have a patch for this one in #1067 (i.e. https://github.com/libusb/libusb/pull/1067/commits/3c792a9b21ec07d9a3f369b9a79eec1e0f999823).

I never understood the warning though. I think coverity is just confused and takes a branch incorrectly for some reason.

mcuee, 2026-03-12:
We will need to run coverity again. Close this one as it seems not a real issue.

closed 2026-03-12 by mcuee