libusb/libusb #1108: segmentation fault: 11 libusb @1.0.25_0 under macOS

chriskuku, 2022-04-03:
When doing an `st-info --probe`  I'm occasionally getting 

```
  dev-type:   unknown
libusb: debug [darwin_event_thread_main] darwin event thread exiting
Segmentation fault: 11

```
Why?

tormodvolden, 2022-04-03:
Can you please try with libusb 1.0.26rc-1 ? It is possible that the application code is doing something wrong, but still it shouldn't segfault in libusb. Do you have the source code to your st-info?

tormodvolden, 2022-04-03:
Once you have libusb 1.0.26-rc1 e.g. with homebrew
```
brew unlink libusb
brew install libusb --HEAD
```

run with libusb debugging and attach the output:
`LIBUSB_DEBUG=4 st-info --probe`
or
`LIBUSB_DEBUG=4 st-info --probe > st-info-debug.log 2>&`1

chriskuku, 2022-04-04:
Thanks. I’m not using homebrew. Can you lead me to git checkout  or how to do that using macports?
—
Christoph

chriskuku, 2022-04-04:
> Can you please try with libusb 1.0.26rc-1 ? It is possible that the application code is doing something wrong, but still it shouldn't segfault in libusb. Do you have the source code to your st-info?

I checked it out yesterday or before yesterday from https://github.com/stlink-org/stlink
Do you want me to send you a tar ball from my checked out directory? Otherwise you can check out the source from github.
--
Christoph

tormodvolden, 2022-04-04:
You can download libusb binaries from Code -> Releases here on GitHub, or build yourself from a git checkout. macports has a libusb-devel package but it is unfortunately not up to date.

Thanks, if your st-info is latest/recent git from the official stlink repo then it's fine.

chriskuku, 2022-04-04:
I followed the homebrew advice above but it seems I'm getting into troubled waters with it:

```
$ brew install libusb --HEAD
Running 'brew update --preinstall'...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> Updated Formulae
Updated 3 formulae.
==> Updated Casks
Updated 6 casks.

==> Downloading https://ghcr.io/v2/homebrew/core/automake/manifests/1.16.5
Already downloaded: /Users/kuku/Library/Caches/Homebrew/downloads/6863d3ffa73c5a5cbf1f1ea9e47b60784d09b232ac9e56149c1b931d9ec44d8b--automake-1.16.5.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/automake/blobs/sha256:ae77a247a13ea860236a29b02769f532
Already downloaded: /Users/kuku/Library/Caches/Homebrew/downloads/25e6690671ae9c3741dac14ff17e9904701143701ad52a24789d60a14dce5077--automake--1.16.5.big_sur.bottle.tar.gz
==> Downloading https://ghcr.io/v2/homebrew/core/libtool/manifests/2.4.7
Already downloaded: /Users/kuku/Library/Caches/Homebrew/downloads/583066a50f4afbd11e5d6bd1d66fe9c16c2770a6d8a9f1b1dc5be930fa7194c5--libtool-2.4.7.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libtool/blobs/sha256:bae30841faca7f4b91e5388c0db13cd6c
Already downloaded: /Users/kuku/Library/Caches/Homebrew/downloads/fe7470debd69c21262f5d2692a4fef57e5626651d8a4f5088c4827a18312c303--libtool--2.4.7.big_sur.bottle.tar.gz
==> Cloning https://github.com/libusb/libusb.git
Updating /Users/kuku/Library/Caches/Homebrew/libusb--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at bfbef17 windows: Update isochronous OUT packet actual length
Warning: Your Xcode (13.1) is outdated.
Please update to Xcode 13.2.1 (or delete it).
Xcode can be updated from the App Store.

==> Installing dependencies for libusb: automake and libtool
==> Installing libusb dependency: automake
==> Pouring automake--1.16.5.big_sur.bottle.tar.gz
🍺  /usr/local/Cellar/automake/1.16.5: 131 files, 3.5MB
==> Installing libusb dependency: libtool
==> Pouring libtool--2.4.7.big_sur.bottle.tar.gz
🍺  /usr/local/Cellar/libtool/2.4.7: 75 files, 3.8MB
==> Installing libusb --HEAD
==> ./autogen.sh
==> ./configure --prefix=/usr/local/Cellar/libusb/HEAD-bfbef17
==> make install
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink include/libusb-1.0/libusb.h
Target /usr/local/include/libusb-1.0/libusb.h
already exists. You may want to remove it:
  rm '/usr/local/include/libusb-1.0/libusb.h'

To force the link and overwrite all conflicting files:
  brew link --overwrite libusb

To list all files that would be deleted:
  brew link --overwrite --dry-run libusb

Possible conflicting files are:
/usr/local/include/libusb-1.0/libusb.h
/usr/local/lib/libusb-1.0.0.dylib
/usr/local/lib/libusb-1.0.a
/usr/local/lib/libusb-1.0.dylib -> /usr/local/lib/libusb-1.0.0.dylib
/usr/local/lib/pkgconfig/libusb-1.0.pc
==> Summary
🍺  /usr/local/Cellar/libusb/HEAD-bfbef17: 22 files, 532.2KB, built in 54 seconds
==> Running `brew cleanup libusb`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
==> Upgrading 5 dependents of upgraded formulae:
Disable this behaviour by setting HOMEBREW_NO_INSTALLED_DEPENDENTS_CHECK.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
gnutls 3.6.16 -> 3.7.4, guile 3.0.7 -> 3.0.8, ffmpeg 4.4_2 -> 5.0, imagemagick 7.1.0-4_2 -> 7.1.0-29, libmicrohttpd 0.9.73 -> 0.9.75
```

tormodvolden, 2022-04-04:
Did you forget the unlink comment that I posted?

chriskuku, 2022-04-05:
> Did you forget the unlink comment that I posted?

No, I did:
$ brew unlink libusb
Unlinking /usr/local/Cellar/libusb/1.0.24... 0 symlinks removed.
$ brew install libusb --HEAD
Updating Homebrew...
....
I now did:
$ brew link --overwrite libusb
Linking /usr/local/Cellar/libusb/1.0.24... 6 symlinks created.

But that's probably not what I want, do I?

How do I proceed from here now?

chriskuku, 2022-04-05:
You said, macport libusb-devel is broken. Broken from which point of view? macports-wise? I did a port install libusb-devel
and switched 

$ port installed libusb
The following ports are currently installed:
  libusb @1.0.24_0
  libusb @1.0.25_0
$ port installed libusb-devel
The following ports are currently installed:
  libusb-devel @20220221-74059154_0 (active)
$ 
$ ~/Downloads/stlink/build/Release/bin/st-info --probe
Failed to parse flash type or unrecognized flash type
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V2J37S26
  serial:     0679FF525056805087052236
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000

detected chip_id parametres

# Device Type: unknown
# Reference Manual: RM0000
#
chip_id 0x0
flash_type 0
flash_size_reg 0x0
flash_pagesize 0x0
sram_size 0x0
bootrom_base 0x0
bootrom_size 0x0
option_base 0x0
option_size 0x0
flags 0

  dev-type:   unknown
libusb: debug [darwin_event_thread_main] darwin event thread exiting
Segmentation fault: 11

Although the SIG11 doesn't occur everytime I invoke st-info.

chriskuku, 2022-04-05:
Question: I found the binary release 1.0.26-rc1 binaries for macOS. How do I install these under macOS?

chriskuku, 2022-04-05:
OK, a new attempt:
[log1.txt](https://github.com/libusb/libusb/files/8419946/log1.txt)

chriskuku, 2022-04-06:
BTW, found that the stlink utilities in question are statically linked, so the DYLD_LIBRARY_PATH was 

> You can download libusb binaries from Code -> Releases here on GitHub, or build yourself from a git checkout. macports has a libusb-devel package but it is unfortunately not up to date.
>
Someone from macports pointed out that libusb-devel is at libusb-devel: update to 20220403-bfbef17 
at the moment. Would that fit for testing?

chriskuku, 2022-04-06:
BTW, found that the stlink utilities in question are statically linked, so the DYLD_LIBRARY_PATH was 

> You can download libusb binaries from Code -> Releases here on GitHub, or build yourself from a git checkout. macports has a libusb-devel package but it is unfortunately not up to date.
>

The macport is at:
libusb-devel: update to 20220403-bfbef17 

Hope, the following log is with the correct version now:

[log.txt](https://github.com/libusb/libusb/files/8425048/log.txt)

mcuee, 2022-04-06:
This is good already in terms of libusb version.
```
$ LIBUSB_DEBUG=4 ./st-info --probe
Failed to parse flash type or unrecognized flash type
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000034] [0011fe19] libusb: debug [libusb_init] created default context
[ 0.000056] [0011fe19] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
```

chriskuku, 2022-04-06:
But what about the segfault?
libusb: debug [darwin_event_thread_main] darwin event thread exiting
Segmentation fault: 11

mcuee, 2022-04-06:
I thought the segfault comes from st-link and not libusb.

tormodvolden, 2022-04-06:
You'd need to run it in a debugger, e.g:
```
LIBUSB_DEBUG=4 gdb --args ./st-info --probe
(gdb) run
...
(gdb) backtrace
```

tormodvolden, 2022-04-06:
> The macport is at: libusb-devel: update to 20220403-bfbef17

So they updated it since the first time you tried:

chriskuku, 2022-04-06:
> I thought the segfault comes from st-link and not libusb.

The funny thing was, everytime I got the segfault 11 at the same time the thread_exit message in libusb occured. But I could not catch the segfault message using >log.txt 2>&1 . It appeared in the Terminal. So what I sent was a copy/paste from the terminal contents.

mcuee, 2022-04-06:
> The funny thing was, everytime I got the segfault 11 at the same time the thread_exit message in libusb occured. But I could not catch the segfault message using >log.txt 2>&1 . It appeared in the [...]

It is okay, I just reopened the issue. Please run what Tormod suggested to get the backtrace by using gdb or lldb.

tormodvolden, 2022-04-06:
> But I could not catch the segfault message using >log.txt 2>&1

The Windows terminal console is so ridiculously slow and it holds back execution while printing messages. So it changes the timing and in this case there is a race condition between threads which makes it crash intermittently.

mcuee, 2022-04-06:
> > But I could not catch the segfault message using >log.txt 2>&1
> 
> The Windows terminal console is so ridiculously slow and it holds back execution while printing messages. So it changes the timing and in this case there is a race condition between threads which makes it crash intermittently.

macOS terminal should be not bad.

Under Windows I use Windows Terminal and it seems to be pretty okay.

tormodvolden, 2022-04-06:
Right, this was macOS :) Still the macOS Terminal is also slow in my experience. On Linux, the Gnome or Mate terminals can be dogs too. The fastest is good old xterm :)

chriskuku, 2022-04-06:
```

  dev-type:   unknown
[ 0.018425] [00009d33] libusb: debug [libusb_close]  
[ 0.018429] [00009d33] libusb: debug [libusb_release_interface] interface 0
[ 0.018681] [00009d33] libusb: debug [libusb_exit]  
libusb: debug [darwin_event_thread_main] darwin event thread exiting
[New Thread 0x3317 of process 2769]

Thread 4 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x3317 of process 2769]
0x00000000a073770d in ?? ()
(gdb) backtrace
#0  0x00000000a073770d in ?? ()
#1  0x000000010000de3d in usbi_log ()
#2  0x000000010001624f in darwin_event_thread_main ()
#3  0x00007fff2065a8fc in ?? ()
#4  0x0000000000000000 in ?? ()
(gdb) 
```

Unfortunately I don't have symbols. Would I have to buid libusb with symbols/debug or the debugee (stlink)?

tormodvolden, 2022-04-06:
It looks like one of the issues that showed up in 1.0.25 when applications do things in the wrong order. It would be easier to pin-point with debug symbols from libusb and stlink, but I think the issue is that the context used for logging in frame #1 was already deleted when the application called libusb_exit() earlier.

mcuee, 2022-04-06:
Just for info, no issues here with st-link release version (1.7.0) under my Mac Mini M1 with an ST Nucleo F303ZE, using libusb git head.

```
mcuee@mcuees-Mac-mini ~ % brew info stlink 
stlink: stable 1.7.0 (bottled), HEAD
STM32 discovery line Linux programmer
https://github.com/texane/stlink
/opt/homebrew/Cellar/stlink/1.7.0 (30 files, 1.4MB) *
  Poured from bottle on 2021-04-26 at 19:45:49
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/stlink.rb
License: BSD-3-Clause
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: libusb ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 428 (30 days), 1,314 (90 days), 5,995 (365 days)
install-on-request: 429 (30 days), 1,316 (90 days), 5,988 (365 days)
build-error: 15 (30 days)

mcuee@mcuees-Mac-mini ~ % st-info --probe 
Found 1 stlink programmers
  version:    V2J38S27
  serial:     066CFF565251887067013337
  flash:      524288 (pagesize: 2048)
  sram:       65536
  chipid:     0x0446
  descr:      F303 high density
```

mcuee, 2022-04-06:
No issue with st-link git head either.

```
mcuee@mcuees-Mac-mini ~ % brew install stlink --HEAD
==> Cloning https://github.com/texane/stlink.git
Cloning into '/Users/mcuee/Library/Caches/Homebrew/stlink--git'...
==> Checking out branch develop
Already on 'develop'
Your branch is up to date with 'origin/develop'.
==> cmake .
==> make install
🍺  /opt/homebrew/Cellar/stlink/HEAD-c4762e6: 95 files, 1.5MB, built in 6 seconds
==> Running `brew cleanup stlink`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
mcuee@mcuees-Mac-mini ~ % st-info --probe           
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32F302_F303_F398_HD
# Reference Manual: RM0365           // also RM0316 (Rev 5)
#
chip_id 0x446
flash_type 1
flash_size_reg 0x1ffff7cc
flash_pagesize 0x800
sram_size 0x10000
bootrom_base 0x1fffd800
bootrom_size 0x2000
option_base 0x1ffff800
option_size 0x10
flags 2

Found 1 stlink programmers
  version:    V2J38S27
  serial:     066CFF565251887067013337
  flash:      524288 (pagesize: 2048)
  sram:       65536
  chipid:     0x446

detected chip_id parametres

# Device Type: STM32F302_F303_F398_HD
# Reference Manual: RM0365           // also RM0316 (Rev 5)
#
chip_id 0x446
flash_type 1
flash_size_reg 0x1ffff7cc
flash_pagesize 0x800
sram_size 0x10000
bootrom_base 0x1fffd800
bootrom_size 0x2000
option_base 0x1ffff800
option_size 0x10
flags 2

  dev-type:   STM32F302_F303_F398_HD

```

mcuee, 2022-04-06:
@kriskiku
Just to rule out hardware stability issue, is this a know good hardware? What if you use OpenOCD (git head, release version has issues with libusb git head)?

chriskuku, 2022-04-07:
Thanks. The problem occurs with aspecial target attached to the SWD port of the nucleo. That is, a DIYMORE 
https://stm32-base.org/boards/STM32F407VGT6-diymore.html

```
$ ./st-info --probe
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32F411xC_xE
# Reference Manual: RM0383
#
chip_id 0x431
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x20000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x0
option_size 0x0
flags 2

Found 1 stlink programmers
  version:    V2J37S26
  serial:     0679FF525056805087010236
  flash:      524288 (pagesize: 16384)
  sram:       131072
  chipid:     0x431

detected chip_id parametres

# Device Type: STM32F411xC_xE
# Reference Manual: RM0383
#
chip_id 0x431
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x20000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x0
option_size 0x0
flags 2

  dev-type:   STM32F411xC_xE
libusb: debug [darwin_event_thread_main] darwin event thread exiting
Segmentation fault: 11
$ 
```

chriskuku, 2022-04-07:
>

tormodvolden, 2022-04-07:
For discussion, an extract from the log. It contains 3 runs where only the last segfaults.
```
egrep 'LIBUSB|init|libusb_exit|fault|warning|thread_main' log.txt 
$ LIBUSB_DEBUG=4 ./st-info --probe
[ 0.000034] [0011fe19] libusb: debug [libusb_init] created default context
[ 0.000056] [0011fe19] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.009543] [0011fe1c] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.009747] [0011fe1c] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.010632] [0011fe19] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.014437] [0011fe19] libusb: warning [libusb_init] installing new context as implicit default
[ 0.018343] [0011fe19] libusb: debug [libusb_exit] destroying default context
[ 0.018692] [0011fe19] libusb: debug [libusb_exit]  
$ LIBUSB_DEBUG=4 ./st-info --probe
[ 0.000033] [0011fe21] libusb: debug [libusb_init] created default context
[ 0.000055] [0011fe21] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.010251] [0011fe24] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.010448] [0011fe24] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.011434] [0011fe21] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.015115] [0011fe21] libusb: warning [libusb_init] installing new context as implicit default
[ 0.018801] [0011fe21] libusb: debug [libusb_exit] destroying default context
[ 0.019140] [0011fe21] libusb: debug [libusb_exit]  
$ LIBUSB_DEBUG=4 ./st-info --probe
[ 0.000035] [0011fe2a] libusb: debug [libusb_init] created default context
[ 0.000059] [0011fe2a] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.009697] [0011fe2d] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.009875] [0011fe2d] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.010672] [0011fe2a] libusb: debug [libusb_init] libusb v1.0.26.11720-rc1
[ 0.014625] [0011fe2a] libusb: warning [libusb_init] installing new context as implicit default
[ 0.018526] [0011fe2a] libusb: debug [libusb_exit] destroying default context
[ 0.018871] [0011fe2a] libusb: debug [libusb_exit]  
libusb: debug [darwin_event_thread_main] darwin event thread exiting
Segmentation fault: 11
```
and this is what we have of a stacktrace (copied from above):
```
  dev-type:   unknown
[ 0.018425] [00009d33] libusb: debug [libusb_close]  
[ 0.018429] [00009d33] libusb: debug [libusb_release_interface] interface 0
[ 0.018681] [00009d33] libusb: debug [libusb_exit]  
libusb: debug [darwin_event_thread_main] darwin event thread exiting
[New Thread 0x3317 of process 2769]

Thread 4 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x3317 of process 2769]
0x00000000a073770d in ?? ()
(gdb) backtrace
#0  0x00000000a073770d in ?? ()
#1  0x000000010000de3d in usbi_log ()
#2  0x000000010001624f in darwin_event_thread_main ()
#3  0x00007fff2065a8fc in ?? ()
#4  0x0000000000000000 in ?? ()
(gdb) 
```
Thinking loud: The `darwin event thread exiting` we see here is the last use of usbi_log() inside darwin_event_thread_main(). So the call that we see in the backtrace must be from another instance. We're supposed to have only one of these running. Has a new one been spawned at this time? The debugger also prints "New Thread" just before but I don't know if these messages are in chronological order or it is delayed. @hjelmn Do you have any idea?

Also, in the 2 first runs the message `darwin event thread exiting` is not printed. I suppose this is because ctx or ctx->log_handler is NULL in log_v() when it gets there.

@chriskuku Symbols would still be nice :) I don't know if your build is missing symbols or if it is just the debugger which cannot find or resolve them. Can you please also reconfigure with --enable-debug-log and attach a new debug log?

I have a suspicion the application is doing something wrong, seeing it uses multiple contexts, some default, some not. But I am not able to confirm that by perusing the log. And anyway I don't see why it would segfault, also with the extra safe-guarding measures we have introduced recently (like @benzea's commit 6b29aeb9).

tormodvolden, 2022-04-07:
> Not sure where I can see the checked out version, is i what I get from -version?
> ```
> v1.7.0-186-gc4762e6
> ```
Yes, the "g" in the string means git commit, so it is c4762e6. All good.

benzea, 2022-04-07:
Ahh, an easy explanation:
 * The application creates two contexts, ctx_a and ctx_b (first a, then b)
 * The `darwin_event_thread_main` stores the pointer to `ctx_a`
 * `ctx_a` is destroyed, but `init_count` is non-zero still
 * `ctx_b` is destroyed, but the logging code is still accessing `ctx_a`

i.e. the thread needs to either:
 * hold on to the context, or
 * switch to another context to pass for logging, or
 * just not try to log to a specific context ever.

Below some outdated thoughts.
----

So, looking at the stack trace and the debug output, I see:
 * We are still in `darwin_event_thread_main`
 * It just logged `darwin event thread exiting`
 * We are in a call to `usbi_log`, i.e. the call probably didn't return yet (even if it printed the message)
 * Which means, probably `ctx->log_handler` is invalid, i.e. we get SIGSEGV when we try to execute the log handler function

Now, various possibilities:
 * Use-after-free, associated with the pointer being wrong at that point -> this does not make any sense, we have a 
 * The application is doing something wrong with the log handler
 * Just a plain old memory corruption

benzea, 2022-04-07:
So, really:
```
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index a457f95..aa21580 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -543,7 +543,9 @@ static void *darwin_event_thread_main (void *arg0) {
 
   usbi_dbg (ctx, "darwin event thread ready to receive events");
 
-  /* signal the main thread that the hotplug runloop has been created. */
+  /* signal the main thread that the hotplug runloop has been created.
+   * Note that ctx may not be valid anmore after this. */
+  ctx = NULL;
   pthread_mutex_lock (&libusb_darwin_at_mutex);
   libusb_darwin_acfl = runloop;
   libusb_darwin_acfls = libusb_shutdown_cfsource;
@@ -553,7 +555,7 @@ static void *darwin_event_thread_main (void *arg0) {
   /* run the runloop */
   CFRunLoopRun();
 
-  usbi_dbg (ctx, "darwin event thread exiting");
+  usbi_dbg (NULL, "darwin event thread exiting");
 
   /* signal the main thread that the hotplug runloop has finished. */
   pthread_mutex_lock (&libusb_darwin_at_mutex);
```

chriskuku, 2022-04-07:
I was trying to build libusb with 
./configure --enable-debug-log
make
make install

The odd thing is: when I start gdb the first time (and that also was the case with all previous attempts for which I sent the logs), it seems to hang forever after issuing the run command. I have to kill the process (^C doesn't work) and start again Then it runs:

```
$ LIBUSB_DEBUG=4 gdb --args ./st-info --probe
GNU gdb (GDB) 11.2
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin20.6.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./st-info...
(gdb) whatis main
type = int (int, char **)
(gdb) run
Starting program: /Users/kuku/Downloads/stlink/build/Debug/bin/st-info --probe
[New Thread 0x2003 of process 16408]
[New Thread 0x1c03 of process 16408]

```

tormodvolden, 2022-04-07:
@chriskuku With --enable-debug-log we would maybe see more in detail what happens, and on the other hand not get the segfault. But benzea's analysis explains everything. It would be great if you can apply his suggested patch and test.

@benzea This makes sense to me, thanks a lot. I understand that the event thread should be context neutral (shared between them), it just happens to be started by the first context being initialized. I would like to include this in 1.0.26, it seems perfectly safe. Will you make a PR?

In general, maybe it would be easier to debug similar issues if we include the context ID in more of our debug messages.

benzea, 2022-04-07:
@tormodvolden, MR created, see #1119. And, thanks for your analysis, you had all the important bits of information already!

chriskuku, 2022-04-07:
@tormodvolden : I can confirm that I didn't get a segfault yet with the --enable-debug-log compiled version. That's why I could post a backtrace yet, how hard I ever tried :)

chriskuku, 2022-04-07:
I'm a little gotten unused to patching. I copied/pasted @benzea post above and did the following:

```
$ cat >patch
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index a457f95..aa21580 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -543,7 +543,9 @@ static void *darwin_event_thread_main (void *arg0) {
 
   usbi_dbg (ctx, "darwin event thread ready to receive events");
 
-  /* signal the main thread that the hotplug runloop has been created. */
+  /* signal the main thread that the hotplug runloop has been created.
+   * Note that ctx may not be valid anmore after this. */
+  ctx = NULL;
   pthread_mutex_lock (&libusb_darwin_at_mutex);
   libusb_darwin_acfl = runloop;
   libusb_darwin_acfls = libusb_shutdown_cfsource;
@@ -553,7 +555,7 @@ static void *darwin_event_thread_main (void *arg0) {
   /* run the runloop */
   CFRunLoopRun();
 
-  usbi_dbg (ctx, "darwin event thread exiting");
+  usbi_dbg (NULL, "darwin event thread exiting");
 
   /* signal the main thread that the hotplug runloop has finished. */
   pthread_mutex_lock (&libusb_darwin_at_mutex);
$ 
$ pwd
/Users/kuku/Downloads/libusb/libusb/os
$ patch -p <patch
patch: option requires an argument -- p
patch: Try `patch --help' for more information.
$ patch  <patch
patching file darwin_usb.c
$ ls -l darwin_*
-rw-r--r--  1 kuku  staff   98625  7 Apr 13:07 darwin_usb.c
-rw-r--r--  1 kuku  staff    6844  5 Apr 11:47 darwin_usb.h
-rw-r--r--  1 kuku  staff     285  7 Apr 11:22 darwin_usb.lo
-rw-r--r--  1 kuku  staff  169920  7 Apr 11:22 darwin_usb.o
$ pwd
/Users/kuku/Downloads/libusb/libusb/os
$ cd ..
$ cd ..
$ ls
AUTHORS			README.git		config.guess		install-sh
COPYING			README.md		config.h		libtool
ChangeLog		TODO			config.h.in		libusb
INSTALL			Xcode			config.log		libusb-1.0.pc
INSTALL_WIN.txt		aclocal.m4		config.status		libusb-1.0.pc.in
Makefile		android			config.sub		ltmain.sh
Makefile.am		appveyor.yml		configure		m4
Makefile.in		autogen.sh		configure.ac		missing
NEWS			autom4te.cache		depcomp			msvc
PORTING			bootstrap.sh		doc			stamp-h1
README			compile			examples		tests
$ pwd
/Users/kuku/Downloads/libusb
$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libusb
  CC       os/darwin_usb.lo
  CCLD     libusb-1.0.la
$ 
```

tormodvolden, 2022-04-07:
It is probably easier to apply by hand in this case. You don't need the comments :)

chriskuku, 2022-04-07:
I'm pretty sure I applied the patch successfully. Also built, installed libusb. Rebuilt stlink. The segfault 11 still occured once in 20 attempts. From what can I see whether the (statically linked) st-info binary was linked agains the patched libusb?

benzea, 2022-04-07:
> I'm pretty sure I applied the patch successfully. Also built, installed libusb. Rebuilt stlink. The segfault 11 still occured once in 20 attempts. From what can I see whether the (statically linked) st-info binary was linked agains the patched libusb?

Maybe just change one of the log messages? Then you can see whether the string changed which would proof you picked up the right version.

chriskuku, 2022-04-07:
> Maybe just change one of the log messages? Then you can see whether the string changed which would
> proof you picked up the right version.

Couldn't believe that you pro guys suggest such stone age method but indeed, I had the same thought and so I did. I changed libusb: in core.c to be my CPK (initials) libusb: and , guess what, it did not fire a single SIG11 in about 100 tries.

I now did a port uninstall libusb and a brew remove libusb first and did a 
make clean all install in libusb. After that: 

```
-rwxr-xr-x  1 root  admin  193376  7 Apr 14:53 /usr/local/lib/libusb-1.0.0.dylib
-rw-r--r--  1 root  admin  534776  7 Apr 14:53 /usr/local/lib/libusb-1.0.a
lrwxr-xr-x  1 root  admin      18  7 Apr 14:53 /usr/local/lib/libusb-1.0.dylib -> libusb-1.0.0.dylib
-rwxr-xr-x  1 root  admin     943  7 Apr 14:53 /usr/local/lib/libusb-1.0.la
```

```
[ 0.029343] [00056af6] CPK libusb: debug [usbi_remove_event_source] remove fd 3
Found 1 stlink programmers
  version:    V2J37S26
  serial:     0679FF525056805087052236
  flash:      524288 (pagesize: 16384)
  sram:       131072
  chipid:     0x431

detected chip_id parametres

# Device Type: STM32F411xC_xE
# Reference Manual: RM0383
#
chip_id 0x431
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x20000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x0
option_size 0x0
flags 2

  dev-type:   STM32F411xC_xE
[ 0.029424] [00056af6] CPK libusb: debug [libusb_close]  
[ 0.029428] [00056af6] CPK libusb: debug [libusb_release_interface] interface 0
[ 0.029698] [00056af6] CPK libusb: debug [libusb_exit]  
[ 0.029725] [00056afa] CPK libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.029945] [00056af6] CPK libusb: debug [libusb_unref_device] destroy device 20.5
[ 0.030061] [00056af6] CPK libusb: debug [usbi_remove_event_source] remove fd 65
$ 
$ 
$ 
```

Looks fine. Now I can turn to the SWD-wise not functioning target. Thanks to all.

tormodvolden, 2022-04-07:
Thanks for reporting the issue and all your testing.

commit ff05357a (2022-04-09): darwin: Fix crash in log handler when stopping event thread

The darwin event thread (in contrast to other OS implementations) tries
to log to the context that created it. However, this context is only
guaranteed to be valid until the thread has started. It may be that
another context is the last one to be destroyed, causing the event
thread to log using an already destroyed context.

Fix this by only passing on ctx where it is acceptable.

Fixes #1108
closed 2022-04-09 by tormodvolden