libusb/libusb #1110: Different behaviour of `libusb_cancel_transfer` on MacOS / Darwin is not documented

martinling, 2022-04-04:
On Darwin, the kernel does not make it possible to cancel a transfer on an endpoint except by aborting all transfers on that endpoint, so `libusb_cancel_transfer` does just that. This OS-specific behaviour is not mentioned in the [API documentation](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#ga685eb7731f9a0593f75beb99727bbe54) for that function. Please document this.

This was raised previously in #549, where it was suggested that this is "documented somewhere", but it doesn't seem to be. The only mentions I found of this libusb behaviour were in that issue, and in [this mailing list thread](https://libusb-devel.narkive.com/wCCiU2eB/libusb-for-the-macintosh), from 13 years ago when Darwin support was first being added to libusb-0.1.

Something else that would be worth documenting here is that `libusb_cancel_transfer` on Darwin calls `ClearPipeStallBothEnds` after aborting transfers, which causes a `CLEAR_FEATURE` / `ENDPOINT_HALT` request to be sent to the device. This does not happen on other operating systems, and doesn't seem to be a common request, so some devices fail to implement it correctly.

In both #549 and our own [recent issue on HackRF](https://github.com/greatscottgadgets/hackrf/issues/1052), this caused a problem due to missing or broken handling of that request. The correct result is to reset the data toggle bit for the specified endpoint and ACK the request, as we [fixed here](https://github.com/greatscottgadgets/hackrf/pull/1081). If the device does not reset the data toggle bit, the next data packet on that endpoint can be silently dropped, leading to data loss. If it fails to ACK the request, things will hang while the host retries it.

hjelmn, 2022-04-04:
That was intended to be documented but looks like it got lost. Probably don't even have the branch with that change anymore.

As for the usage of ```ClearPipeStallBothEnds```. It is recommended by Apple after the abort to clear the data toggle. That seemed desirable at the time and this is the first I have heard of any issue (13 years later).

From the documentation:

```
If there are outstanding asynchronous transactions on the pipe, the callbacks will happen. Note that this command will also clear the halted bit on the endpoint in the controller, but will NOT clear the data toggle bit.  If you want to clear the data toggle bit as well, see @link //apple_ref/C/instm/IOUSBInterfaceInterface/ClearPipeStall/ ClearPipeStall @/link or @link //apple_ref/C/instm/IOUSBInterfaceInterface190/ClearPipeStallBothEnds/ ClearPipeStallBothEnds @/link for more information.  The interface must be open for the pipe to exist.
```

Changing this would be a change of behavior and I do not know who relies on the data toggle being cleared.

martinling, 2022-04-04:
To clarify, I wasn't suggesting to remove the usage of `ClearPipeStallBothEnds`. As far as I can see it's a perfectly correct thing to do, so long as the device on the other end implements the request correctly. I presume that it was added because someone found that the abort could leave the data toggle out of sync otherwise.

I was just suggesting that it would be helpful to mention this behaviour in the documentation, given that it differs from other operating systems and that there's been at least two cases where a device has failed to implement that request, leading to problems. It's not exactly an easy thing to debug from the symptoms.

tormodvolden, 2022-04-04:
Hmm, recommended by Apple or not, changing behaviour or not, I still think there is value in making libusb as consistent as possible across platforms. But I don't know the details here. Will removing ClearPipeStallBothEnds() make it work like on Linux and Windows, or will it cause other problems on macOS? Or should the other backends be fixed?

tormodvolden, 2022-04-04:
BTW @martinling has HackRF been tested on libusb 1.0.26-RC1? We welcome test reports in #1099.

hjelmn, 2022-04-05:
Ah. Sure. The issue is it is hard to keep consistency for cancel between platforms. We should at least document this for sure.

martinling, 2022-04-05:
I have found conflicting information about whether `AbortPipe` resets the data toggle bit or not.

The text that @hjelmn quoted above says:

>  Note that this command will also clear the halted bit on the endpoint in the controller, but will NOT clear the data toggle bit.

However, the current HTML documentation [here](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-630.4.5/Documentation/IOUSBLib/Functions/InterfaceInterface.html) says exactly the opposite:

> Note that this command will also clear both the halted bit and the data toggle bit on the endpoint in the controller.

Digging through the IOUSBFamily [version history](https://opensource.apple.com/source/IOUSBFamily/), I found where this text changed:
- In versions up to [281.4.1](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-281.4.1/IOUSBFamily/Headers/IOUSBLib.h.auto.html), `AbortPipe` resets the data toggle bit.
- In versions from [303.4.5](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-303.4.5/IOUSBFamily/Headers/IOUSBLib.h.auto.html), `AbortPipe` does NOT reset the data toggle bit.

The change only appears in the header comments in IOUSBLib.h. The HTML documentation seems to have never been updated, so in versions after 303.4.5 the two sources contradict each other within the same release.

I don't know whether this was a functional change between 281.4.1 and 303.4.5, or a correction about existing behaviour.

But either way, it seems that on 303.4.5 or later it should be fine to remove the call to `ClearPipeStallBothEnds` entirely. That call is needed to cause the device to clear its data toggle bit, when the host has already cleared its own by calling `AbortPipe`. If the data toggle has not been cleared, there is no need to do this. And it's safer not to do it, because devices might not implement that request correctly.

On versions 281.4.1 and earlier, libusb must assume the data toggle was cleared, so it should call `ClearPipeStallBothEnds` to ensure that the device's data toggle bit to stay in sync with the host's.

Version 303.4.5 shipped with Mac OS X 10.5 "Leopard" in 2007, so this should only matter for very old systems at this point. On all Mac OS versions from the last 15 years, it looks like it should be safe to change this.

martinling, 2022-04-05:
> BTW @martinling has HackRF been tested on libusb 1.0.26-RC1? We welcome test reports in #1099.

Not to my knowledge. Are there particular changes in this release which you think might affect us? I see some discussion of isochronous transfers, but we don't actually use them - our streaming is implemented with bulk transfers.

hjelmn, 2022-04-05:
> The change only appears in the header comments in IOUSBLib.h. The HTML documentation seems to have never been updated, so in versions after 303.4.5 the two sources contradict each other within the same release.

I likely used the online HTML version when developing the feature. Apple has been terrible about updating the documentation. The first node that shows up in Google (https://developer.apple.com/documentation/iokit/iousbinterfaceinterface650/1559196-abortpipe?chan_1) is empty  probably due to how the headers are written.

> I don't know whether this was a functional change between 281.4.1 and 303.4.5, or a correction about existing behaviour.
>
> But either way, it seems that on 303.4.5 or later it should be fine to [...]

Probably true. Before 10.5 we had to always clear the device toggle but it still appears to be recommended. The only issue I have with removing the call is libusb provides no way to clear the toggle if their software has been relying on it being cleared by cancel. Other than that I see no reason not to remove it as the documentation clearly says macOS no longer clears the host side data toggle. 

> On versions 281.4.1 and earlier, libusb must assume the data toggle was cleared, so it should call `ClearPipeStallBothEnds` to ensure that the device's data toggle bit stays in sync with the host's.

Easy enough. I prefer not to support 281.4.1 at all but we still have developers who wish it to be supported so we can protect the call to only occur with older versions of IOUSBFamily.

martinling, 2022-04-05:
> The only issue I have with removing the call is libusb provides no way to clear the toggle if their software has been relying on it being cleared by cancel.

I'm having trouble imagining what software could be relying on the existing behaviour, because the data toggle is such a low level part of the protocol - one that is not visible to or directly controlled by userspace. Indeed, part of how we tracked down this problem in HackRF is that I spotted the libusb message saying:
```
[darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
```
...and immediately thought this was a very suspicious thing for userspace to be trying to do. It turned out of course that this *was* necessary, at least before 10.5 - but only because a call to `AbortPipe` had already had the unnecessary side effect of clearing the data toggle.

From the point of view of a userspace program, the choice of whether the data toggle has been cleared on both sides, or left intact on both sides, ought to be completely unimportant.

The one scenario I can see in which this change might break something is in a buggy program recovering from a halt. If there is some code out there which can cause a device to return a STALL, and which also tries to recover from that condition, then it _should_ call `libusb_cancel_transfer` followed by `libusb_clear_halt`. There could be some program in which the `libusb_clear_halt` call is missing, but which still works because `libusb_cancel_transfer` already clears the halt. Removing the `ClearPipeStallBothEnds` from `darwin_abort_transfers` would break that program. However, it would already fail to work correctly on any other operating system, since this behaviour is specific to Darwin.

So the scenarios I can see where this may break something are limited to a program:
- Cause a device to STALL and halt an endpoint.
- Include code to recover from this condition.
- Are using the asynchronous API.
- Call `libusb_cancel_transfer` without calling `libusb_clear_halt`, which is not supposed to work.
- Have only been tested on MacOS.

There may be other possibilities that I'm not thinking of, but if that _is_ the only scenario, it must be a vanishingly small fraction of programs. And if something _does_ break like this, it should end up returning noticeable errors when it next calls `libusb_submit_transfer` on a halted endpoint. A one line fix to the program would be needed to add the missing call to `libusb_clear_halt`.

On the other hand, the current behaviour can cause problems for any device that does not correctly implement the `CLEAR_FEATURE` request, which is probably a pretty widespread issue given that it does not seem to be commonly used, and is not required for libusb to work on other operating systems. The consequence can be silent data loss without any reported error, and fixing the behaviour on the device side may be impossible.

I think that makes a very strong case for changing this.

hjelmn, 2022-04-06:
I propose we update the documentation to reflect the same suggestion that Apple makes. Canceling a transfer may cause a pipe stall and could require a call to ```libusb_clear_halt```. This is in addition to documenting that we can not cancel a single transfer on macOS, only all transfers.

hjelmn, 2022-04-06:
#1117 will close this.

martinling, 2022-04-06:
> I disagree. libusb may be a userspace library but macOS gives it control over the data toggle. This allows us to implement `libusb_clear_halt` and allowed me to clear the data toggle in this case. It is unnecessary now but it is not wrong.

I didn't say it was wrong, only that I found it suspicious. Once I learned why this was done, I agreed it was the correct thing to do. It is only in the later versions that it has become unnecessary.

> Oh, I agree. But a buggy device that can't handle this basic USB control function will break on a pipe stall anyway.

Only if a pipe stall actually occurs. In practice, many devices may never return a `STALL`, so unless there is a bus error, there is no reason for an endpoint to become halted, and thus no need for a `ClearFeature(ENDPOINT_HALT)` handler.

> I think the only case for changing the behavior is to match the behavior as close to other OSs as possible. Nothing in libusb documentation states the behavior if the cancelation causes the endpoint to stall. You are correct, however, that a well written program will clear the halt if one is detected but it is up to libusb whether or not to actively clear a pipe stall. I chose, based on documentation, to always clear the data toggle.

I don't understand what you mean about cancellation causing the endpoint to stall - I've raised this point on #1117.

> I will likely update the code to not clear the data toggle but I have never been a fan of working around buggy devices. The device should handle clearing the data toggle. If the device can not then `libusb_clear_halt` (which clear the data toggle) will not work and a correct program will not work with the device anyway.

I do not think a correct program ever needs to call `libusb_clear_halt` unless a stall has occurred. This is unrelated to cancellation as I understand it, except recovery from a stall requires any existing transfers to be cancelled before calling `libusb_clear_halt`.

martinling, 2022-04-06:
> I propose we update the documentation to reflect the same suggestion that Apple makes.

I don't think Apple is still making this recommendation. In current versions of IOUSBFamily, this recommendation only appears in the [HTML documentation](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-630.4.5/Documentation/IOUSBLib/Functions/InterfaceInterface.html) of the functions, and the footer on that document indicates that it has not been updated since 2003. The only description that has been updated is the one in the [header file](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-630.4.5/IOUSBFamily/Headers/IOUSBLib.h.auto.html, which has said since ~2006 that the data toggle bit is not reset by `AbortPipe`, and no longer says that the data toggle bit may need to be resynchronized.

I do not think that Apple's failure to update the HTML copy since 2003 represents a recommendation to ignore the later changes.

martinling, 2022-04-06:
Oh, I've just grasped what some the confusion is about I think.

The reason that `ClearPipeStallBothEnds` used to be needed was not because `AbortPipe` might have caused the endpoint to become stalled. It was only needed because `AbortPipe` had cleared the host-side data toggle bit - something that I think was a mistake in the design of that function really, and one that Apple have since corrected.

The only effect of `ClearPipeStallBothEnds` that was needed was its side effect of clearing the device-side data toggle bit, such that the two remained in sync. There was never any actual stall to be cleared on the device, unless the endpoint happened to be already stalled before the call to `libusb_cancel_transfer`.

The only time that `libusb_clear_halt` should be needed is when the endpoint is known to have actually stalled. In that case, the correct behaviour - on all platforms - would be to cancel remaining transfers with `libusb_cancel_transfer`, followed by calling `libusb_endpoint_halt` to clear the halt condition.

That is only scenario in which libusb should need to trigger a `ClearFeature(ENDPOINT_HALT)` request, as far as I can see.

hjelmn, 2022-04-07:
Exactly. I think I added to the confusion. IIRC if the bits get out of sync you will get a stall error code. The easiest way to deal with it is to go ahead and clear both. With the documentation I am reserving the right to remove the call in the cancel path. Since some devices do not implement it correctly I think we need to allow them to call reset instead.

martinling, 2022-04-08:
My concern is with the text proposed in #1117, which is not accurate. A call to `libusb_cancel_transfer` does not cause the endpoint to become stalled, and does not create a need to call `libusb_clear_halt`.

- On versions after OS X 10.5, the data toggle is not cleared on the host side, so the two sides remain in sync. There is no need for a call to `ClearPipeStallBothEnds` to clear the device side, and thus no need to add a `libusb_clear_halt` call.
- On versions before OS X 10.5, `libusb_cancel_transfer` will already call `ClearPipeStallBothEnds`. At that point there are two possible outcomes:
  - The device implements the request correctly, in which case the two sides are back in sync, so there is no need to add a call to `libusb_clear_halt`.
  - The device fails to implement the request, in which case calling `libusb_clear_halt` cannot help - because it will just repeat the call to `ClearPipeStallBothEnds` which already failed to solve the problem.
 
In that last case, the endpoint is still not actually stalled. It's just in a state where the data toggles are out of sync. Some future transfer will send a data packet with the wrong toggle bit, and that could cause a stall, though not necessarily. But that would be something that happens later, and calling `libusb_clear_halt` won't prevent it, or even recover from it when it happens.

When a data packet is sent with the wrong toggle bit, various things can happen. For an outbound transfer, the device could return a STALL and halt the endpoint. But it may also just NAK the packet, and could keep doing so until the transfer times out on the host. The host is not permitted to return a STALL under any condition (8.4.5), so on an inbound transfer all it can do is NAK the packet. It should probably eventually flag the endpoint and raise an error, but what we saw in practice was the silent loss of one packet's worth of data from the transfer.

When a `ClearFeature(ENDPOINT_HALT)` has been sent but not handled correctly by the device, the only reasonable action a program can take is to reset the device.

I think what the documentation should say is this:

> On macOS and iOS it is not possible to cancel a single transfer. On these operating systems, cancelling one transfer on an endpoint will cause all transfers on that endpoint to be cancelled. Your callback function will be invoked with a transfer status of `LIBUSB_TRANSFER_CANCELLED` for each transfer that was cancelled.
> 
> When built for Mac OS X versions prior to 10.5, this function sends a `ClearFeature(ENDPOINT_HALT)` request to the device. Prior to libusb 1.0.26, this request was sent on all Darwin systems. If this request is sent and the device does not handle it correctly, the data toggle bits on the host and device may be left out of sync. This could result in a stall on the next transmission, or to data in the next packet being lost. If faced with this situation, the only action possible may be to reset the device.

(substituting the correct version if this doesn't make it into 1.0.26)

tormodvolden, 2022-04-08:
I don't know if we should introduce a behavioral change now just before rolling the 1.0.26 binaries, since this will be a fix-up release with selected fixes. It could be argued that it fixes an inconsistency between platforms, but if this behavior is documented it is not a high-importance bug.

In any case we should update the documentation, and as per the existing code in a first step.

martinling, 2022-04-08:
Yeah it's up to you and your release process whether to include the functional change.

If that change is left out then the second part could say something like:

> On macOS and iOS, calling this function also sends a `ClearFeature(ENDPOINT_HALT)` request to the device. If the device does not handle this request correctly, the data toggle bits on the host and device can be left out of sync, which can result in a stall on the next transmission, or data in the next packet being lost. A call to `libusb_clear_halt` will not clear this condition, since that function uses the same request. Therefore, if your program runs on Darwin and uses a device that does not correctly implement `ClearFeature` requests, you should not use `libusb_cancel_transfer` unless it is followed by a call to `libusb_reset_device`.

tormodvolden, 2022-04-09:
Unless any of you make a PR, I'll commit this:

> On macOS and iOS it is not possible to cancel a single transfer. On these operating systems, cancelling one transfer on an endpoint will cause all transfers on that endpoint to be cancelled. Your callback function will be invoked with a transfer status of LIBUSB_TRANSFER_CANCELLED for each transfer that was cancelled.
> 
> On macOS and iOS, calling this function also sends a ClearFeature(ENDPOINT_HALT) request to the device. If the device does not handle this request correctly, the data toggle bits on the host and device can be left out of sync, which can result in a stall on the next transmission, or data in the next packet being lost. A call to libusb_clear_halt will not clear this condition, since that function uses the same request. Therefore, if your program runs on Darwin and uses a device that does not correctly implement ClearFeature requests, you should not use libusb_cancel_transfer unless it is followed by a call to libusb_reset_device.

martinling, 2022-04-09:
Since I proposed that text I suppose I should take the git blame for it - so a bit of further redrafting I've submitted this as #1121.

tormodvolden, 2022-04-09:
Perfect, I also didn't want to receive any assumed credit for this.

Thanks a lot for your contribution!

#1117 will be queued for 1.0.27.

commit f1fafe8a (2022-04-10): darwin: Document OS-specific behaviour of libusb_cancel_transfer

The behaviour of libusb_cancel_transfer is different on Darwin:

- Cancellation cancels all transfers on the same endpoint.

- A ClearFeature(ENDPOINT_HALT) request is sent after cancellation.

This documents both differences and their implications.

Fixes #1110
Closes #1121
closed 2022-04-10 by tormodvolden