libusb/libusb #1123: Boostrap issues in linux with 1.0.26 release

svonohr, 2022-04-11:
I just tried updating to the latest 1.0.26 release. However, the build produced an error on linux during the configure step:
```
...
checking if gcc recognizes -pthread... yes
checking for library containing pthread_create... -lpthread
checking for library containing __atomic_fetch_add_4... -latomic
../libusb/configure: line 16709: syntax error near unexpected token `UMOCKDEV,'
../libusb/configure: line 16709: `		PKG_CHECK_MODULES(UMOCKDEV, umockdev-1.0 >= 0.16.0, ac_have_umockdev=yes, ac_have_umockdev=no)'
```
I have generated the [configure](https://github.com/libusb/libusb/files/8461873/configure.txt) script myself using the `bootstrap.sh` since I checkout the source using git. It was generated in an Ubuntu 18.04 build environment, though I didn't have this problem in my Ubuntu VM. The build was fine with the previous libusb version.

mcuee, 2022-04-11:
Do you have issue with libusb-1.0.26 release tar ball?

It is not clear what you are trying to do. It could be the build-tool in Ubuntu 18.04 is too old.

demhademha, 2022-04-11:
I think the issue is 
```

../libusb/configure: line 16709: syntax error near unexpected token `UMOCKDEV,'
../libusb/configure: line 16709: `		PKG_CHECK_MODULES(UMOCKDEV, umockdev-1.0 >= 0.16.0, ac_have_umockdev=yes, ac_have_umockdev=no)'

```

tormodvolden, 2022-04-11:
You are missing autotools macros for pkg-config. bootstrap.sh is anyway only needed if you build from git, and is not meant for releases.

To build the 1.0.26 release you should download the official tar.bz tarball. It includes configure so you just run ./configure and make.

svonohr, 2022-04-11:
Thanks, installing the `pkgconf` package in Ubuntu solved the issue.

closed 2022-04-11 by mcuee

Timmmm, 2022-04-15:
Thanks @svonohr  - just ran into exactly the same problem and `sudo apt install pkg-config` worked. It would be nice if there was a better error message. Or if you guys could ditch autotools. It's 2022!

hjelmn, 2022-04-15:
Not switching from autotools. It works well and the other options are nowhere near as good.

hjelmn, 2022-04-15:
Since we the ```pkg-config``` macros are only used for ```umockdev``` I opened #1129 to allow configure to continue even if ```pkg-config``` is missing. This fixes a regression since we did not require these macros until fairly recently. I need to build a VM to test but if someone gets to this first please let me know if it works as intended.

Timmmm, 2022-04-15:
CMake is ... well it's still insane but even so it's *significantly* better than autotools, and much more popular. There are better build systems but CMake is pretty much the de facto standard these days. Anyway it sounds like you've already made up your mind so I won't try and persuade you!

Thanks for fixing this issue anyway. Fewer requirements are always better.