libusb/libusb #1124: 1.0.26 Release assert on darwin_init

a1ien, 2022-04-11:
In rust wrapper darwin asserted on simple test.
https://github.com/a1ien/rusb/blob/master/libusb1-sys/tests/test.rs#L20
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file os/darwin_usb.c, line 603. 
https://github.com/a1ien/rusb/runs/5970785517?check_suite_focus=true#step:7:32

In libusb we have same test https://github.com/libusb/libusb/blob/master/tests/stress.c#L29 but looks like we didn't run test on mac os.

I am also didn't have macos so I can't reproduce

mcuee, 2022-04-11:
I just tested 1.0.26 release and there is no issue under my Mac Mini M1 (macOS 12.3.1 ARM64). And you can see I have nested USB hubs with some USB devices attached.

Please help to check with the reporter to see if it is possible to provide more details, like the macOS version and 

```

mcuee@mcuees-Mac-mini libusb % ./tests/stress 
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests

mcuee@mcuees-Mac-mini libusb % lsusb1
Bus 999 Device 001: ID 05ac:AppleT8103USBXHCI Apple Inc. USB 3.1 Bus, AppleT8103USBXHCI
Bus 999 Device 001: ID 05ac:AppleT8103USBXHCI Apple Inc. USB 3.1 Bus, AppleT8103USBXHCI
Bus 999 Device 001: ID 05ac:AppleEmbeddedUSBXHCIFL1100 Apple Inc. USB 3.0 Bus, AppleEmbeddedUSBXHCIFL1100
Bus 002 Device 002: ID 0bda:0411 Generic 4-Port USB 3.0 Hub
Bus 002 Device 003: ID 0b95:1790 ASIX Elec. Corp. AX88179, Serial: 000000000001E9
Bus 002 Device 004: ID 0bda:0411 Generic 4-Port USB 3.0 Hub
Bus 002 Device 005: ID 05e3:0749 Generic USB3.0 Card Reader, Serial: 000000001536
Bus 002 Device 001: ID 0bda:5411 Generic 4-Port USB 2.0 Hub
Bus 002 Device 006: ID 0bda:5411 Generic 4-Port USB 2.0 Hub
Bus 002 Device 007: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT]
Bus 002 Device 008: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT]
Bus 002 Device 009: ID 1915:1025 ZY.Ltd ZY Control Mic
```

mcuee, 2022-04-11:
The CI from rusb seems to use libusb 1.0.25 version.
https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true

```
Run cargo test --workspace --verbose --all
  cargo test --workspace --verbose --all
  shell: /bin/bash -e {0}
  env:
    CARGO_TERM_COLOR: always
       Fresh pkg-config v0.3.[2](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:2)5
       Fresh cc v1.0.7[3](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:3)
       Fresh regex-syntax v0.6.25
       Fresh libc v0.2.122
       Fresh memchr v2.[4](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:4).1
       Fresh aho-corasick v0.7.18
   Compiling libusb1-sys v0.6.0 (/Users/runner/work/rusb/rusb/libusb1-sys)
       Fresh regex v1.[5](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:5).5
     Running `rustc --crate-name libusb1_sys --edition=2018 libusb1-sys/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,future-incompat --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --test -C metadata=2323930a9f[6](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:6)846d4 -C extra-filename=-2323930a9f6846d4 --out-dir /Users/runner/work/rusb/rusb/target/debug/deps -C incremental=/Users/runner/work/rusb/rusb/target/debug/incremental -L dependency=/Users/runner/work/rusb/rusb/target/debug/deps --extern libc=/Users/runner/work/rusb/rusb/target/debug/deps/liblibc-e35d6d2ad358f94[7](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:7).rlib -L native=/usr/local/Cellar/libusb/1.0.25/lib -l usb-1.0`
     Running `rustc --crate-name test --edition=201[8](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:8) libusb1-sys/tests/test.rs --error-format=json --json=diagnostic-rendered-ansi,future-incompat --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --test -C metadata=cd08f0[9](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:9)4d71299ea -C extra-filename=-cd08f094d71299ea --out-dir /Users/runner/work/rusb/rusb/target/debug/deps -C incremental=/Users/runner/work/rusb/rusb/target/debug/incremental -L dependency=/Users/runner/work/rusb/rusb/target/debug/deps --extern libc=/Users/runner/work/rusb/rusb/target/debug/deps/liblibc-e35d6d2ad358f947.rlib --extern libusb1_sys=/Users/runner/work/rusb/rusb/target/debug/deps/liblibusb1_sys-e608e6bebf1d9aa3.rlib -L native=/usr/local/Cellar/libusb/1.0.25/lib`
   Compiling rusb v0.9.0 (/Users/runner/work/rusb/rusb)
     Running `rustc --crate-name rusb --edition=2018 src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,future-incompat --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --test -C metadata=c6b13d1d91242d0d -C extra-filename=-c6b13d1d91242d0d --out-dir /Users/runner/work/rusb/rusb/target/debug/deps -C incremental=/Users/runner/work/rusb/rusb/target/debug/incremental -L dependency=/Users/runner/work/rusb/rusb/target/debug/deps --extern libc=/Users/runner/work/rusb/rusb/target/debug/deps/liblibc-e35d6d2ad358f947.rlib --extern libusb1_sys=/Users/runner/work/rusb/rusb/target/debug/deps/liblibusb1_sys-e608e6bebf1d9aa3.rlib --extern regex=/Users/runner/work/rusb/rusb/target/debug/deps/libregex-90bdc284956a8[10](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:10)b.rlib --cfg libusb_hotplug_get_user_data -L native=/usr/local/Cellar/libusb/1.0.25/lib`
    Finished test [unoptimized + debuginfo] target(s) in 10.48s
     Running `/Users/runner/work/rusb/rusb/target/debug/deps/libusb1_sys-2323930a9f6846d4`

running 0 tests

test result: ok. 0 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s
     Running `/Users/runner/work/rusb/rusb/target/debug/deps/test-cd08f094d7[12](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:12)99ea`

running 6 tests
test test_fill_bulk_transfer ... ok
test test_fill_control_transfer ... ok
test test_fill_control_setup ... ok
test test_version ... ok
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file os/darwin_usb.c, line 603.
error: test failed, to rerun pass '-p libusb1-sys --test test'

Caused by:
  process didn't exit successfully: `/Users/runner/work/rusb/rusb/target/debug/deps/test-cd08f094d71[29](https://github.com/a1ien/rusb/runs/5971098066?check_suite_focus=true#step:7:29)9ea` (signal: 6, SIGABRT: process abort signal)
Error: Process completed with exit code 101.
```

mcuee, 2022-04-11:
But it seems that I can reproduce the issue with rusb. So it seems to be a rusb probelm.

```
 mcuee@mcuees-Mac-mini rusb % cargo test --workspace --verbose --all

mcuee@mcuees-Mac-mini libusb % git clone https://github.com/a1ien/rusb.git
Cloning into 'rusb'...
remote: Enumerating objects: 2003, done.
remote: Counting objects: 100% (305/305), done.
remote: Compressing objects: 100% (148/148), done.
remote: Total 2003 (delta 216), reused 207 (delta 155), pack-reused 1698
Receiving objects: 100% (2003/2003), 1.60 MiB | 11.39 MiB/s, done.
Resolving deltas: 100% (1267/1267), done.
mcuee@mcuees-Mac-mini libusb % cd rusb                                    
mcuee@mcuees-Mac-mini rusb % cargo test --workspace --verbose --all     
    Updating crates.io index
   Compiling cc v1.0.73
   Compiling pkg-config v0.3.25
   Compiling libc v0.2.122
   Compiling memchr v2.4.1
   Compiling rusb v0.9.0 (/Users/mcuee/build/libusb/rusb)
   Compiling regex-syntax v0.6.25
     Running `rustc --crate-name cc --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/cc-1.0.73/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=7b0a6dd5d7761f14 -C extra-filename=-7b0a6dd5d7761f14 --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name pkg_config /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/pkg-config-0.3.25/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=009f644021626a1a -C extra-filename=-009f644021626a1a --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name build_script_build /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.122/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=259ac4164a779056 -C extra-filename=-259ac4164a779056 --out-dir /Users/mcuee/build/libusb/rusb/target/debug/build/libc-259ac4164a779056 -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name build_script_build --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/memchr-2.4.1/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=d2a70c4b9a120c3c -C extra-filename=-d2a70c4b9a120c3c --out-dir /Users/mcuee/build/libusb/rusb/target/debug/build/memchr-d2a70c4b9a120c3c -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name build_script_build --edition=2018 build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=9b369dd1945c466a -C extra-filename=-9b369dd1945c466a --out-dir /Users/mcuee/build/libusb/rusb/target/debug/build/rusb-9b369dd1945c466a -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps`
     Running `rustc --crate-name regex_syntax --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/regex-syntax-0.6.25/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="unicode"' --cfg 'feature="unicode-age"' --cfg 'feature="unicode-bool"' --cfg 'feature="unicode-case"' --cfg 'feature="unicode-gencat"' --cfg 'feature="unicode-perl"' --cfg 'feature="unicode-script"' --cfg 'feature="unicode-segment"' -C metadata=718f4134841ec3a4 -C extra-filename=-718f4134841ec3a4 --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `/Users/mcuee/build/libusb/rusb/target/debug/build/memchr-d2a70c4b9a120c3c/build-script-build`
     Running `/Users/mcuee/build/libusb/rusb/target/debug/build/libc-259ac4164a779056/build-script-build`
     Running `rustc --crate-name memchr --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/memchr-2.4.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=896ab57c429d8962 -C extra-filename=-896ab57c429d8962 --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name libc /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.122/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=3f635f1017ba60d2 -C extra-filename=-3f635f1017ba60d2 --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -L dependency=/Users/mcuee/build/libusb/rusb/ta [... cut ...]
     Running `/Users/mcuee/build/libusb/rusb/target/debug/deps/test-dc1ae4ce25b54bb6`

running 6 tests
test test_fill_bulk_transfer ... ok
test test_fill_control_transfer ... ok
test test_fill_control_setup ... ok
test test_version ... ok
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file darwin_usb.c, line 605.
error: test failed, to rerun pass '-p libusb1-sys --test test'

Caused by:
  process didn't exit successfully: `/Users/mcuee/build/libusb/rusb/target/debug/deps/test-dc1ae4ce25b54bb6` (signal: 6, SIGABRT: process abort signal)

```

a1ien, 2022-04-11:
Oh... yes you are right. CI use 1.0.25 version. This explain why when I bump to 1.0.25 we have no problem.
CI use 1.0.24 https://github.com/a1ien/rusb/runs/5034846662?check_suite_focus=true#step:7:14 when I think libusb build from source for mac

>But it seems that I can reproduce the issue with rusb. So it seems to be a rusb probelm.

Hmm.. interesting thing that rusb use pkg-config version of libusb and didn't build libusb from source.

a1ien, 2022-04-11:
I will close this issues. Because it's looks like problem in `rusb` but I can understand where is exactly.

So if you can help with investigating I will be appreciate

mcuee, 2022-04-11:
> I will close this issues. Because it's looks like problem in `rusb` but I can understand where is exactly.
> 
> So if you can help with investigating I will be appreciate

Yes I can help with the investigation. Just remember I am not a programmer but only a tester.

a1ien, 2022-04-11:
Can you try run `cargo test --verbose --features vendored` this must build and link static version of libusb

mcuee, 2022-04-11:
Not so sure why the static build failed. Take note it needs to link to the security framework. Not so sure if this is the reason though.

```
mcuee@mcuees-Mac-mini rusb % pkg-config --static --libs libusb-1.0
-L/opt/homebrew/Cellar/libusb/HEAD-4239bc3/lib -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security
```

mcuee, 2022-04-11:
More detailed log.

```

mcuee@mcuees-Mac-mini rusb % export RUST_BACKTRACE='full'            
mcuee@mcuees-Mac-mini rusb % cargo test --verbose --features vendored
       Fresh cc v1.0.73
       Fresh pkg-config v0.3.25
       Fresh regex-syntax v0.6.25
   Compiling libusb1-sys v0.6.0 (/Users/mcuee/build/libusb/rusb/libusb1-sys)
       Fresh libc v0.2.122
       Fresh memchr v2.4.1
       Fresh aho-corasick v0.7.18
       Fresh regex v1.5.5
     Running `/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-9edb2b1485255256/build-script-build`
error: failed to run custom build command for `libusb1-sys v0.6.0 (/Users/mcuee/build/libusb/rusb/libusb1-sys)`

Caused by:
  process didn't exit successfully: `/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-9edb2b1485255256/build-script-build` (exit code: 101)
  --- stdout
  cargo:vendored=1
  cargo:static=1

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', libusb1-sys/build.rs:85:6
  stack backtrace:
     0:        0x1047f7218 - std::backtrace_rs::backtrace::libunwind::trace::h6635c546760c01d7
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/../../backtrace/src/backtrace/libunwind.rs:90:5
     1:        0x1047f7218 - std::backtrace_rs::backtrace::trace_unsynchronized::h1672fde63351961a
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
     2:        0x1047f7218 - std::sys_common::backtrace::_print_fmt::h55998ee5d5c4bd49
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:67:5
     3:        0x1047f7218 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hcdaace98b5f99f33
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:46:22
     4:        0x10480ee40 - core::fmt::write::h4e46426f9591205d
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/core/src/fmt/mod.rs:1096:17
     5:        0x1047f3664 - std::io::Write::write_fmt::hd71db88d6aa1df24
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/io/mod.rs:1568:15
     6:        0x1047f95e4 - std::sys_common::backtrace::_print::heada759bfede9436
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:49:5
     7:        0x1047f95e4 - std::sys_common::backtrace::print::ha4efc73542d9f131
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:36:9
     8:        0x1047f95e4 - std::panicking::default_hook::{{closure}}::h744546d1495ac83a
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panicking.rs:208:50
     9:        0x1047f917c - std::panicking::default_hook::hfa180b571e8756b1
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panicking.rs:225:9
    10:        0x1047f9b98 - std::panicking::rust_panic_with_hook::h0ba91efd9a619f41
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panicking.rs:591:17
    11:        0x1047f9774 - std::panicking::begin_panic_handler::{{closure}}::h016fbcc3b9442b8a
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panicking.rs:497:13
    12:        0x1047f76d0 - std::sys_common::backtrace::__rust_end_short_backtrace::h7d78c8c5fa9da52e
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:141:18
    13:        0x1047f96dc - rust_begin_unwind
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panicking.rs:493:5
    14:        0x104814508 - core::panicking::panic_fmt::h2a9bc29c5973805e
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/core/src/panicking.rs:92:14
    15:        0x104814460 - core::option::expect_none_failed::h99a737a43330cbf7
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/core/src/option.rs:1300:5
    16:        0x104759a58 - core::result::Result<T,E>::unwrap::hd38fc2919285d645
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/core/src/result.rs:1037:23
    17:        0x1047581ec - build_script_build::make_source::h58e6c9c39850c05a
                                 at /Users/mcuee/build/libusb/rusb/libusb1-sys/build.rs:81:5
    18:        0x1047596b4 - build_script_build::main::h6bf840b9b23d4094
                                 at /Users/mcuee/build/libusb/rusb/libusb1-sys/build.rs:183:9
    19:        0x1047563f4 - core::ops::function::FnOnce::call_once::hf0ed04541316fe1d
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/core/src/ops/function.rs:227:5
    20:        0x104759ba0 - std::sys_common::backtrace::__rust_begin_short_backtrace::h46f0b47a8862fd99
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/sys_common/backtrace.rs:125:18
[... cut ...]
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/panic.rs:431:14
    26:        0x1047f9ef8 - std::rt::lang_start_internal::hbc1be5b0e12e91fd
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/rt.rs:51:25
    27:        0x1047560c0 - std::rt::lang_start::h8337ae00f33bbf1e
                                 at /rustc/2fd73fabe469357a12c2c974c140f67e7cdd76d0/library/std/src/rt.rs:65:5
    28:        0x10475972c - _main
```

a1ien, 2022-04-11:
Ahh this is because you need clone with submodule. 
`git clone --recursive` or run `git submodule update --init`

mcuee, 2022-04-11:
> Ahh this is because you need clone with submodule. `git clone --recursive` or run `git submodule update --init`

I see. No issues any more.
```

mcuee@mcuees-Mac-mini rusb % cargo test --verbose --features vendored
       Fresh pkg-config v0.3.25
       Fresh cc v1.0.73
       Fresh regex-syntax v0.6.25
   Compiling libusb1-sys v0.6.0 (/Users/mcuee/build/libusb/rusb/libusb1-sys)
       Fresh libc v0.2.122
       Fresh memchr v2.4.1
       Fresh aho-corasick v0.7.18
     Running `/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-9edb2b1485255256/build-script-build`
       Fresh regex v1.5.5
   Compiling rusb v0.9.0 (/Users/mcuee/build/libusb/rusb)
     Running `rustc --crate-name libusb1_sys --edition=2018 libusb1-sys/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="vendored"' -C metadata=c231060d34fbc7fe -C extra-filename=-c231060d34fbc7fe --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --extern libc=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibc-3f635f1017ba60d2.rmeta -L native=/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-b1a59d507529f9d7/out -l framework=CoreFoundation -l framework=IOKit -l framework=Security -l static=usb`
     Running `/Users/mcuee/build/libusb/rusb/target/debug/build/rusb-c10586c631a407c7/build-script-build`
     Running `rustc --crate-name rusb --edition=2018 src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="vendored"' -C metadata=37d4bc149cceff8c -C extra-filename=-37d4bc149cceff8c --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --extern libc=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibc-3f635f1017ba60d2.rmeta --extern libusb1_sys=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibusb1_sys-c231060d34fbc7fe.rmeta --cfg libusb_hotplug_get_user_data -L native=/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-b1a59d507529f9d7/out`
     Running `rustc --crate-name rusb --edition=2018 src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --test --cfg 'feature="vendored"' -C metadata=2db42a2363c32fdd -C extra-filename=-2db42a2363c32fdd --out-dir /Users/mcuee/build/libusb/rusb/target/debug/deps -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --extern libc=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibc-3f635f1017ba60d2.rlib --extern libusb1_sys=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibusb1_sys-c231060d34fbc7fe.rlib --extern regex=/Users/mcuee/build/libusb/rusb/target/debug/deps/libregex-cf2b2a5e07d22bad.rlib --cfg libusb_hotplug_get_user_data -L native=/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-b1a59d507529f9d7/out`
     Running `rustc --crate-name libusb_info --edition=2018 examples/libusb_info.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="vendored"' -C metadata=145e2d4024c010de --out-dir /Users/mcuee/build/libusb/rusb/target/debug/examples -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --extern libc=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibc-3f635f1017ba60d2.rlib --extern libusb1_sys=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibusb1_sys-c231060d34fbc7fe.rlib --extern regex=/Users/mcuee/build/libusb/rusb/target/debug/deps/libregex-cf2b2a5e07d22bad.rlib --extern rusb=/Users/mcuee/build/libusb/rusb/target/debug/deps/librusb-37d4bc149cceff8c.rlib --cfg libusb_hotplug_get_user_data -L native=/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-b1a59d507529f9d7/out`
     Running `rustc --crate-name list_devices --edition=2018 examples/list_devices.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="vendored"' -C metadata=145e2d4024c010de --out-dir /Users/mcuee/build/libusb/rusb/target/debug/examples -C incremental=/Users/mcuee/build/libusb/rusb/target/debug/incremental -L dependency=/Users/mcuee/build/libusb/rusb/target/debug/deps --extern libc=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibc-3f635f1017ba60d2.rlib --extern libusb1_sys=/Users/mcuee/build/libusb/rusb/target/debug/deps/liblibusb1_sys-c231060d34fbc7fe.rlib --extern regex=/Users/mcuee/build/libusb/rusb/target/debug/deps/libregex-cf2b2a5e07d22bad.rlib --extern rusb=/Users/mcuee/build/libusb/rusb/target/debug/deps/librusb-37d4bc149cceff8c.rlib --cfg libusb_hotplug_get_user_data -L native=/Users/mcuee/build/libusb/rusb/target/debug/build/libusb1-sys-b1a59d507529f9d7/out`
     Running `rustc --crate-name hotplug --edition=2018 examples/hotplug.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="vendored"' -C metadata=145e2d4024c010de --out-dir /Users/mcuee/ [... cut ...] -extern regex=/Users/mcuee/build/libusb/rusb/target/debug/deps/libregex-cf2b2a5e07d22bad.rlib --extern rusb=/Users/mcuee/build/libusb/rusb/target/debug/deps/librusb-37d4bc149cceff8c.rlib -C embed-bitcode=no --cfg 'feature="vendored"' --cfg libusb_hotplug_get_user_data`

running 2 tests
test src/fields.rs - fields::request_type (line 204) ... ok
test src/fields.rs - fields::Version (line 138) ... ok

test result: ok. 2 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.38s
```

a1ien, 2022-04-11:
yes, of course without `--all` we didn't run tests inside `test` dir.
Hmm... I am still don't understand why it's happening. 
With 1.0.24 `rusb` work correct. But with this commit https://github.com/libusb/libusb/commit/676076ce6cbed8331a22c33ca84b9cae326cff5b we see fail

a1ien, 2022-04-11:
Single init pass test.  And if wee sleep 100ms between iteration tests also pass.
So now I am not sure about something wrong with rusb

a1ien, 2022-04-11:
@mcuee It's possible that tests are building without optimizations? 
So if we build `stress` with optimization we can see crash on stress test?

tormodvolden, 2022-04-11:
libusb builds by default with -O2 (from configure), however the XCode build interestingly seems to use -Os AFAICS from the CI build.

mcuee, 2022-04-12:
I will reopen this ticket. I can test -O3 after work.

mcuee, 2022-04-12:
Maybe it is also interesting to see if other language bindings got similar problems or not.

mcuee, 2022-04-12:
Just tried under Windows 11 (Windows Terminal) and it seems to be okay.

```

C:\work\libusb\rusb [master ≡]> git submodule update --init
Submodule 'libusb' (https://github.com/libusb/libusb.git) registered for path 'libusb1-sys/libusb'
Cloning into 'C:/work/libusb/rusb/libusb1-sys/libusb'...
Submodule path 'libusb1-sys/libusb': checked out '4239bc3a50014b8e6a5a2a59df1fff3b7469543b'
C:\work\libusb\rusb [master ≡]> cargo test --workspace --verbose --all
       Fresh pkg-config v0.3.25
       Fresh cc v1.0.73
       Fresh vcpkg v0.2.15
       Fresh regex-syntax v0.6.25
   Compiling libusb1-sys v0.6.0 (C:\work\libusb\rusb\libusb1-sys)
       Fresh libc v0.2.122
       Fresh memchr v2.4.1
     Running `C:\work\libusb\rusb\target\debug\build\libusb1-sys-9c7d0abba62bdce1\build-script-build`
       Fresh aho-corasick v0.7.18
   Compiling regex v1.5.5
     Running `rustc --crate-name regex --edition=2018 C:\Users\xfchen\.cargo\registry\src\github.com-1ecc6299db9ec823\regex-1.5.5\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg "feature=\"aho-corasick\"" --cfg "feature=\"default\"" --cfg "feature=\"memchr\"" --cfg "feature=\"perf\"" --cfg "feature=\"perf-cache\"" --cfg "feature=\"perf-dfa\"" --cfg "feature=\"perf-inline\"" --cfg "feature=\"perf-literal\"" --cfg "feature=\"std\"" --cfg "feature=\"unicode\"" --cfg "feature=\"unicode-age\"" --cfg "feature=\"unicode-bool\"" --cfg "feature=\"unicode-case\"" --cfg "feature=\"unicode-gencat\"" --cfg "feature=\"unicode-perl\"" --cfg "feature=\"unicode-script\"" --cfg "feature=\"unicode-segment\"" -C metadata=74c30b94c1c5328d -C extra-filename=-74c30b94c1c5328d --out-dir C:\work\libusb\rusb\target\debug\deps -L dependency=C:\work\libusb\rusb\target\debug\deps --extern aho_corasick=C:\work\libusb\rusb\target\debug\deps\libaho_corasick-a6c08234c710d095.rmeta --extern memchr=C:\work\libusb\rusb\target\debug\deps\libmemchr-097ac1aaa105a9e5.rmeta --extern regex_syntax=C:\work\libusb\rusb\target\debug\deps\libregex_syntax-8b10c5ab22bdc4b0.rmeta --cap-lints allow`
   Compiling rusb v0.9.0 (C:\work\libusb\rusb)
     Running `rustc --crate-name libusb1_sys --edition=2018 libusb1-sys\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=bfbc308db5a4a83b -C extra-filename=-bfbc308db5a4a83b --out-dir C:\work\libusb\rusb\target\debug\deps -C incremental=C:\work\libusb\rusb\target\debug\incremental -L dependency=C:\work\libusb\rusb\target\debug\deps --extern libc=C:\work\libusb\rusb\target\debug\deps\liblibc-a0d4f40aedaf84f0.rmeta -L native=C:\work\libusb\rusb\target\debug\build\libusb1-sys-2b992af499e30d89\out -l dylib=user32 -l static=usb`
     Running `rustc --crate-name libusb1_sys --edition=2018 libusb1-sys\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --test -C metadata=d764183e7a9b5753 -C extra-filename=-d764183e7a9b5753 --out-dir C:\work\libusb\rusb\target\debug\deps -C incremental=C:\work\libusb\rusb\target\debug\incremental -L dependency=C:\work\libusb\rusb\target\debug\deps --extern libc=C:\work\libusb\rusb\target\debug\deps\liblibc-a0d4f40aedaf84f0.rlib -L native=C:\work\libusb\rusb\target\debug\build\libusb1-sys-2b992af499e30d89\out -l dylib=user32 -l static=usb`
     Running `C:\work\libusb\rusb\target\debug\build\rusb-cf3591f1f35347e6\build-script-build`
     Running `rustc --crate-name ffi_list_devices --edition=2018 libusb1-sys\examples\ffi_list_devices.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=c7f90bd9da316fc2 --out-dir C:\work\libusb\rusb\target\debug\examples -C incremental=C:\work\libusb\rusb\target\debug\incremental -L dependency=C:\work\libusb\rusb\target\debug\deps --extern libc=C:\work\libusb\rusb\target\debug\deps\liblibc-a0d4f40aedaf84f0.rlib --extern libusb1_sys=C:\work\libusb\rusb\target\debug\deps\liblibusb1_sys-bfbc308db5a4a83b.rlib -L native=C:\work\libusb\rusb\target\debug\build\libusb1-sys-2b992af499e30d89\out`
     Running `rustc --crate-name test --edition=2018 libusb1-sys\tests\test.rs --error-format=json --json=diagnostic-rendered-ansi --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --test -C metadata=3b9425f72ffcdc8f -C extra-filename=-3b9425f72ffcdc8f --out-dir C:\work\libusb\rusb\target\debug\deps -C incremental=C:\work\libusb\rusb\target\debug\incremental -L dependency=C:\work\libusb\rusb\target\debug\deps --extern libc=C:\work\libusb\rusb\target\debug\deps\liblibc-a0d4f40aedaf84f0.rlib --extern libusb1_sys=C:\work\libusb\rusb\target\debug\deps\liblibusb1_sys-bfbc308db5a4a83b.rlib -L native=C:\work\libusb\rusb\target\debug\build\libusb1-sys-2b992af499e30d89\out`
     Running `rustc --crate-name ffi_info --edition=2018 libusb1-sys\examples\ffi_info.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=c7f90bd9da316fc2 --out-dir C:\work\libusb\rusb\target\debug\examples -C incremental=C:\work\libusb\rusb\target\debug\incremental -L dependency=C:\work\libusb\rusb\target\debug\deps --extern libc=C:\work\libusb\rusb\target\debug\deps\liblibc-a0d4f40aedaf84f0.rlib --extern libusb1_sys=C:\work\libusb\rusb\target\debug\deps\liblibusb1_sys-bfbc308db5a4a83b.rlib -L native=C:\w [... cut ...] ebug\deps\liblibusb1_sys-bfbc308db5a4a83b.rlib --extern regex=C:\work\libusb\rusb\target\debug\deps\libregex-74c30b94c1c5328d.rlib --extern rusb=C:\work\libusb\rusb\target\debug\deps\librusb-6f464728bf19a9cd.rlib -C embed-bitcode=no --cfg libusb_hotplug_get_user_data`

running 2 tests
test src\fields.rs - fields::request_type (line 204) ... ok
test src\fields.rs - fields::Version (line 138) ... ok

test result: ok. 2 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.74s

```

a1ien, 2022-04-12:
Other OS haven't problems. In current master I add sleep to test for macOS https://github.com/a1ien/rusb/blob/master/libusb1-sys/tests/test.rs#L30-L31

a1ien, 2022-04-12:
Now  I more understand problem. I try to reproduce with simple rust application. But failed with this [variant](https://github.com/a1ien/libusb-stress-test/blob/3af51f3b75c1dceb2b42063cecfd647c04d847c6/src/main.rs) but after I and two threads a can reproduce [main.rs ](https://github.com/a1ien/libusb-stress-test/blob/eafc12792a3ba2b03a030f4e7311d66039e8c499/src/main.rs)

So I think if we run [this](https://github.com/libusb/libusb/blob/master/tests/stress.c#L29) function in multiple thread we also fail.
But problem only at macOS.

a1ien, 2022-04-12:
This C code also crash on macOS https://github.com/a1ien/libusb-stress-test/blob/ba45ba0b7220bff818dfceb97fcbafd9632ee9e2/main.c

mcuee, 2022-04-12:
Same issue here with the C program.

Without any external USB devices then there are no issues.
```

mcuee@mcuees-Mac-mini libusb-stress-test % ./libusb-stress-test 
Thread done
Thread done
All Done
```

With some USB devices, it will fail.
```

mcuee@mcuees-Mac-mini libusb-stress-test % ./libusb-stress-test
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file darwin_usb.c, line 605.
zsh: abort      ./libusb-stress-test
```

a1ien, 2022-04-12:
>BTW, stress.c with -O3 has no issues.

Bug appear only in multi-threads. Also it's can be reproduce if run multiply instance of stress(but I am not sure)

benzea, 2022-04-12:
I am thinking the refcount of the cached devices is getting confused somehow. Possibly, a device is kept alive, which causes it to remain in the list of devices for too long somehow (the code appears to be susceptible to the libusb device being kept alive from the outside even). Or, the reference counting is just getting out of sync.

To debug, I would suggest adding logging in
 * `darwin_ref_cached_device`
 * `darwin_deref_cached_device`
 * `darwin_cleanup_devices? 
so that one sees the reference count and whether it drops to 0 in `darwin_cleanup_devices` (after which the list should be empty really).

But, as said before, I am not sure the code is quite right. i.e. my hunch (didn't fully tests this) is that a device object might be kept alive longer than the context by user code. And this will then keep the device cached even after `libusb_exit`.

benzea, 2022-04-12:
> But, as said before, I am not sure the code is quite right. i.e. my hunch (didn't fully tests this) is that a device object might be kept alive longer than the context by user code. And this will then keep the device cached even after `libusb_exit`.

Said differently, doing:
 * `libusb_init`
 * `libusb_get_device_list`
 * `libusb_exit`
 * `libusb_init`

Should trigger the assertion (if the system has USB devices). i.e. the code leaks the device list retrieved using `libusb_get_device_list`.

hjelmn, 2022-04-12:
Will take a look shortly. My initialization code was changed to destroy the list on the last context exit. It is likely that introduced this bug.

benzea, 2022-04-12:
@hjelmn, there is also the fact that I added code to transfers to hold on to the cached device. So, I suspect that a late transfer free could also cause this corner case these days.

hjelmn, 2022-04-12:
That code is incorrect. You must destroy the list before exiting. You leaked the list and the backend asserted because of it. Now, we shouldn't assert-fail but that is an incorrect usage.

hjelmn, 2022-04-12:
Note also that ALL transfers MUST be freed before libusb_exit.

hjelmn, 2022-04-12:
Also note that the cached device cleanup was intended to be called by the library destructor function. When that changed it needed to be protected by the ```darwin_cached_devices_lock``` so that is an error that would likely cause issues with multithreaded programs.

hjelmn, 2022-04-12:
I made the mutual exclusion even stricter. Taking the cached device mutex in ```darwin_exit``` to ensure ordering between calls of ```darwin_exit``` and ```darwin_init```. The lock is held when decrementing the init count so that ```libusb_init``` blocks waiting for the lock before accessing the cached device list.

benzea, 2022-04-12:
> Note also that ALL transfers MUST be freed before libusb_exit. This has always been the case but probably not well documented.

So, we have at least some code users who free transfers with a delay. libgusb for example does so from an idle handler in a later mainloop iteration, as such, anything can happen in between (even `libusb_exit` theoretically, but that is an extreme case). In the case of libgusb, the code would become really awkward if it need to change the behaviour in this regard.

So, right now, I would assume that a `struct libusb_device` may be referenced after `libusb_exit`. However, a `struct libusb_device_handle` will be closed before `libusb_exit`.

hjelmn, 2022-04-12:
Ok, that is no allowed. All libusb object must be released before ```libusb_exit``` is called. pyusb had this issue as well.

hjelmn, 2022-04-13:
If you can test #1125 let me know. It fixes a very real bug in multithreaded support and changes out the assert for a log message and error code. It should make things easier to debug.

mcuee, 2022-04-13:
The C test program from https://github.com/a1ien/libusb-stress-test still failed after applying pull request #1125.
* #1125

```
mcuee@mcuees-Mac-mini libusb-stress-test % make
cc -g3 -Wall -Wextra  -std=c99 -D_GNU_SOURCE=1 -D _POSIX_C_SOURCE=199309L -I/opt/homebrew/Cellar/libusb/HEAD-221b092/include/libusb-1.0   -c -o main.o main.c
main.c:7:33: warning: unused parameter 'arg' [-Wunused-parameter]
void* test_init_and_exit(void * arg)
                                ^
main.c:30:9: warning: variable 'err' set but not used [-Wunused-but-set-variable]
    int err = pthread_create(&threadId1, NULL, &test_init_and_exit, NULL);
        ^
main.c:32:9: warning: variable 'err2' set but not used [-Wunused-but-set-variable]
    int err2 = pthread_create(&threadId2, NULL, &test_init_and_exit, NULL);
        ^
3 warnings generated.
cc  -o libusb-stress-test main.o -L/opt/homebrew/Cellar/libusb/HEAD-221b092/lib -lusb-1.0

mcuee@mcuees-Mac-mini libusb-stress-test % ./libusb-stress-test 
Failed to init libusb on iteration 131: -99Thread done
All Done
```

mcuee, 2022-04-13:
On the other hand, the rust test program seems to work fine (not so sure if I did it correctly or not).

```
mcuee@mcuees-Mac-mini libusb-stress-test % cargo test --workspace --verbose --all
    Updating crates.io index
  Downloaded libusb1-sys v0.6.1
  Downloaded libc v0.2.123
  Downloaded 2 crates (861.1 KB) in 2.00s
   Compiling libc v0.2.123
   Compiling pkg-config v0.3.25
   Compiling cc v1.0.73
     Running `rustc --crate-name build_script_build /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.123/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=82cb71419935f65e -C extra-filename=-82cb71419935f65e --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/build/libc-82cb71419935f65e -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name pkg_config /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/pkg-config-0.3.25/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 -C metadata=0ffeecf47fc222c3 -C extra-filename=-0ffeecf47fc222c3 --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name cc --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/cc-1.0.73/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 -C metadata=be4d0e6aa55d6ed6 -C extra-filename=-be4d0e6aa55d6ed6 --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --cap-lints allow`
     Running `/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/build/libc-82cb71419935f65e/build-script-build`
   Compiling libusb1-sys v0.6.1
     Running `rustc --crate-name build_script_build --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libusb1-sys-0.6.1/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 -C metadata=3eca560decbd2467 -C extra-filename=-3eca560decbd2467 --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/build/libusb1-sys-3eca560decbd2467 -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --extern cc=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps/libcc-be4d0e6aa55d6ed6.rlib --extern pkg_config=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps/libpkg_config-0ffeecf47fc222c3.rlib --cap-lints allow`
     Running `/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/build/libusb1-sys-3eca560decbd2467/build-script-build`
     Running `rustc --crate-name libc /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.123/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=b58f6f353f6bed39 -C extra-filename=-b58f6f353f6bed39 --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --cap-lints allow --cfg freebsd11 --cfg libc_priv_mod_use --cfg libc_union --cfg libc_const_size_of --cfg libc_align --cfg libc_int128 --cfg libc_core_cvoid --cfg libc_packedN --cfg libc_cfg_target_vendor --cfg libc_non_exhaustive --cfg libc_ptr_addr_of --cfg libc_underscore_const_names`
     Running `rustc --crate-name libusb1_sys --edition=2018 /Users/mcuee/.cargo/registry/src/github.com-1ecc6299db9ec823/libusb1-sys-0.6.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 -C metadata=e1b94379cd7bfb75 -C extra-filename=-e1b94379cd7bfb75 --out-dir /Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps -L dependency=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps --extern libc=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps/liblibc-b58f6f353f6bed39.rmeta --cap-lints allow -L native=/opt/homebrew/Cellar/libusb/HEAD-221b092/lib -l usb-1.0`
   Compiling libusb-stress-test v0.1.0 (/Users/mcuee/build/libusb/libusb_test/libusb-stress-test)
     Running `rustc --crate-name libusb_stress_test --edition=2021 src/main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --emit=dep-info,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --test -C metadata=a16cd43549db0c27 -C extra-filename=-a16cd43549db0c27 --out-dir /Users/mcuee/build/libusb/libusb_ [... cut ...] /deps --extern libusb1_sys=/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps/liblibusb1_sys-e1b94379cd7bfb75.rlib -L native=/opt/homebrew/Cellar/libusb/HEAD-221b092/lib`
    Finished test [unoptimized + debuginfo] target(s) in 7.02s
     Running `/Users/mcuee/build/libusb/libusb_test/libusb-stress-test/target/debug/deps/libusb_stress_test-a16cd43549db0c27`

running 0 tests

test result: ok. 0 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s

```

a1ien, 2022-04-13:
In rust I have add sleep between init and exit.

mcuee, 2022-04-13:
I think I did it wrongly, the rust code stillf failed.

```
                               
mcuee@mcuees-Mac-mini libusb-stress-test % cargo run libusb-stress-test
    Finished dev [unoptimized + debuginfo] target(s) in 0.00s
     Running `target/debug/libusb-stress-test libusb-stress-test`
thread '<unnamed>' panicked at 'Failed to init libusb on iteration 122: -99', src/main.rs:10:24
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Any { .. }', src/main.rs:31:16
```

hjelmn, 2022-04-13:
Also need #1126. The exit ordering is wrong.

hjelmn, 2022-04-13:
#1126 is merged. With that the assumption made by the backend (that all devices are cleaned up by the end of the last call to ```darwin_exit```-- as long as the user does not have any references) is correct.

hjelmn, 2022-04-13:
Ok, I think #1125 is good to go. Thanks for bringing this to our attention. We have insufficient testing of multithreaded init/finalize so it went far too long.

mcuee, 2022-04-14:
> Ok, I think #1125 is good to go. Thanks for bringing this to our attention. We have insufficient testing of multithreaded init/finalize so it went far too long.

Shall we change stress test  to multithreaded test?

mcuee, 2022-04-14:
Yes #1125 is good now.

```

mcuee@mcuees-Mac-mini libusb-stress-test % make
cc -g3 -Wall -Wextra  -std=c99 -D_GNU_SOURCE=1 -D _POSIX_C_SOURCE=199309L -I/opt/homebrew/Cellar/libusb/HEAD-602197d/include/libusb-1.0   -c -o main.o main.c
main.c:7:33: warning: unused parameter 'arg' [-Wunused-parameter]
void* test_init_and_exit(void * arg)
                                ^
main.c:30:9: warning: variable 'err' set but not used [-Wunused-but-set-variable]
    int err = pthread_create(&threadId1, NULL, &test_init_and_exit, NULL);
        ^
main.c:32:9: warning: variable 'err2' set but not used [-Wunused-but-set-variable]
    int err2 = pthread_create(&threadId2, NULL, &test_init_and_exit, NULL);
        ^
3 warnings generated.
cc  -o libusb-stress-test main.o -L/opt/homebrew/Cellar/libusb/HEAD-602197d/lib -lusb-1.0
mcuee@mcuees-Mac-mini libusb-stress-test % ./libusb-stress-test 
Thread done
Thread done
All Done
```

tormodvolden, 2022-04-14:
Thanks a lot for the quick fixing up @hjelmn. I have also tested #1125 (on macOS 11.6) and can confirm it fixes the crash here. I used the attached multithreaded test, I tested up to 64 threads but I guess much more than the number of cores doesn't stress it further (and only 2 are needed for this issue to manifest). I don't know if we should replace the old test or keep both. The old tests runs 10000 times, i don't think this high number helps for detecting these issues, but it could make sense for detecting memory leaks if memory usage is monitored (which typically is not done for automatic testing though).

[stress-mt.c.txt](https://github.com/libusb/libusb/files/8489458/stress-mt.c.txt)

commit 4e2de881 (2022-04-14): darwin: add missing locking around cached device list cleanup

The cleanup function darwin_cleanup_devices was intented to be called only once
at program exit. When the initialization/finalization code was changed to
destroy the cached device list on last exit this function should have been
modified to require a lock on darwin_cached_devices_lock. This commit updates
cleanup to protect the cached device list with the cached devices mutex and
updates darwin_init to print out an error and return if a reference leak is
detected.

Also using this opportunity to correct the naming of the mutex. Cha [... cut]
closed 2022-04-14 by hjelmn