libusb/libusb #1128: Enhance libusb stress test

mcuee, 2022-04-15:
From https://github.com/libusb/libusb/issues/1124#issuecomment-1099188287 by Tormod Volden.
> I used the attached multithreaded test, I tested up to 64 threads but I guess much more than the number of cores doesn't stress it further (and only 2 are needed for this issue to manifest). I don't know if we should replace the old test or keep both. The old tests runs 10000 times, i don't think this high number helps for detecting these issues, but it could make sense for detecting memory leaks if memory usage is monitored (which typically is not done for automatic testing though).

https://github.com/libusb/libusb/files/8489458/stress-mt.c.txt

mcuee, 2022-04-15:
Sample run log under macOS (Mac Mini M1).
```

mcuee@mcuees-Mac-mini stress_mt % ./stress-mt 
Thread 0 started
Thread 3 started
Thread 1 started
Thread 4 started
Thread 2 started
Thread 5 started
Thread 6 started
Thread 7 started
Thread 0 done
Thread 1 done
Thread 2 done
Thread 7 done
Thread 6 done
Thread 4 done
Thread 3 done
Thread 5 done
All Done
```

mcuee, 2022-04-16:
Note: as a general test code, then it needs to be cross-platform.

a1ien, 2022-04-16:
oh if we can use C11 we can try [thrd_create](https://en.cppreference.com/w/c/thread/thrd_create)

mcuee, 2022-04-17:
> oh if we can use C11 we can try [thrd_create](https://en.cppreference.com/w/c/thread/thrd_create)

I believe it is not available in VS 2019/2022.

mcuee, 2022-04-17:
Maybe we can use similar workaround for Windows like this libusb dfp example.
https://github.com/libusb/libusb/blob/master/examples/dpfp.c

Or probably just use Windows specific codes for Windows.
https://docs.microsoft.com/en-us/cpp/parallel/multithreading-with-c-and-win32?view=msvc-160

tormodvolden, 2023-01-24:
We can keep this open until someone adds Windows support to stress-mt.

mcuee, 2023-02-11:
@tormodvolden 

Actually it seems to work under MSYS2 mingw. There are some warnings though.

```
MINGW64 /c/work/libusb/libusb/tests
$ gcc -I /mingw64/include/libusb-1.0 stress_mt.c -pthread -o stress_mt -lusb-1.0
stress_mt.c: In function 'test_init_and_exit':
stress_mt.c:12:29: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   12 |         long int threadno = (long int) arg;
      |                             ^
stress_mt.c: In function 'main':
stress_mt.c:37:73: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   37 |                 pthread_create(&threadId[t], NULL, &test_init_and_exit, (void *) t);
      |                                                                         ^

$ ./stress_mt.exe
Starting multithreaded init and exit test...
Thread 0 started
Thread 2 started
Thread 3 started
Thread 1 started
Thread 4 started
Thread 5 started
Thread 6 started
Thread 7 started
Thread 7 done
Thread 2 done
Thread 3 done
Thread 1 done
Thread 0 done
Thread 4 done
Thread 6 done
Thread 5 done
All Done

```

mcuee, 2023-02-11:
The other possibility is to use `std:thread` which should work under both MSY2 MinGW64 and Visual Studio.
https://en.cppreference.com/w/cpp/thread/thread

tormodvolden, 2023-02-11:
That's great. I must figure out why it is not built when I cross-build. The warnings shouldn't be to difficult to fix.

mcuee, 2023-02-11:
> That's great. I must figure out why it is not built when I cross-build. The warnings shouldn't be to difficult to fix.
> 
> EDIT: Now I get it, you build it manually and not via tests/Makefile.am

That should be easy to fix, just remove this line.
https://github.com/libusb/libusb/blob/master/tests/Makefile.am#L9

```
if PLATFORM_POSIX 
stress_mt_SOURCES = stress_mt.c
```

mcuee, 2023-02-11:
Quick test shows that it works.

```
PS C:\work\libusb\libusb> git diff
diff --git a/tests/Makefile.am b/tests/Makefile.am
index d47d4ed7..82c34776 100644
--- a/tests/Makefile.am
+++ b/tests/Makefile.am
@@ -6,11 +6,11 @@ stress_SOURCES = stress.c libusb_testlib.h testlib.c

 noinst_PROGRAMS = stress

-if PLATFORM_POSIX
+# if PLATFORM_POSIX
 stress_mt_SOURCES = stress_mt.c

 noinst_PROGRAMS += stress_mt
-endif
+# endif

 if BUILD_UMOCKDEV_TEST
 # NOTE: We add libumockdev-preload.so so that we can run tests in-process

MINGW64 /c/work/libusb/libusb
$ ./autogen.sh
...

$ make
...
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb/tests'
  CC       stress.o
  CCLD     stress.exe
  CC       stress_mt.o
stress_mt.c: In function 'test_init_and_exit':
stress_mt.c:12:29: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   12 |         long int threadno = (long int) arg;
      |                             ^
stress_mt.c: In function 'main':
stress_mt.c:37:73: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   37 |                 pthread_create(&threadId[t], NULL, &test_init_and_exit, (void *) t);
      |                                                                         ^
  CCLD     stress_mt.exe
make[2]: Leaving directory '/c/work/libusb/libusb/tests'
make[2]: Entering directory '/c/work/libusb/libusb'
make[2]: Leaving directory '/c/work/libusb/libusb'
make[1]: Leaving directory '/c/work/libusb/libusb'

$ ./tests/stress_mt.exe
Starting multithreaded init and exit test...
Thread 0 started
Thread 1 started
Thread 2 started
Thread 3 started
Thread 4 started
Thread 5 started
Thread 6 started
Thread 7 started
Thread 7 done
Thread 6 done
Thread 2 done
Thread 0 done
Thread 3 done
Thread 5 done
Thread 4 done
Thread 1 done
All Done
```

tormodvolden, 2023-02-11:
Well, we have to guard it somehow from platforms without pthread. But maybe solve that with ifdef HAVE_PTHREAD around the whole program.

The warnings are silenced with this:

```
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index c59a5c7..f08ca81 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -9,7 +9,7 @@
 
 static void *test_init_and_exit(void * arg)
 {
-       long int threadno = (long int) arg;
+       long int threadno = (long int)(intptr_t) arg;
 
        printf("Thread %ld started\n", threadno);
        for (int i = 0; i < ITERS; ++i) {
@@ -34,7 +34,7 @@ int main(void)
 
        printf("Starting multithreaded init and exit test...\n");
        for(t = 0; t < NTHREADS; t++)
-               pthread_create(&threadId[t], NULL, &test_init_and_exit, (void *) t);
+               pthread_create(&threadId[t], NULL, &test_init_and_exit, (void *)(intptr_t) t);
 
        for(t = 0; t < NTHREADS; t++)
                pthread_join(threadId[t], NULL);

```

mcuee, 2023-02-25:
@tormodvolden 

I just asked ChatGPT to conver the code to C++ 11 threading.

Here is what it comes out.

```
#include <iostream>
#include <chrono>
#include <thread>
#include <vector>
#include <mutex>
#include <condition_variable>

#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    libusb_context *ctx;
    int rc;
};

std::mutex mtx;
std::condition_variable cv;
bool ready = false;

void wait_for_ready() {
    std::unique_lock<std::mutex> lock(mtx);
    while (!ready) {
        cv.wait(lock);
    }
}

void signal_ready() {
    {
        std::unique_lock<std::mutex> lock(mtx);
        ready = true;
    }
    cv.notify_all();
}

void *thread_func(void *arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_device **devs;
        ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
        if (cnt < 0) {
            data->rc = (int)cnt;
            libusb_free_device_list(devs, 1);
            break;
        }

        libusb_free_device_list(devs, 1);
    }

    return NULL;
}

int main() {
    libusb_context *ctx;
    int rc = libusb_init(&ctx);
    if (rc < 0) {
        std::cerr << "Failed to initialize libusb" << std::endl;
        return 1;
    }

    std::vector<std::thread> threads;

    struct thread_data data[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].ctx = ctx;
        data[i].rc = 0;
        threads.emplace_back(thread_func, &data[i]);
    }

    signal_ready();

    for (auto& thread : threads) {
        thread.join();
    }

    for (int i = 0; i < NUM_THREADS; i++) {
        if (data[i].rc < 0) {
            std::cerr << "Thread " << i << " failed with error code " << data[i].rc << std::endl;
            return 1;
        }
    }

    std::cout << "All threads completed successfully" << std::endl;

    libusb_exit(ctx);

    return 0;
}
```

- We use `std::thread` instead of `pthread_t` to create threads.
- We use `std::mutex` and `std::condition_variable` to synchronize the threads instead of `pthread_mutex_t` and `pthread_cond_t`.
- The `wait_for_ready()` and `signal_ready()` functions are used to synchronize the threads before they start executing their tasks.
- Instead of passing a `void*` pointer to the thread function, we define a struct that contains the data needed by the thread, and pass a pointer to this struct to the thread function.
- We use range-based for loop to iterate over the threads vector instead of a for loop with an index.
- We use `std::cout` and `std::cerr` instead of `printf()`.
- We use `std::chrono` to measure the time taken by the program. However, the original program did not have any timing code, so we did not add any timing code to the converted program.
-

mcuee, 2023-02-25:
@tormodvolden 

I do not know much about C++ 11 but the above codes do not seem to work.

```
$ g++ -o stress_mt_cpp11 -I /mingw64/include/libusb-1.0 stress_mt_cpp11.cpp -lusb-1.0
stress_mt_cpp11.cpp: In function 'int main()':
stress_mt_cpp11.cpp:60:25: warning: 'int libusb_init(libusb_context**)' is deprecated: Use libusb_init_context instead [-Wdeprecated-declarations]
   60 |     int rc = libusb_init(&ctx);
      |              ~~~~~~~~~~~^~~~~~
In file included from stress_mt_cpp11.cpp:8:
C:/msys64/mingw64/include/libusb-1.0/libusb.h:1568:17: note: declared here
 1568 | int LIBUSB_CALL libusb_init(libusb_context **ctx);
      |                 ^~~~~~~~~~~

$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1275
```

mcuee, 2023-02-25:
gdb output.

```
$ gdb ./stress_mt_cpp11.exe
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./stress_mt_cpp11.exe...
(gdb) run
Starting program: C:\work\libusb\libusb_test\stress_mt_cpp11\stress_mt_cpp11.exe
[New Thread 17772.0x528]
[New Thread 17772.0x22fc]
[New Thread 17772.0x3190]
warning:
warning: AllK required contiguous memory = 675016 (64bit)
warning:   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
warning:   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
warning:   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
warning:   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
warning:   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
warning:   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
warning:   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
warning:   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
warning:   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)
warning:
warning: Dynamically allocated as needed:
warning:        KLST_DEVINFO = 2596 bytes each
[New Thread 17772.0x1538]
[New Thread 17772.0x14c8]
[New Thread 17772.0x1380]
[New Thread 17772.0x2a9c]
[New Thread 17772.0x12f0]
[New Thread 17772.0x3f9c]
[New Thread 17772.0x429c]
[New Thread 17772.0x4560]
[New Thread 17772.0xe50]
[New Thread 17772.0x126c]
[New Thread 17772.0x447c]
[New Thread 17772.0x2850]
[New Thread 17772.0x2118]
[New Thread 17772.0x4530]
[New Thread 17772.0x41d8]
[New Thread 17772.0x35fc]
[New Thread 17772.0x40c0]

Thread 19 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 17772.0x41d8]
0x00007ff90bbd1e80 in strcmp () from C:\WINDOWS\System32\msvcrt.dll
(gdb) bt
#0  0x00007ff90bbd1e80 in strcmp () from C:\WINDOWS\System32\msvcrt.dll
#1  0x00007ff88c8939cb in winusb_get_device_list (ctx=<optimized out>, _discdevs=<optimized out>) at os/windows_winusb.c:1787
#2  0x00007ff88c8821e5 in libusb_get_device_list (ctx=0xd82b80, list=0x440fd00) at core.c:853
#3  0x00007ff60e711614 in thread_func(void*) ()
#4  0x00007ff60e714756 in void* std::__invoke_impl<void*, void* (*)(void*), thread_data*>(std::__invoke_other, void* (*&&)(void*), thread_data*&&) ()
#5  0x00007ff60e714cac in std::__invoke_result<void* (*)(void*), thread_data*>::type std::__invoke<void* (*)(void*), thread_data*>(void* (*&&)(void*), thread_data*&&) ()
#6  0x00007ff60e713ea5 in void* std::thread::_Invoker<std::tuple<void* (*)(void*), thread_data*> >::_M_invoke<0ull, 1ull>(std::_Index_tuple<0ull, 1ull>) ()
#7  0x00007ff60e713ec8 in std::thread::_Invoker<std::tuple<void* (*)(void*), thread_data*> >::operator()() ()
#8  0x00007ff60e713d0c in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void* (*)(void*), thread_data*> > >::_M_run() ()
#9  0x00007ff88d13bf9f in ?? () from C:\msys64\mingw64\bin\libstdc++-6.dll
#10 0x00007ff8d7944d0b in ?? () from C:\msys64\mingw64\bin\libwinpthread-1.dll
#11 0x00007ff90bbae634 in msvcrt!_beginthreadex () from C:\WINDOWS\System32\msvcrt.dll
#12 0x00007ff90bbae70c in msvcrt!_endthreadex () from C:\WINDOWS\System32\msvcrt.dll
#13 0x00007ff90c3326bd in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#14 0x00007ff90d5cdfb8 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#15 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) quit
A debugging session is active.

        Inferior 1 [process 17772] will be killed.

Quit anyway? (y or n) y

```

mcuee, 2023-02-25:
@tormodvolden 

FYI, no issues under Linux.

```
mcuee@UbuntuSwift3:~/build/libusb_test/stress_mt_cpp11$ g++ -std=c++11 -o stress_mt_cpp11 -I/usr/local/include/libusb-1.0   stress_mt_cpp11_chatgpt.cpp -lusb-1.0 -lpthread
stress_mt_cpp11_chatgpt.cpp: In function ‘int main()’:
stress_mt_cpp11_chatgpt.cpp:60:30: warning: ‘int libusb_init(libusb_context**)’ is deprecated: Use libusb_init_context instead [-Wdeprecated-declarations]
   60 |     int rc = libusb_init(&ctx);
      |                              ^
In file included from stress_mt_cpp11_chatgpt.cpp:8:
/usr/local/include/libusb-1.0/libusb.h:1568:17: note: declared here
 1568 | int LIBUSB_CALL libusb_init(libusb_context **ctx);
      |                 ^~~~~~~~~~~
stress_mt_cpp11_chatgpt.cpp:60:30: warning: ‘int libusb_init(libusb_context**)’ is deprecated: Use libusb_init_context instead [-Wdeprecated-declarations]
   60 |     int rc = libusb_init(&ctx);
      |                              ^
In file included from stress_mt_cpp11_chatgpt.cpp:8:
/usr/local/include/libusb-1.0/libusb.h:1568:17: note: declared here
 1568 | int LIBUSB_CALL libusb_init(libusb_context **ctx);
      |                 ^~~~~~~~~~~
mcuee@UbuntuSwift3:~/build/libusb_test/stress_mt_cpp11$ ./stress_mt_cpp11 
All threads completed successfully
```

mcuee, 2023-02-25:
Some times it does not cause segemtation fault but rather assetion failure.

```
$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1275

MINGW64 /c/work/libusb/libusb_test/stress_mt_cpp11
$ gdb ./stress_mt_cpp11.exe
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./stress_mt_cpp11.exe...
(gdb) run
Starting program: C:\work\libusb\libusb_test\stress_mt_cpp11\stress_mt_cpp11.exe
[New Thread 3028.0x1180]
[New Thread 3028.0x2f3c]
[New Thread 3028.0x2500]
warning:
warning: AllK required contiguous memory = 675016 (64bit)
warning:   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
warning:   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
warning:   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
warning:   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
warning:   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
warning:   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
warning:   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
warning:   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
warning:   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)
warning:
warning: Dynamically allocated as needed:
warning:        KLST_DEVINFO = 2596 bytes each
[New Thread 3028.0x2984]
[New Thread 3028.0x21a0]
[New Thread 3028.0x2918]
[New Thread 3028.0x3a0]
[New Thread 3028.0x268c]
[New Thread 3028.0x2c78]
[New Thread 3028.0x1b5c]
[New Thread 3028.0xed4]
[New Thread 3028.0x231c]
[New Thread 3028.0x1b8c]
[New Thread 3028.0x366c]
[New Thread 3028.0x314c]
[New Thread 3028.0x23bc]
[New Thread 3028.0x17e4]
[New Thread 3028.0x2198]
[New Thread 3028.0x2284]
[New Thread 3028.0x1820]

Thread 9 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 3028.0x268c]
0x00007fff90a21e80 in strcmp () from C:\WINDOWS\System32\msvcrt.dll
(gdb) bt
#0  0x00007fff90a21e80 in strcmp () from C:\WINDOWS\System32\msvcrt.dll
#1  0x00007fff37e139cb in winusb_get_device_list (ctx=<optimized out>, _discdevs=<optimized out>) at os/windows_winusb.c:1787
#2  0x00007fff37e021e5 in libusb_get_device_list (ctx=0x622b80, list=0x307fd00) at core.c:853
#3  0x00007ff75ede1614 in thread_func(void*) ()
#4  0x00007ff75ede4756 in void* std::__invoke_impl<void*, void* (*)(void*), thread_data*>(std::__invoke_other, void* (*&&)(void*), thread_data*&&) ()
[... 50 lines cut ...]
Assertion failed: refcnt >= 2, file core.c, line 1275
[Thread 10796.0x1e9c exited with code 3]
[Thread 10796.0x3688 exited with code 3]
[Thread 10796.0x9e4 exited with code 3]
[Thread 10796.0x2690 exited with code 3]
[Thread 10796.0x2088 exited with code 3]
[Thread 10796.0x9b4 exited with code 3]
[Thread 10796.0x16a0 exited with code 3]
[Thread 10796.0x2c54 exited with code 3]
[Thread 10796.0x544 exited with code 3]
[Thread 10796.0x20f4 exited with code 3]
[Thread 10796.0x15e8 exited with code 3]
[Thread 10796.0x29fc exited with code 3]
[Thread 10796.0x13f4 exited with code 3]
[Thread 10796.0x19e4 exited with code 3]
[Thread 10796.0x1094 exited with code 3]
[Thread 10796.0x160c exited with code 3]
[Thread 10796.0x1a90 exited with code 3]
[Thread 10796.0x15b8 exited with code 3]
[Thread 10796.0xb54 exited with code 3]
[Thread 10796.0x78c exited with code 3]
[Inferior 1 (process 10796) exited with code 03]
(gdb) quit

```

mcuee, 2023-02-25:
No issue with the existing stress_mt.c code, even if I increase the number of threads and iteration to match the ChatGPT version.

```
PS C:\work\libusb\libusb> git diff
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index c59a5c7a..7ae558a1 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -4,8 +4,8 @@

 /* Test that creates and destroys contexts repeatedly */

-#define NTHREADS 8
-#define ITERS 64
+#define NTHREADS 16
+#define ITERS 10000

 static void *test_init_and_exit(void * arg)
 {

$ ./tests/stress_mt.exe
Starting multithreaded init and exit test...
Thread 0 started
Thread 2 started
Thread 1 started
Thread 3 started
Thread 4 started
Thread 5 started
Thread 6 started
Thread 7 started
Thread 8 started
Thread 9 started
Thread 10 started
Thread 11 started
Thread 12 started
Thread 13 started
Thread 14 started
Thread 15 started
Thread 1 done
Thread 2 done
Thread 6 done
Thread 13 done
Thread 9 done
Thread 14 done
Thread 10 done
Thread 5 done
Thread 7 done
Thread 0 done
Thread 3 done
Thread 12 done
Thread 11 done
Thread 15 done
Thread 8 done
Thread 4 done
All Done

```

mcuee, 2023-02-25:
Using `-g` in the compiling stage somehow makes the segfault less often and the assertion fault more often.

```
$ g++ -std=c++11 -g -o stress_mt_cpp11 -I /mingw64/include/libusb-1.0 stress_mt_cpp11.cpp -lusb-1.0
stress_mt_cpp11.cpp: In function 'int main()':
stress_mt_cpp11.cpp:60:25: warning: 'int libusb_init(libusb_context**)' is deprecated: Use libusb_init_context instead [-Wdeprecated-declarations]
   60 |     int rc = libusb_init(&ctx);
      |              ~~~~~~~~~~~^~~~~~
In file included from stress_mt_cpp11.cpp:8:
C:/msys64/mingw64/include/libusb-1.0/libusb.h:1568:17: note: declared here
 1568 | int LIBUSB_CALL libusb_init(libusb_context **ctx);
      |                 ^~~~~~~~~~~

$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1275

xiaof@LAPTOP-913SIMUD MINGW64 /c/work/libusb/libusb_test/stress_mt_cpp11
$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1275

$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1275

$ ./stress_mt_cpp11.exe
Segmentation fault

$ gdb ./stress_mt_cpp11.exe
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./stress_mt_cpp11.exe...
(gdb) run
Starting program: C:\work\libusb\libusb_test\stress_mt_cpp11\stress_mt_cpp11.exe
[New Thread 13680.0x298c]
[New Thread 13680.0x1fec]
[New Thread 13680.0x1a48]
warning:
warning: AllK required contiguous memory = 675016 (64bit)
warning:   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
warning:   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
warning:   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
warning:   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
warning:   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
warning:   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
warning:   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
warning:   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
warning:   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)
warning:
warning: Dynamically allocated as needed:
warning:        KLST_DEVINFO = 2596 bytes each
[New Thread 13680.0x2a20]
[New Thread 13680.0x3ec8]
[New Thread 13680.0x27b8]
[New Thread 13680.0x32f0]
[New Thread 13680.0x26a4]
[New Thread 13680.0x3a44]
[New Thread 13680.0x540]
[New Thread 13680.0x2674]
[... 85 lines cut ...]
#7  0x00007ff606693ec8 in std::thread::_Invoker<std::tuple<void* (*)(void*), thread_data*> >::operator() (this=0xd29548)
    at C:/msys64/mingw64/include/c++/12.2.0/bits/std_thread.h:259
#8  0x00007ff606693d0c in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void* (*)(void*), thread_data*> > >::_M_run (this=0xd29540)
    at C:/msys64/mingw64/include/c++/12.2.0/bits/std_thread.h:210
#9  0x00007fff19a6bf9f in ?? () from C:\msys64\mingw64\bin\libstdc++-6.dll
#10 0x00007fff8af54d0b in ?? () from C:\msys64\mingw64\bin\libwinpthread-1.dll
#11 0x00007fff909fe634 in msvcrt!_beginthreadex () from C:\WINDOWS\System32\msvcrt.dll
#12 0x00007fff909fe70c in msvcrt!_endthreadex () from C:\WINDOWS\System32\msvcrt.dll
#13 0x00007fff8f0c26bd in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#14 0x00007fff911edfb8 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#15 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) quit
A debugging session is active.

        Inferior 1 [process 5100] will be killed.

Quit anyway? (y or n) y

```

mcuee, 2023-02-25:
The following is generated by ChatGPT for C11 threading, but it will not work under Windows as MinGW and Visual Studio do not seem to support `<threads>`

```
#include <stdio.h>
#include <stdlib.h>
#include <threads.h>
#include <time.h>
#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    libusb_context *ctx;
    int rc;
};

mtx_t mtx;
cnd_t cv;
int ready = 0;

void wait_for_ready() {
    mtx_lock(&mtx);
    while (!ready) {
        cnd_wait(&cv, &mtx);
    }
    mtx_unlock(&mtx);
}

void signal_ready() {
    mtx_lock(&mtx);
    ready = 1;
    cnd_broadcast(&cv);
    mtx_unlock(&mtx);
}

int thread_func(void *arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_device **devs;
        ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
        if (cnt < 0) {
            data->rc = (int)cnt;
            libusb_free_device_list(devs, 1);
            break;
        }

        libusb_free_device_list(devs, 1);
    }

    return 0;
}

int main() {
    libusb_context *ctx;
    int rc = libusb_init(&ctx);
    if (rc < 0) {
        fprintf(stderr, "Failed to initialize libusb\n");
        return 1;
    }

    thrd_t threads[NUM_THREADS];
    struct thread_data data[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].ctx = ctx;
        data[i].rc = 0;
        thrd_create(&threads[i], thread_func, &data[i]);
    }

    signal_ready();

    for (int i = 0; i < NUM_THREADS; i++) {
        int ret;
        thrd_join(threads[i], &ret);
        if (ret != 0) {
            fprintf(stderr, "Thread %d failed with error code %d\n", i, ret);
            return 1;
        }
    }

    printf("All threads completed successfully\n");

    libusb_exit(ctx);

    return 0;
}

```

So I asked it to generate another code with Windows native threading.

```
#include <stdio.h>
#include <stdlib.h>
#include <Windows.h>
#include <time.h>
#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    libusb_context *ctx;
    int rc;
};

CRITICAL_SECTION cs;
CONDITION_VARIABLE cv;
BOOL ready = FALSE;

void wait_for_ready() {
    EnterCriticalSection(&cs);
    while (!ready) {
        SleepConditionVariableCS(&cv, &cs, INFINITE);
    }
    LeaveCriticalSection(&cs);
}

void signal_ready() {
    EnterCriticalSection(&cs);
    ready = TRUE;
    WakeAllConditionVariable(&cv);
    LeaveCriticalSection(&cs);
}

DWORD WINAPI thread_func(LPVOID arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_device **devs;
        ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
        if (cnt < 0) {
            data->rc = (int)cnt;
            libusb_free_device_list(devs, 1);
            break;
        }

        libusb_free_device_list(devs, 1);
    }

    return 0;
}

int main() {
    libusb_context *ctx;
    int rc = libusb_init(&ctx);
    if (rc < 0) {
        fprintf(stderr, "Failed to initialize libusb\n");
        return 1;
    }

    HANDLE threads[NUM_THREADS];
    struct thread_data data[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].ctx = ctx;
        data[i].rc = 0;
        threads[i] = CreateThread(NULL, 0, thread_func, &data[i], 0, NULL);
    }

    signal_ready();

    WaitForMultipleObjects(NUM_THREADS, threads, TRUE, INFINITE);

    for (int i = 0; i < NUM_THREADS; i++) {
        DWORD ret;
        GetExitCodeThread(threads[i], &ret);
        if (ret != 0) {
            fprintf(stderr, "Thread %d failed with error code %d\n", i, ret);
            return 1;
        }
    }

    printf("All threads completed successfully\n");

    libusb_exit(ctx);

    return 0;
}

```

mcuee, 2023-02-25:
The above code failed very fast. Maybe the code is not correct.

```
$ gcc -std=c11 -o stress_mt_winthread -I /mingw64/include/libusb-1.0 stress_mt_winthread.c -lusb-1.0
stress_mt_winthread.c: In function 'main':
stress_mt_winthread.c:57:5: warning: 'libusb_init' is deprecated: Use libusb_init_context instead [-Wdeprecated-declarations]
   57 |     int rc = libusb_init(&ctx);
      |     ^~~
In file included from stress_mt_winthread.c:5:
C:/msys64/mingw64/include/libusb-1.0/libusb.h:1568:17: note: declared here
 1568 | int LIBUSB_CALL libusb_init(libusb_context **ctx);
      |                 ^~~~~~~~~~~

$ ./stress_mt_winthread.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001169] [00002564] libusb: debug [libusb_init_context] libusb v1.0.26.11784
[ 0.001483] [00002564] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000058 events 0
[ 0.001756] [00002564] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.002033] [00002564] libusb: debug [usbi_add_event_source] add HANDLE 000000000000005c events 0
[ 0.002337] [00002564] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002453] [00002564] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003998] [00002564] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005768] [00002564] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007597] [00002564] libusb: info [windows_init] UsbDk backend is not available
[ 0.007786] [00002564] libusb: debug [libusb_init_context] installing new context as implicit default
[ 0.007968] [00000ba8] libusb: debug [windows_iocp_thread] I/O completion thread started
Segmentation fault

$ gdb ./stress_mt_winthread.exe
GNU gdb (GDB) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./stress_mt_winthread.exe...
(gdb) run
Starting program: C:\work\libusb\libusb_test\stress_mt_c11\stress_mt_winthread.exe
[New Thread 6176.0x1e44]
[New Thread 6176.0x3a7c]
[New Thread 6176.0x3db4]
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001156] [00001cbc] libusb: debug [libusb_init_context] libusb v1.0.26.11784
[ 0.001457] [00001cbc] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000bc events 0
[ 0.001681] [00001cbc] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001943] [00001cbc] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000c0 events 0
[ 0.002030] [00001cbc] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002210] [00001cbc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.065722] [00001cbc] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.163101] [00001cbc] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
warning:
warning: AllK required contiguous memory = 675016 (64bit)
warning:   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
warning:   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
warning:   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
warning:   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
warning:   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
warning:   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
warning:   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
[... 23 lines cut ...]

Thread 6 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 6176.0x1ccc]
0x00007fff911c419a in ntdll!TpWorkOnBehalfClearTicket () from C:\WINDOWS\SYSTEM32\ntdll.dll
(gdb) bt
#0  0x00007fff911c419a in ntdll!TpWorkOnBehalfClearTicket () from C:\WINDOWS\SYSTEM32\ntdll.dll
#1  0x00007fff911b1f64 in ntdll!RtlEnterCriticalSection () from C:\WINDOWS\SYSTEM32\ntdll.dll
#2  0x00007fff911b1d52 in ntdll!RtlEnterCriticalSection () from C:\WINDOWS\SYSTEM32\ntdll.dll
#3  0x00007ff742bc1502 in wait_for_ready ()
#4  0x00007ff742bc15b6 in thread_func ()
#5  0x00007fff8f0c26bd in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#6  0x00007fff911edfb8 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#7  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) quit
A debugging session is active.

        Inferior 1 [process 6176] will be killed.

Quit anyway? (y or n) y
```

mcuee, 2023-02-25:
ChatGPT suggested to add a mutex but that does not help.

```
#include <iostream>
#include <chrono>
#include <thread>
#include <vector>
#include <mutex>
#include <condition_variable>

#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    libusb_context *ctx;
    int rc;
    std::mutex *mtx;
};

std::mutex mtx;
std::condition_variable cv;
bool ready = false;

void wait_for_ready() {
    std::unique_lock<std::mutex> lock(mtx);
    while (!ready) {
        cv.wait(lock);
    }
}

void signal_ready() {
    {
        std::unique_lock<std::mutex> lock(mtx);
        ready = true;
    }
    cv.notify_all();
}

void *thread_func(void *arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_device **devs;
        ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
        if (cnt < 0) {
            std::unique_lock<std::mutex> lock(*data->mtx);
            data->rc = (int)cnt;
            libusb_free_device_list(devs, 1);
            break;
        }

        libusb_free_device_list(devs, 1);
    }

    return NULL;
}

int main() {
    libusb_context *ctx;
    int rc = libusb_init(&ctx);
    if (rc < 0) {
        std::cerr << "Failed to initialize libusb" << std::endl;
        return 1;
    }

    std::vector<std::thread> threads;

    struct thread_data data[NUM_THREADS];
    std::mutex thread_mtx[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].ctx = ctx;
        data[i].rc = 0;
        data[i].mtx = &thread_mtx[i];
        threads.emplace_back(thread_func, &data[i]);
    }

    signal_ready();

    for (auto& thread : threads) {
        thread.join();
    }

    for (int i = 0; i < NUM_THREADS; i++) {
        if (data[i].rc < 0) {
            std::cerr << "Thread " << i << " failed with error code " << data[i].rc << std::endl;
            return 1;
        }
    }

    std::cout << "All threads completed successfully" << std::endl;

    libusb_exit(ctx);

    return 0;
}

```

tormodvolden, 2023-02-25:
> I just asked ChatGPT to conver the code to C++ 11 threading. How do you like the idea?

Ha ha, the future of programming is here :)

I think this autogenerated code sometimes have bugs that are even more tricky to track down than the bugs that humans create.

Note also that some projects refuse to include such code due to an unclear copyright situation.

But seriously, for the libusb tests I think we should stick to C for the basic tests. It could make sense to have tests for other languages as well, but it would be a lot more work to maintain. If you look at examples/dpfp.c you can see that porting posix threads to native Windows threads is not so difficult. Want to give that a try? You could even grab the thread_create() and thread_join() abstractions from there.

Youw, 2023-02-25:
```cpp
void signal_ready() {
    mtx_lock(&mtx);
    ready = 1;
    cnd_broadcast(&cv);
    mtx_unlock(&mtx);
}
```
Btw - this is a typical (performance) error many developer make: mutex has to be unlocked _before_ notifying a conditional.

mcuee, 2023-02-25:
> Ha ha, the future of programming is here :)
> 
> I think this autogenerated code sometimes have bugs that are even more tricky to track down than the bugs that humans create.

Good point. That is probably the danger of using ChatGPT for real production codes.

> Note also that some projects refuse to include such code due to an unclear copyright situation.

I agree with that. My intention is to use ChatGPT to generate test codes for libusb (not to be merged) to find potential bugs. For example, if the generated codes work under Linux and macOS but not Windows, potentially the Windows codes may have an issue.

mcuee, 2023-02-25:
> But seriously, for the libusb tests I think we should stick to C for the basic tests. It could make sense to have tests for other languages as well, but it would be a lot more work to maintain. If [...]

Haha, my C language skill is still quite lacking. I will see what I can do here. I am actually very scare of Windows native threading stuff.

mcuee, 2023-02-25:
> > I just asked ChatGPT to conver the code to C++ 11 threading. How do you like the idea?
> 
> Ha ha, the future of programming is here :)

@tormodvolden
BTW, ChatGPT does not seem to do a good job at converting the following libusb based codes to C++11 threading. It failed badly after many tries.
https://github.com/libusb/hidapi/blob/master/libusb/hid.c

@Youw thinks it should not be difficult at all.
https://github.com/libusb/hidapi/issues/319#issuecomment-1444973552

mcuee, 2023-02-25:
> FYI, no issues under Linux.
> 
> ```
> mcuee@UbuntuSwift3:~/build/libusb_test/stress_mt_cpp11$ ./stress_mt_cpp11 
> All threads completed successfully
> ```

No issues under macOS as well.

```

mcuee@mcuees-Mac-mini stress_mt_cpp11 % g++ -std=c++11 -g -I /opt/homebrew/include/libusb-1.0  stress_mt_cpp11.cpp -L /opt/homebrew/lib  -o stress_mt_cpp11 -lusb-1.0
stress_mt_cpp11.cpp:60:14: warning: 'libusb_init' is deprecated [-Wdeprecated-declarations]
    int rc = libusb_init(&ctx);
             ^
/opt/homebrew/include/libusb-1.0/libusb.h:1567:1: note: 'libusb_init' has been explicitly marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_init_context)
^
/opt/homebrew/include/libusb-1.0/libusb.h:76:50: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__ ((deprecated))
                                                 ^
1 warning generated.

mcuee@mcuees-Mac-mini stress_mt_cpp11 % ./stress_mt_cpp11         
All threads completed successfully

```

mcuee, 2023-02-25:
> The following is generated by ChatGPT for C11 threading, but it will not work under Windows as MinGW and Visual Studio do not seem to support `<threads.h>`

@Youw 
Apple clang does not support it either.
```
mcuee@mcuees-Mac-mini stress_mt_c11 % gcc -std=c11 -g -I /opt/homebrew/include/libusb-1.0  stress_mt_c11.c -L /opt/homebrew/lib  -o stress_mt_c11 -lusb-1.0 -pthreads
stress_mt_c11.c:3:10: fatal error: 'threads.h' file not found
#include <threads.h>
         ^~~~~~~~~~~
1 error generated.
```

mcuee, 2023-02-26:
Take note the above C++ 11 threading code does not match the intention of `stress_mt.c` which is to test libusb by creating and destroying contexts repeatedly.

So chatgpt changes the code to the following.

```
#include <iostream>
#include <chrono>
#include <thread>
#include <vector>
#include <mutex>
#include <condition_variable>

#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    int rc;
};

std::mutex mtx;
std::condition_variable cv;
bool ready = false;

void wait_for_ready() {
    std::unique_lock<std::mutex> lock(mtx);
    while (!ready) {
        cv.wait(lock);
    }
}

void signal_ready() {
    {
        std::unique_lock<std::mutex> lock(mtx);
        ready = true;
    }
    cv.notify_all();
}

void *thread_func(void *arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_context *ctx;
        int rc = libusb_init(&ctx);
        if (rc < 0) {
            data->rc = rc;
            break;
        }
        libusb_exit(ctx);
    }

    return NULL;
}

int main() {
    std::vector<std::thread> threads;

    struct thread_data data[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].rc = 0;
        threads.emplace_back(thread_func, &data[i]);
    }

    signal_ready();

    for (auto& thread : threads) {
        thread.join();
    }

    for (int i = 0; i < NUM_THREADS; i++) {
        if (data[i].rc < 0) {
            std::cerr << "Thread " << i << " failed with error code " << data[i].rc << std::endl;
            return 1;
        }
    }

    std::cout << "All threads completed successfully" << std::endl;

    return 0;
}

```

Now there is no issue to run under Windows, which matches the result of `stress_mt.c`.

```
$ ./stress_mt_cpp11_context.exe
All threads completed successfully
```

mcuee, 2023-02-26:
@tormodvolden 

That being said, I tend to think we may have a ref count issue with the Windows backend. I tend to believe the code generated by ChatGPT earlier is still correct, even though it does not match `stress_mt.c`.

```
#include <iostream>
#include <chrono>
#include <thread>
#include <vector>
#include <mutex>
#include <condition_variable>

#include <libusb.h>

#define NUM_THREADS 16
#define NUM_ITERATIONS 10000

struct thread_data {
    int thread_id;
    libusb_context *ctx;
    int rc;
};

std::mutex mtx;
std::condition_variable cv;
bool ready = false;

void wait_for_ready() {
    std::unique_lock<std::mutex> lock(mtx);
    while (!ready) {
        cv.wait(lock);
    }
}

void signal_ready() {
    {
        std::unique_lock<std::mutex> lock(mtx);
        ready = true;
    }
    cv.notify_all();
}

void *thread_func(void *arg) {
    struct thread_data *data = (struct thread_data*)arg;

    wait_for_ready();

    for (int i = 0; i < NUM_ITERATIONS; i++) {
        libusb_device **devs;
        ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
        if (cnt < 0) {
            data->rc = (int)cnt;
            libusb_free_device_list(devs, 1);
            break;
        }

        libusb_free_device_list(devs, 1);
    }

    return NULL;
}

int main() {
    libusb_context *ctx;
    int rc = libusb_init(&ctx);
    if (rc < 0) {
        std::cerr << "Failed to initialize libusb" << std::endl;
        return 1;
    }

    std::vector<std::thread> threads;

    struct thread_data data[NUM_THREADS];
    for (int i = 0; i < NUM_THREADS; i++) {
        data[i].thread_id = i;
        data[i].ctx = ctx;
        data[i].rc = 0;
        threads.emplace_back(thread_func, &data[i]);
    }

    signal_ready();

    for (auto& thread : threads) {
        thread.join();
    }

    for (int i = 0; i < NUM_THREADS; i++) {
        if (data[i].rc < 0) {
            std::cerr << "Thread " << i << " failed with error code " << data[i].rc << std::endl;
            return 1;
        }
    }

    std::cout << "All threads completed successfully" << std::endl;

    libusb_exit(ctx);

    return 0;
}

```

mcuee, 2023-02-26:
> That being said, I tend to think we may have a ref count issue with the Windows backend. I tend to believe the code generated by ChatGPT earlier is still correct, even though it does not match the intention of `stress_mt.c`.

I am not sure now. No issues with the following quick hack of `stress_mt.c`. Haha, I do not need the help from ChatGPT here. 

```
PS C:\work\libusb\libusb_pr1246> git diff
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index 55c11635..c6ca9803 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -8,8 +8,8 @@

 /* Test that creates and destroys contexts repeatedly */

-#define NTHREADS 8
-#define ITERS 64
+#define NTHREADS 16
+#define ITERS 10000

 static void *test_init_and_exit(void * arg)
 {
@@ -18,11 +18,17 @@ static void *test_init_and_exit(void * arg)
        printf("Thread %ld started\n", threadno);
        for (int i = 0; i < ITERS; ++i) {
                libusb_context *ctx = NULL;
+               libusb_device **devs;
                int r;

                r = libusb_init_context(&ctx, /*options=*/NULL, /*num_options=*/0);
                if (r != LIBUSB_SUCCESS) {
-                       printf("Failed to init libusb on iteration %d: %d", i, r);
+                       printf("Failed to init libusb on iteration %d: %d \n", i, r);
+                       return NULL;
+               }
+               r = libusb_get_device_list(ctx, &devs);
+               if (r != 7) {
+                       printf("Failed to get device list on iteration %d: %d \n", i, r);
                        return NULL;
                }
                libusb_exit(ctx);

PS C:\work\libusb\libusb_pr1246> .\examples\testlibusb.exe
Dev (bus 1, device 5): 8087 - 0026 speed: 12M
Dev (bus 1, device 8): 0403 - 6001 speed: 12M
  Manufacturer:              FTDI
  Product:                   US232R
Dev (bus 1, device 4): 04F2 - B6DD speed: 480M
Dev (bus 1, device 2): 1C7A - 0575 speed: 480M
Dev (bus 1, device 0): 8086 - A0ED speed: 5G
Dev (bus 1, device 1): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 0): 8086 - 9A13 speed: 5G
PS C:\work\libusb\libusb_pr1246> .\tests\stress_mt.exe
Starting multithreaded init and exit test...
Thread 0 started
Thread 1 started
Thread 2 started
Thread 3 started
Thread 4 started
Thread 5 started
Thread 6 started
Thread 7 started
Thread 5 done
Thread 1 done
Thread 3 done
Thread 7 done
Thread 2 done
Thread 4 done
Thread 6 done
Thread 0 done
All Done

```

mcuee, 2023-02-26:
> I think this autogenerated code sometimes have bugs that are even more tricky to track down than the bugs that humans create.

@tormodvolden 

I think you are absolute right here.

I believe I have found the bug of the ChatGPT code since ` libusb_free_device_list(devs, 1);` may not be thread safe. This may cause either the ref cnt assertion fault or segmentation fault.

Please help to check if my root cause analysis and the fix are correct or not. Thanks.

```
$ ./stress_mt_cpp11.exe
Segmentation fault

$ ./stress_mt_cpp11_1.exe
All threads completed successfully

$ diff -u stress_mt_cpp11.cpp stress_mt_cpp11_1.cpp
--- stress_mt_cpp11.cpp 2023-02-26 18:35:02.624901400 +0800
+++ stress_mt_cpp11_1.cpp       2023-02-26 18:34:00.070289400 +0800
@@ -45,11 +45,11 @@
         ssize_t cnt = libusb_get_device_list(data->ctx, &devs);
         if (cnt < 0) {
             data->rc = (int)cnt;
-            libusb_free_device_list(devs, 1);
+            libusb_free_device_list(devs, 0);
             break;
         }

-        libusb_free_device_list(devs, 1);  // I believe this line is the problem
+        libusb_free_device_list(devs, 0);
     }

     return NULL;
```

tormodvolden, 2023-02-26:
I think it is safe to say that winusb_get_device_list() is not thread-safe. I don't know if the solution is a mutex around this monster enumeration loop.

A brute-force trial:
```
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -850,7 +850,9 @@ ssize_t API_EXPORTED libusb_get_device_list(libusb_context *ctx,
                usbi_mutex_unlock(&ctx->usb_devs_lock);
        } else {
                /* backend does not provide hotplug support */
+               usbi_mutex_lock(&ctx->usb_devs_lock);
                r = usbi_backend.get_device_list(ctx, &discdevs);
+               usbi_mutex_unlock(&ctx->usb_devs_lock);
        }
 
        if (r < 0) {

```

mcuee, 2023-02-26:
> I think it is safe to say that winusb_get_device_list() is not thread-safe. I don't know if the solution is a mutex around this monster enumeration loop.
> 
> A brute-force trial:

So looks like the root cause analysis is correct but my fix is not correct.

Do you think it is important to make `libusb_get_device_list()` thread safe? 

I have tried your patch but it does not seem to work.

```
PS C:\work\libusb\libusb_pr1246> git diff
diff --git a/libusb/core.c b/libusb/core.c
index e3746217..b7673613 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -850,7 +850,9 @@ ssize_t API_EXPORTED libusb_get_device_list(libusb_context *ctx,
                usbi_mutex_unlock(&ctx->usb_devs_lock);
        } else {
                /* backend does not provide hotplug support */
+               usbi_mutex_lock(&ctx->usb_devs_lock);
                r = usbi_backend.get_device_list(ctx, &discdevs);
+               usbi_mutex_unlock(&ctx->usb_devs_lock);
        }

        if (r < 0) {

$ ./stress_mt_cpp11.exe
Assertion failed: refcnt >= 2, file core.c, line 1277

```

mcuee, 2023-02-26:
@tormodvolden 

I think we should improve `stress_mt.c` further by including `libusb_get_device_list` and `libusb_free_device_list`.

Ref:
* https://github.com/libusb/libusb/issues/1124#issuecomment-1096717824

mcuee, 2023-02-27:
> I think it is safe to say that winusb_get_device_list() is not thread-safe.

@tormodvolden 

Just wondering how you come to this conclusion. As long as there is no hotplug, I think the result should still be the same.

I was thinking `libusb_free_device_list` with the ref count decrease was not thread-safe but after looking at the codes I am not sure now. At least we are using atomic increase/decrease for ref count changes.

mcuee, 2023-02-27:
> I think it is safe to say that winusb_get_device_list() is not thread-safe. I don't know if the solution is a mutex around this monster enumeration loop.

@tormodvolden

We need to confirm this is a real problem for Windows.

libusb documentation says that **"libusb is a thread-safe library, but extra considerations must be applied to applications which interact with libusb from multiple threads"** and **"libusb is designed to be completely thread-safe, but as with any API it cannot prevent a user from sabotaging themselves, either intentionally or otherwise"**.

I tend to think that the user will probably assume that `libusbget_device_list()` is thread-safe without much-needed considerations, as the following documnetations do not mention anything special about this API.

Ref:
https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html
https://libusb.sourceforge.io/api-1.0/libusb_caveats.html

tormodvolden, 2023-02-27:
Maybe you are right in that as long as all the ref counting is atomic there shouldn't be any problem with winusb_get_device_list().

mcuee, 2023-02-28:
> Maybe you are right in that as long as all the ref counting is atomic there shouldn't be any problem with winusb_get_device_list(). I've been thinking there could be something between usbi_get_device_by_session_id() and usbi_alloc_device(), if two threads from the same context step on each other. But if they are from different contexts it should be fine. In our current stress_mt test we create a new context per thread.

@tormodvolden 

Hmm, that can still pose a problem for some applications.

As we know, we do not have hotplug support for Windows. People are using the trick of repeat calling `libusb_get_device_list` which will in turn call `winusb_get_device_list`. I do not think they are using different context.

mcuee, 2023-02-28:
>  I've been thinking there could be something between usbi_get_device_by_session_id() and usbi_alloc_device(), if two threads from the same context step on each other. 

@tormodvolden 

This is a quick hack of `stress_mt.c` from PR #1252 to test `libusb_get_device_list` with single context (the default context). It seems to work fine. Still I am not so sure if the test is good enough to prove `winusb_get_device_list` is thread-safe.

```
PS C:\work\libusb\libusb_pr1252_singlecontext> git diff
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index 07a9a8b2..ca9a3608 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -89,10 +89,9 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
        struct thread_info *ti = (struct thread_info *) arg;

        for (int i = 0; i < ITERS; ++i) {
-               libusb_context *ctx = NULL;
                int r;

-               r = libusb_init_context(&ctx, /*options=*/NULL, /*num_options=*/0);
+               r = libusb_init_context(NULL, /*options=*/NULL, /*num_options=*/0);
                if (r != LIBUSB_SUCCESS) {
                        printf("Failed to init libusb on iteration %d: %d", i, r);
                        ti->err = r;
@@ -100,7 +99,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                }
                if (ti->enumerate) {
                        libusb_device **devs;
-                       ssize_t cnt = libusb_get_device_list(ctx, &devs);
+                       ssize_t cnt = libusb_get_device_list(NULL, &devs);
                        if (cnt < 0) {
                                libusb_free_device_list(devs, 1);
                                break;
@@ -109,7 +108,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                        libusb_free_device_list(devs, 1);
                }

-               libusb_exit(ctx);
+               libusb_exit(NULL);
        }
        return (thread_return_t) THREAD_RETURN_VALUE;
 }

PS C:\work\libusb\libusb_pr1252_singlecontext> .\build\v143\x64\Release\stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 discovered 9 devices
Thread 1 discovered 9 devices
Thread 2 discovered 9 devices
Thread 3 discovered 9 devices
Thread 4 discovered 9 devices
Thread 5 discovered 9 devices
Thread 6 discovered 9 devices
Thread 7 discovered 9 devices
All done, 0 errors
```

mcuee, 2023-02-28:
> I think this autogenerated code sometimes have bugs that are even more tricky to track down than the bugs that humans create.

I guess we can forget about the ChatGPT based C++11 test code for now, since PR #1252 works fine.

mcuee, 2023-02-28:
> > I've been thinking there could be something between usbi_get_device_by_session_id() and usbi_alloc_device(), if two threads from the same context step on each other.
> 
> Edit to add: **Now I can see segfault or ref count assetion after some more tests. You are probably right in the end.**

With the brute-force patch, it still has the same problem.

```
PS C:\work\libusb\libusb_pr1252_singlecontext> git diff
diff --git a/libusb/core.c b/libusb/core.c
index e3746217..b7673613 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -850,7 +850,9 @@ ssize_t API_EXPORTED libusb_get_device_list(libusb_context *ctx,
                usbi_mutex_unlock(&ctx->usb_devs_lock);
        } else {
                /* backend does not provide hotplug support */
+               usbi_mutex_lock(&ctx->usb_devs_lock);
                r = usbi_backend.get_device_list(ctx, &discdevs);
+               usbi_mutex_unlock(&ctx->usb_devs_lock);
        }

        if (r < 0) {
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index 07a9a8b2..ca9a3608 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -89,10 +89,9 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
        struct thread_info *ti = (struct thread_info *) arg;

        for (int i = 0; i < ITERS; ++i) {
-               libusb_context *ctx = NULL;
                int r;

-               r = libusb_init_context(&ctx, /*options=*/NULL, /*num_options=*/0);
+               r = libusb_init_context(NULL, /*options=*/NULL, /*num_options=*/0);
                if (r != LIBUSB_SUCCESS) {
                        printf("Failed to init libusb on iteration %d: %d", i, r);
                        ti->err = r;
@@ -100,7 +99,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                }
                if (ti->enumerate) {
                        libusb_device **devs;
-                       ssize_t cnt = libusb_get_device_list(ctx, &devs);
+                       ssize_t cnt = libusb_get_device_list(NULL, &devs);
                        if (cnt < 0) {
                                libusb_free_device_list(devs, 1);
                                break;
@@ -109,7 +108,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                        libusb_free_device_list(devs, 1);
                }

-               libusb_exit(ctx);
+               libusb_exit(NULL);
        }
        return (thread_return_t) THREAD_RETURN_VALUE;
 }
```

tormodvolden, 2023-02-28:
> As we know, we do not have hotplug support for Windows. People are using the trick of repeat calling `libusb_get_device_list` which will in turn call `winusb_get_device_list`. I do not think they are using different context.

Then the mentioned disclaimer in the docs apply, as well as common sense: They should make sure that all device discovery is running in the same thread. We shouldn't need to explicitly warn about this.

mcuee, 2023-02-28:
> libusb documentation says that **"libusb is a thread-safe library, but extra considerations must be applied to applications which interact with libusb from multiple threads"** and **"libusb is [...]

Yes I have seen these two. However, it does not mention much about device discovery.
https://libusb.sourceforge.io/api-1.0/libusb_caveats.html
https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html

And the single context test does seem to pass under Linux.

tormodvolden, 2023-02-28:
The assert(refcnt >= 0) in libusb_unref_device() makes sense because the device should be destroyed (somewhere) if refcnt hits zero, and dev should be NULL. The problem can arise if it is unref'ed twice without "somewhere" happening in between.

mcuee, 2023-02-28:
> The assert(refcnt >= 0) in libusb_unref_device() makes sense because the device should be destroyed (somewhere) if refcnt hits zero, and dev should be NULL. The problem can arise if it is unref'ed twice without "somewhere" happening in between.

Yes this makes sense.

What about the following?
```
> assert(refcnt >= 2);
```

tormodvolden, 2023-02-28:
> And the single context test does seem to pass under Linux. Do you think it will fail as well if I increase the number of iterations?

I was thinking this is because of the locks. Not sure why the brute-force lock doesn't help.

In my experience more iterations wouldn't help much, since the issues are seen on the first iterations. BTW, this would also be seen easier in stress_mt if we improve it to display the iteration it failed.

tormodvolden, 2023-02-28:
assert(refcnt >= 2) in libusb_ref_device(): If the device is accessed, it must already have been ref'ed during enumeration (otherwise it cannot be assumed to be available). Then after incrementing, it must be at least 2.

mcuee, 2023-03-01:
> > They should make sure that all device discovery is running in the same thread. We shouldn't need to explicitly warn about this.
> 
> This seems to be a good one to mention in the FAQ.

Wiki updated to reflect this one.
**Please take note all device discovery should be running in the same thread.**

https://github.com/libusb/libusb/wiki/FAQ#user-content-What_are_the_extra_considerations_to_be_applied_to_applications_which_interact_with_libusb_from_multiple_threads

Youw, 2023-03-01:
> Please take note all device discovery should be running in the same thread.

Within the same context or across multiple contexts? (I'd assume the furst).

tormodvolden, 2023-03-01:
Within same context. Also, the device discovery must not necessarily run in one thread. But if it runs in several threads, the programmer should make sure (through some synchronization mechanism) that only one is running at a time. So a typical sane setup would be to have it in one thread.

Youw, 2023-03-01:
As I suspected.
I'd suggest a bit different wording:

Please take note that a device discovery is not thread-safe within a single context. It should be performed either in the same thread or serialized using an external (user-code) syncronisation mechanism.

mcuee, 2023-03-01:
I updated the wiki entry.

_It is recommended that all device discovery should be running in the same thread. If the device discovery runs in several threads, the application should make sure that only one is running at a time through some synchronization mechanism._

mcuee, 2023-03-01:
@Youw and @tormodvolden 

Feel free to update the wiki entry to make it more accurate. Thanks.

mcuee, 2023-03-01:
> Within same context. Also, the device discovery must not necessarily run in one thread. But if it runs in several threads, the programmer should make sure (through some synchronization mechanism) that only one is running at a time. So a typical sane setup would be to have it in one thread.
> 
> And this is only important for non-hotplug platforms. But that includes all portable applications.

What will be the rule for hotplug platforms (Linux and macOS)?

tormodvolden, 2023-03-01:
For hot-plug there shouldn't be any issue since there is the mutex around get_device_list() - although I don't see why adding a similar mutex for Windows didn't help there. Also get_device_list() doesn't perform much "discovery" in this case, it just returns the list that has been maintained by hot-plug events.

mcuee, 2023-03-02:
> For hot-plug there shouldn't be any issue since there is the mutex around get_device_list() - although I don't see why adding a similar mutex for Windows didn't help there. Also get_device_list() [...]

I will see if I can debug the Windows further to see why the single context `stress_mt` does not work even with the mutex added.

mcuee, 2023-03-02:
@tormodvolden and @Youw 

I have updated the Wiki.
<B>
For non-hotplug platforms (like Windows), it is recommended that all device discovery should be running in the same thread. If the device discovery runs in several threads, the application should make sure that only one is running at a time through some synchronization mechanism, because device discovery may not be thread-safe within a single context for these platforms.
</B>

This is not an issue with hotplug capable platforms (Linux and macOS).

mcuee, 2023-03-02:
> @tormodvolden and @Youw
> 
> I have updated the Wiki. Please review the wording to see if it is good now. Thanks. For non-hotplug platforms (like Windows), it is recommended that all device discovery should be running in the same thread. If the device discovery runs in several threads, the application should make sure that only one is running at a time through some synchronization mechanism, because device discovery may not be thread-safe within a single context for these platforms.
> 
> This is not an issue with hotplug capable platforms (Linux and macOS).

@tormodvolden 
What will be an example to deal with the issue for Windows from the user application side, since changing `core.c` to add mutex does not help? 

Example, what kind of synchronization mechanism should be used if I want to get the single default context version of `stress_mt` to work under Windows?

```
$ git diff
diff --git a/libusb/core.c b/libusb/core.c
index e3746217..b7673613 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -850,7 +850,9 @@ ssize_t API_EXPORTED libusb_get_device_list(libusb_context *ctx,
                usbi_mutex_unlock(&ctx->usb_devs_lock);
        } else {
                /* backend does not provide hotplug support */
+               usbi_mutex_lock(&ctx->usb_devs_lock);
                r = usbi_backend.get_device_list(ctx, &discdevs);
+               usbi_mutex_unlock(&ctx->usb_devs_lock);
        }

        if (r < 0) {
diff --git a/tests/stress_mt.c b/tests/stress_mt.c
index 5b89194d..fc19750e 100644
--- a/tests/stress_mt.c
+++ b/tests/stress_mt.c
@@ -89,10 +89,10 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
        struct thread_info *ti = (struct thread_info *) arg;

        for (int i = 0; i < ITERS; ++i) {
-               libusb_context *ctx = NULL;
+       //      libusb_context *ctx = NULL;
                int r;

-               r = libusb_init_context(&ctx, /*options=*/NULL, /*num_options=*/0);
+               r = libusb_init_context(NULL, /*options=*/NULL, /*num_options=*/0);
                if (r != LIBUSB_SUCCESS) {
                        printf("Failed to init libusb on iteration %d: %d", i, r);
                        ti->err = r;
@@ -100,7 +100,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                }
                if (ti->enumerate) {
                        libusb_device **devs;
-                       ti->devcount = libusb_get_device_list(ctx, &devs);
+                       ti->devcount = libusb_get_device_list(NULL, &devs);
                        if (ti->devcount < 0) {
                                libusb_free_device_list(devs, 1);
                                break;
@@ -108,7 +108,7 @@ static thread_return_t THREAD_CALL_TYPE init_and_exit(void * arg)
                        libusb_free_device_list(devs, 1);
                }

-               libusb_exit(ctx);
+               libusb_exit(NULL);
        }
        return (thread_return_t) THREAD_RETURN_VALUE;
 }

$ ./tests/stress_mt.exe
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Assertion failed: refcnt >= 2, file core.c, line 1277

```

[7 more replies]

commit ab612960 (2023-03-12): tests/stress_mt: Add Windows threads support

Build the test on all platforms.

Closes #1128
closed 2023-03-12 by tormodvolden