libusb/libusb #1130: Pckaging and nomenclature with release 1.0.26

hamlibdk, 2022-04-18:
Hi,

The way that the v1.0.26 binary release (at **https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26-binaries.7z**) has been packaged causes some issues with software development now as nomenclature and package release structures have changed.

(a) Can the "Include" directory as delivered in the past containing the header file please be packaged with the "binaries"? It is NOT supplied with the **VS2015-xx** versions .

(b) Can this release (and future releases) please be repackaged in accordance with past nomenclature structures i.e. 

C:.
├───examples
│   ├───bin32
│   ├───bin64
│   └───source
├───include
│   └───libusb-1.0
├───MinGW32
│   ├───dll
│   └───static
├───MinGW64
│   ├───dll
│   └───static
├───VS2015
│   ├───MS32
│   │   ├───dll
│   │   └───static
│   └───MS64
│       ├───dll
│       └───static
(etc)

(c) Can the DLL's please be named consistently i.e. the MinGW version as **libusb-1.0.dll** rather than **msys-usb-1.0.dll** ?    

All of this makes life so much easier for developers and SDK maintainers.

Regards,

Hamlib SDK Developers

tormodvolden, 2022-04-19:
Hi, I will try to follow this if I make the binaries the next time. Wrapping these up for 1.0.25 and 1.0.26 I wasn't aware there was a defined structure that somebody depends upon, and I did it by hand. I don't know how it was generated previously, if somebody has a build script for instance. Nobody around these days seems to know.

Since I am fishing out these files from a build, but not knowing what they all can be used for, I wonder:
Which files do you need under static/ ? Do you need anything else than the .dll under dll/ ? Can we skip examples/source/ ?
I guess it doesn't matter which compiler is used for the examples?

a) There were actually include directories included, under other targets (but it surely makes sense to have one at the top instead since they are identical).

c) Yes, this ends up as msys-usb-1.0.dll in the build but we can rename it.

tormodvolden, 2022-04-19:
Please also see #1116. Maybe you can help making the CI build lay out the wanted structure also.

mcuee, 2022-04-19:
No idea how the older 7z binary were built. I believe last time Pete was using a build script, using Windows WDK and then his customized  mutli-arch MinGW-w64 toolchain for MinGW.
Ref: https://github.com/libusb/libusb/blob/master/.private/bm.sh
Ref: https://github.com/libusb/libusb/commit/9ee81db53e4367bbb90650ec6443f81429454971

But later since WDK is kind of outdated so that things changed a bit when Chris took over.
https://github.com/libusb/libusb/commit/28630ad489bfc81ee683ddf569e9d9e0f4da95f7

From what I see, in the past, for VS, DLL directory will include the DLL, the import lib file and the pdb file (useful for debugging). The stactic director just includes the static lib.

For MinGW, usually the DLL directory will include the DLL file and the import lib file. Then the static lib should just include the static lib file.

hamlibdk, 2022-04-20:
All that is being asked for is compatibility with past nomenclature - if not a fix in this instance with this release - as everyone wants to be able to use the same versions of Libraries and Software on MS-Windows as is available for Linux-based systems.

If the fix is not easy for this release then perhaps an early 1.0.27 release may be in order?

We should just be able to drop in a MinGW, CygWIN or VS-built DLL from the package - and it should not matter what "environment" was used to build this or how it was built for the MS-Windows systems.

What is supplied as of Version 1.0.26 this cannot and will not occur without adding considerable layers of complexity that may impact development (and scripts) at later developmental stages.

The ask here is to continue and respect the past delivery structure for The Version 1.0.x - stream especially for those of us suffering with making software portable across WIndows and Linux environments. 

If change is necessary pleae introduce it with Version 2 - and pre-warn of structural changes so that us Developers and kt maintainers can work with change.

Just so the team is aware of what we do and maintain here are references to what we do:

- https://sourceforge.net/projects/hamlib-sdk/ <== this is what we do and what is impacted.
- https://sourceforge.net/projects/hamlib/ <== Libraries that use LibUSB extensively
- https://physics.princeton.edu/pulsar/k1jt/wsjtx.html <== This is what its for: so we can build this software written by a Nobel Laureate (Joe Taylor of "Pulsar" fame) and used by over 30,000 people at any one time.

Keep up the great work. Again our regrets at having to ask for this direction refocus.

Thanks and Regards,

Hamlib SDK Developers
https://hamlib-sdk.sourceforge.io/

mcuee, 2022-04-21:
@tormodvolden 
BTW, last time we decided that we would not package for Cygwin. For Windows libusb-1.0.dll is the common name for the DLL with the exception of Cygwin.

And then we also kind decided not to support MinGW.org unless there are non-intrusive patches coming from contributors (but some activities are on-going in the new web site: https://osdn.net/projects/mingw/releases/) as most of the people move to MinGW-w64 based 32/64bit toolchain.

mcuee, 2022-04-21:
> We should just be able to drop in a MinGW, CygWIN or VS-built DLL from the package - and it should not matter what "environment" was used to build this or how it was built for the MS-Windows systems.

The idea is to have compatibility between the MinGW and VS build DLL. Cygwin DLL will not be compatible.

hamlibdk, 2022-04-22:
> Cygwin DLL will not be compatible.

That is not always the case. As long as the DLL has access to additional libraries etc. then there is no conceivable reason why it should not work as a drop-in. That is the whole point of DLL's and Dynamic Libraries. They are resolved at run time.

MS compiler-built components is always preferred when working with Windows.... but ...

Yet life is not that simple when writing code that will compile across paltforms and environments. That is where MinGW excels (and of course the fact that the Qt environment also works with MinGW). In that case it is "good karma" to use the MinGW-built DLL if possible.

I reiterate that any DLL compiled under any environment - including CygWIN - should just work as a drop-in (ideally). Yes its a little more complex with CygWIN as one needs also to supply other dependencies to make things work.... and this can become quite cumbersome and messy.

Back with 1.0.24 we had to use the VS2019 version with our Hamlib-compilation work (as the preferred version) as for some reason the supplied MinGW DLL just would not work !

Ending comment here as I think that the point has well and truly been made and comment here is in very general terms and open to too much argument :-)

closed 2022-04-22 by hamlibdk

mcuee, 2022-04-22:
Thanks. But just remeber we do not package Cygwin in the past and probably will not package Cygwin binaries in the future as well.