libusb/libusb #1131: [Windows] Open device failed with access error.

FuChengYiXiao, 2022-04-19:
I want to open a usb composite device(uisng winusb driver) which has 2 interfaces.  When another device use one interface, I get an LIBUSB_ERROR_ACCESS error while calling libusb_open.
I found that winusbx_open funtion will return error immediately when open device failed and not try all interface paths.
I make a demo just like this  and it`s ok.
`	int ret = ERROR_FILE_NOT_FOUND;
	// WinUSB requires a separate handle for each interface
	for (i = 0; i < USB_MAXINTERFACES; i++) {
		if ((priv->usb_interface[i].path != NULL)
				&& (priv->usb_interface[i].apib->id == USB_API_WINUSBX)) {
			file_handle = windows_open(dev_handle, priv->usb_interface[i].path, GENERIC_READ | GENERIC_WRITE);
			if (file_handle == INVALID_HANDLE_VALUE) {
				usbi_err(HANDLE_CTX(dev_handle), "could not open device %s (interface %d): %s", priv->usb_interface[i].path, i, windows_error_str(0));
// 				switch (GetLastError()) {
// 				case ERROR_FILE_NOT_FOUND: // The device was disconnected
// 					return LIBUSB_ERROR_NO_DEVICE;
// 				case ERROR_ACCESS_DENIED:
// 					return LIBUSB_ERROR_ACCESS;
// 				default:
// 					return LIBUSB_ERROR_IO;
// 				}
				ret = GetLastError();
				continue;
			}

			handle_priv->interface_handle[i].dev_handle = file_handle;
		}
		else
		{
			break;
		}
	}`

tormodvolden, 2022-04-19:
Can you please post the diff? 

See also #965 which may also deal with this.

mcuee, 2022-04-19:
> I want to open a usb composite device(uisng winusb driver) which has 2 interfaces. When another device use one interface, I get an LIBUSB_ERROR_ACCESS error while calling libusb_open.

Windows will normally treat the two interfaces as two devices. Did you install the WinUSB driver with the interface you are interested? 

Is this a USB IAD device? If yes, please try #965 as well.

FuChengYiXiao, 2022-04-22:
I will try #965, thanks.