libusb/libusb #1132: Seems to fail when i use a [bmRequestType].[recipient] of Interface - LINUX

brendena, 2022-04-19:
All code bellow is run from a linux ubuntu 20.04

What i'm trying to do is setup some code to talk to a extension unit on UVC class.  To do that i need to talk to a specific unit in the USB class.  But when i set th**e bmRequestType.recipient = 1 (Interface),**  it just fails to send anything.

So to simplify the problem i tried to do a simple **GET_STATUS** command an a device and a interface.  The GET_STATUS worked fine on the device but on the interface version it failed.  But i have similar code made with 4lv2 and it seems to work.    

```c
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <libusb-1.0/libusb.h>
#include <err.h>
#include <errno.h>

#define MFGR_ID 0x8087 // given manufacturer ID 
#define DEV_ID 0x0b6d  // given device ID

#define bmRequestType_DEVICE     0x80
#define bmRequestType_INTERFACE  0x81

int main() {
    int init = libusb_init(NULL); // NULL is the default libusb_context
    if (init < 0) {errx(1,"\n\nERROR: Cannot Initialize libusb\n\n");  }
    struct libusb_device_handle *dh = NULL; // The device handle
    dh = libusb_open_device_with_vid_pid(NULL,MFGR_ID,DEV_ID);
    if (!dh) {errx(1,"\n\nERROR: Cannot connect to device %d\n\n",DEV_ID);}

    unsigned char dataBuff[100];
    uint8_t       bmReqType = bmRequestType_DEVICE;   // 0x81 - interface 0x80 device
    uint8_t            bReq = 0;   // the request field for this packet
    uint16_t           wVal = 0;   // the value field for this packet
    uint16_t         wIndex = 0;   // the index field for this packet
    unsigned char*   data = dataBuff;   // the data buffer for the in/output data
    uint16_t           wLen = 2;   // length of this setup packet 
    unsigned int     to = 1000;       // timeout duration (if transfer fails)

    int config = libusb_control_transfer(dh,bmReqType,bReq,wVal,wIndex,data,wLen,to);
    if (config < 0) {
        printf("errno %d\n", errno);
        errx(1,"\n\nERROR: No data transmitted to device %d\n\n",DEV_ID);
    }
    // now you can use libusb_bulk_transfer to send raw data to the device
    printf("worked fine\n");
    libusb_exit(NULL);
}
```

build command **gcc  -o hostapp hostapp.c  -lusb-1.0**

I don't entirely know how to enable the debug logs in the C lib.  But There's a javascript wrapper for libusb where i'm also getting the exact same problem.  I would post the packet that it's would be sending out but the error seems to be before any data leave the usb.  

timestamp - threadID -facility level - function call <message>
--------------------------------------------------------------------------------
[ 0.081986] [00005da8] libusb: debug [libusb_alloc_transfer] transfer 0x69dc5f0
[ 0.082039] [00005da8] libusb: debug [libusb_submit_transfer] transfer 0x69dc5f0
[ 0.082046] [00005da8] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.082067] [00005da8] libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
[ 0.082073] [00005da8] libusb: debug [disarm_timerfd] 
[ 0.082078] [00005da8] libusb: debug [libusb_submit_transfer] transfer 0x69dc5f0
[ 0.082082] [00005da8] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.082089] [00005da8] libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
[ 0.082093] [00005da8] libusb: debug [disarm_timerfd] 

I'm fairly new to USB so sorry if i screwed up some of the terminology.

diabolo38, 2022-04-19:
for std request targeting an int reface  windex shall be the interface number , are your sure 0 is the right interface number ?

brendena, 2022-04-19:
So the actual interface that i'm trying to reach is 0x4, but to get a base line what i'm trying to do is what V2LC does.  The first thing it does it a get_status on both the device and interface of 0.  

This is the raw bytes from a wire shark capture of  V2LC get status on interface 0

0000   c0 49 a9 a8 bc 89 ff ff 53 02 80 02 06 00 00 3c
0010   f9 04 5f 62 00 00 00 00 f1 67 00 00 8d ff ff ff
0020   02 00 00 00 00 00 00 00 **[81 00 00 00 00 00 02 00]**  -- this is the line where the usb packets at 
0030   00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00

mcuee, 2022-04-20:
As mentioned by the answer of michel, your code is not correct and error number 16 (resource busy) is expected. 

Sorry but what is V2LC? 

And just wondering why you do not want to use libuvc ( https://github.com/libuvc/libuvc ) if your device is compliant to UVC class.

mcuee, 2022-04-20:
@brendena Also you need to detach the kernel driver and set up proper permission (say using udev rules) in order to use libusb to talk to a UVC device which has already the kernel driver attached.

The Get_Status call is a standard USB request which typical kernel driver will do during the enumeration, similar to Set_Address. Usually you do not need to deal with them in libusb.

Ref: https://www.usbmadesimple.co.uk/ums_4.htm

mcuee, 2022-04-20:
Good info
https://people.redhat.com/zaitcev/notes/usb_stern_cheat.txt

```
Author: Alan Stern (at harward.edu)
Date posted: Thu, 1 Mar 2007 14:05:58 -0500 (EST)

		Standard device requests

RQtype Req	Value		Index	Length
----------------------------------------------
  0	1	dev-feat	0	0	Clear device feature
  0	3	dev-feat	0	0	Set device feature
  0	5	address		0	0	Set device address
  0	9	config-value	0	0	Set configuration
  1	b	altsetting	intf	0	Set interface
  1	1	intf-feat	intf	0	Clear interface feature
  1	3	intf-feat	intf	0	Set interface feature
  2	1	ep-feat		ep	0	Clear endpoint feature
  2	3	ep-feat		ep	0	Set endpoint feature
 80	0	0		0	2	Get device status
 80	6	descr-type/	0/	len	Get descriptor
		index		lang-ID
 80	8	0		0	1	Get configuration
 81	0	0		intf	2	Get interface status
 81	a	0		intf	1	Get interface (altsetting)
 82	0	0		ep	2	Get endpoint status

Device features:	0 = self-powered, 1 = remote wakeup
Interface features:	None
Endpoint features:	0 = halt
Descriptor types:	1 = device, 2 = config, 3 = string, (4 = interface,
	5 = endpoint), 6 = device qualifier, 7 = other-speed config

		Hub class requests

RQtype Req	Value		Index	Length
----------------------------------------------
 20	1	hub-feat	0	0	Clear hub feature
 20	3	hub-feat	0	0	Set hub feature
 23	1	port-feat	sel/	0	Clear port feature
				port
 23	3	port-feat	sel/	0	Set port feature
				port
 a0	0	0		0	4	Get hub status
 a0	6	2900		0	len	Get hub descriptor
 a3	0	0		port	4	Get port status

Hub features:	0 = hub local power, 1 = hub over current,
		10 = ch hub local power, 11 = ch hub over current
Port features:	0 = connect, 1 = enable, 2 = suspend, 3 = over current,
		4 = reset, 8 = power, 9 = low speed, a = high speed,
		b = test mode, c = indicator,
		10 = ch connect, 11 = ch enable, 12 = ch suspend,
		13 = ch over current, 14 = ch reset
Selector is used for port indicators

		HID class requests

RQtype Req	Value		Index	Length
----------------------------------------------
 21	9	report-type/ID	intf	len	Set report
 21	a	duration/ID	intf	0	Set idle
 21	b	proto		intf	0	Set protocol
 81	6	descr-type/set	intf	len	Get class descriptor
 a1	1	report-type/ID	intf	len	Get report
 a1	2	0/ID		intf	1	Get idle
 a1	3	0		intf	1	Get protocol

Report type:		1 = input, 2 = output, 3 = feature
Descriptor type:	21 = HID, 22 = Report, 23 = Physical
Protocol:		0 = boot, 1 = report

```

mcuee, 2022-04-20:
This may help for UVC device.
http://www.ideasonboard.org/uvc/usbmon.py
[usbmon.py.txt](https://github.com/libusb/libusb/files/8517834/usbmon.py.txt)

diabolo38, 2022-04-20:
some basic : 
to get control of uvc or any device that has a kernel driver you shall detach it's kernel driver first if any .. 
then to do most ops on an interface you shall start by claiming it first

brendena, 2022-04-20:
>  In you case, just wondering why you choose to use libusb.

* Cross compatible with both windows and linux.  
* There's a [node wrapper](https://www.npmjs.com/package/usb) which is important because what i'm making is a electron nodejs application 

>Sorry but what is V2LC? Do you mean Video4Linux?

Typo it was suppose to be V4L2.

> Does the kernel V4L2 driver and API work for you?

Yes i can get it to work, but since it's not cross compatible i don't really want to use it.

> If not, you may want to use libuvc ( https://github.com/libuvc/libuvc ) which is based on libusb. Then you should not go through the hardwork of reverse engineering.

I think i'll run into the same problems.  To me it seems like the software just doesn't work with the given call.  Also i was planning on grabbing the image with opencv.  So i just needed a way to send a few of these extension units.  Which i don't see a option for with libuvc.  But it does look like somebody did create a [wrapper for it](https://github.com/joelpurra/node-libuvc)

brendena, 2022-04-20:
@diabolo38 @mcuee yep i think you where both right about the detaching kernel thing.  I was able to get the camera feed and use a extension with libuvc.  Thanks a lot.  I was figuring if i didn't open the driver then i would be able to connect to the usb device with a different driver.  

Just for reference the libuvc code looks like this
uvc_extension_unit_t * extension = uvc_get_extension_units(devh);
unsigned char data[1];
uvc_get_ctrl(devh,extension->bUnitID, 1,data,1,UVC_GET_CUR);