libusb/libusb #1135: 1.0.26 unnecessary link with -latomic

abelbeck, 2022-04-23:
After cross-compiling 1.0.26 it unnecessarily links with `-latomic` which we don't include in our project.

x86_64-unknown-linux-gnu-gcc (crosstool-NG 1.24.0) 6.5.0

As a temporary fix, I removed this line in `configure.ac`:
```
AC_SEARCH_LIBS([__atomic_fetch_add_4], [atomic])
```
and `lsusb` works just fine.

This line got added here: https://github.com/libusb/libusb/commit/561dbdacfbcd8f26c81900d1a1a16d2f44093e0e

mcuee, 2022-04-29:
The temporary fix is not really correct. Just wondering if you have a better fix? Maybe you can try to debug why the check in https://github.com/libusb/libusb/commit/561dbdacfbcd8f26c81900d1a1a16d2f44093e0e is not correct for your case.

tormodvolden, 2022-05-01:
@ffontaine Does the suggested patch work on your (sparc) platforms?

ffontaine, 2022-05-08:
Yes, the suggested patch work on sparc platforms.

tormodvolden, 2022-05-08:
@abelbeck Can you please make a pull request?

abelbeck, 2022-05-08:
@tormodvolden : PR #1139

commit 95e601ce (2022-06-26): configure.ac: Link with -latomic only if no atomic builtins

Follow-up to 561dbda, a check of GCC atomic builtins needs to be done
first.

I'm no autoconf guru, but using this:
https://github.com/mesa3d/mesa/blob/0df485c285b73c34ba9062f0c27e55c3c702930d/configure.ac#L469
as inspiration, I created a pre-check before calling AC_SEARCH_LIBS(...)

Fixes #1135
Closes #1139
closed 2022-06-26 by tormodvolden