libusb/libusb #1137: got "unrecognised urb status -1" issue when getting data.

fsuper, 2022-04-28:
I am using libusb+libuvc to get frames from usb camera. But sometimes, the size of the frame is not right. From the log of libusb, I found some logs like this:
libusb: warning [handle_iso_completion] packet 13 - unrecognised urb status -1

[image]

I checked the code flow as following:
1. reap_for_handle：r = ioctl(hpriv->fd, IOCTL_USBFS_REAPURBNDELAY, &urb);
2. handle_control_completion： check the urb->status, if urb->status is not in the defined list, the error log will occur.

code is : src/main/jni/libusb-android/libusb/os/linux_usbfs.c   #2641 line.

I am wondering what the “urb status = -1” stands for? or how can I debug it?  thank you

mcuee, 2022-04-28:
Does it work under normal Linux distribution? If it is Android, what is your Soc and Android version?
Can you post the full debug log in plain text format (you can zip it).

fsuper, 2022-04-29:
> Does it work under normal Linux distribution? If it is Android, what is your Soc and Android version? Can you post the full debug log in plain text format (you can zip it).

Dear mcuee,
Thanks for your reply. 
It is running on Android. 
Mobile name: Galaxy S22.
Model number: SM-S9010
Android version: 12

here is the debug logs. 
[2022-4-29_debug.zip](https://github.com/libusb/libusb/files/8587634/2022-4-29_debug.zip)

mcuee, 2022-04-29:
You are not generating the debug log correctly since it includes many other non-related stuff. What you should do it to set LIBUSB_DEBUG=4 and then run your application and redirect the debug log to a file. I have no idea about Android so I am not so sure how you can achieve that though.

What is the libusb version you are using? libusb release or git tree does not work well under non-rooted Android. So I tend to believe you are not really using the official libusb release.

You may want to try out #874.
* #874

fsuper, 2022-05-05:
> You are not generating the debug log correctly since it includes many other non-related stuff. What you should do it to set LIBUSB_DEBUG=4 and then run your application and redirect the debug log to a file.
> 
> ```
> set LIBUSB_DEBUG=4
> my_application > debug.log 2>&1
> ```
> 
> I have no idea about Android so I am not so sure how you can achieve that under Android though.
> 
> What is the libusb version you are using? libusb release or git tree does not work well under non-rooted Android. So I tend to believe you are not really using the official libusb release.
> 
> You may want to try out #874.
> 
> * [Better Android Support #874](https://github.com/libusb/libusb/pull/874)

I used both of the git tree and “libusb-android” version. The same issue happend. I am wondering if anyone also encountered the same issue, "unrecognised urb status -1"

mcuee, 2022-05-05:
> I used both of the git tree and “libusb-android” version. The same issue happend. I am wondering if anyone also encountered the same issue, "unrecognised urb status -1"

Did you try under normal Linux machine to see if the issue happens there as well? If not, then the problem is specific to your Andorid platform. Android is rather fragmental so things happens.

fsuper, 2022-05-06:
[2022-5-6_log.zip](https://github.com/libusb/libusb/files/8637201/2022-5-6_log.zip)

@mcuee  thanks for your reply.  I changed the log level in android code and now it contains full debug log.

mcuee, 2022-05-06:
You are leaving out the important initial debug log which shows libusb version used.

Anyway, not so sure if someone can look at the debug log to see what "urb status -1" means.
```
05-06 10:22:07.877 11102 11258 D libusb  : [139.602694] [00002bfa] libusb: debug [libusb_submit_transfer] transfer 0x797bbd01b8
05-06 10:22:07.877 11102 11258 D libusb  : [139.602732] [00002bfa] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 1858560
05-06 10:22:07.878 11102 11258 D libusb  : [139.603655] [00002bfa] libusb: debug [libusb_get_next_timeout] next timeout in 4.971501s
05-06 10:22:07.878 11102 11258 D libusb  : [139.604093] [00002bfa] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
05-06 10:22:07.878 11102 11258 D libusb  : [139.604123] [00002bfa] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 4972ms
05-06 10:22:07.882 11102 11258 D libusb  : [139.608172] [00002bfa] libusb: debug [usbi_wait_for_events] poll() returned 1
05-06 10:22:07.883 11102 11258 D libusb  : [139.608638] [00002bfa] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=660
05-06 10:22:07.883 11102 11258 D libusb  : [139.608684] [00002bfa] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
05-06 10:22:07.883 11102 11258 D libusb  : [139.608697] [00002bfa] libusb: debug [handle_iso_completion] all URBs in transfer reaped --> complete!
05-06 10:22:07.883 11102 11258 D libusb  : [139.608710] [00002bfa] libusb: debug [usbi_handle_transfer_completion] transfer 0x797bbd12f8 has callback 0x77fdbd5fc4
05-06 10:22:07.883 11102 11258 D libusb  : [139.608729] [00002bfa] libusb: debug [libusb_submit_transfer] transfer 0x797bbd12f8
05-06 10:22:07.883 11102 11258 D libusb  : [139.608742] [00002bfa] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 1858560
05-06 10:22:07.883 11102 11258 D libusb  : [139.609239] [00002bfa] libusb: debug [libusb_get_next_timeout] next timeout in 4.973453s
05-06 10:22:07.883 11102 11258 D libusb  : [139.609285] [00002bfa] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
05-06 10:22:07.883 11102 11258 D libusb  : [139.609299] [00002bfa] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 4974ms
05-06 10:22:07.889 11102 11258 D libusb  : [139.615277] [00002bfa] libusb: debug [usbi_wait_for_events] poll() returned 1
05-06 10:22:07.891 11102 11258 D libusb  : [139.616754] [00002bfa] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=147860
05-06 10:22:07.891 11102 11258 D libusb  : [139.616900] [00002bfa] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
05-06 10:22:07.891 11102 11258 D libusb  : [139.616997] [00002bfa] libusb: debug [handle_iso_completion] all URBs in transfer reaped --> complete!
05-06 10:22:07.891 11102 11258 D libusb  : [139.617046] [00002bfa] libusb: debug [usbi_handle_transfer_completion] transfer 0x797bbb9778 has callback 0x77fdbd5fc4
05-06 10:22:07.891 11102 11258 D libusb  : [139.617257] [00002bfa] libusb: debug [libusb_submit_transfer] transfer 0x797bbb9778
05-06 10:22:07.891 11102 11258 D libusb  : [139.617341] [00002bfa] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 1858560
05-06 10:22:07.893 11102 11258 D libusb  : [139.618591] [00002bfa] libusb: debug [libusb_get_next_timeout] next timeout in 4.970696s
05-06 10:22:07.893 11102 11258 D libusb  : [139.618673] [00002bfa] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
05-06 10:22:07.893 11102 11258 D libusb  : [139.618721] [00002bfa] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 4971ms
05-06 10:22:07.896 11102 11258 D libusb  : [139.621963] [00002bfa] libusb: debug [usbi_wait_for_events] poll() returned 1
05-06 10:22:07.897 11102 11258 D libusb  : [139.622811] [00002bfa] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=347848
05-06 10:22:07.897 11102 11258 D libusb  : [139.622910] [00002bfa] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
05-06 10:22:07.897 11102 11258 W libusb  : [139.622958] [00002bfa] libusb: warning [handle_iso_completion] packet 50 - unrecognised urb status -1
05-06 10:22:07.897 11102 11258 D libusb  : [139.623000] [00002bfa] libusb: debug [handle_iso_completion] all URBs in transfer reaped --> complete!
05-06 10:22:07.897 11102 11258 D libusb  : [139.623044] [00002bfa] libusb: debug [usbi_handle_transfer_completion] transfer 0x797bbed7c8 has callback 0x77fdbd5fc4
05-06 10:22:07.897 11102 11258 D libuvc  : [stream.c:844/_uvc_stream_callback] bad packet (isochronous transfer); status: 1

```

blackox626, 2023-08-25:
Any progress on this issue? I am experiencing the same problem. I am using libusb version 1.0.26, and the video display is flickering on Android 12, with an error log reporting 'unrecognized urb status -1.' Additionally, I have run libuvc's example code on Ubuntu, and the video displaying normally with no flickering when using OpenCV.

mcuee, 2023-08-29:
@blackox626 

You have to provide more info like the full debug log.

I tend to think it is a Android Linux Kernel issue or USB stack issue and not a libusb issue.

blackox626, 2023-08-30:
> 

```
2023-08-30 12:18:24.006 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 11 - unrecognised urb status -1
2023-08-30 12:18:24.006 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 11
2023-08-30 12:18:24.006 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:31.770 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 15 - unrecognised urb status -1
2023-08-30 12:18:31.770 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 15
2023-08-30 12:18:31.770 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:31.770 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:38.758 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 22 - unrecognised urb status -1
2023-08-30 12:18:38.758 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 22
2023-08-30 12:18:38.758 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:38.758 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:39.338 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 20 - unrecognised urb status -1
2023-08-30 12:18:39.338 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 20
2023-08-30 12:18:39.338 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:39.338 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:39.938 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 22 - unrecognised urb status -1
2023-08-30 12:18:39.938 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 22
2023-08-30 12:18:39.938 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:40.542 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 26 - unrecognised urb status -1
2023-08-30 12:18:40.542 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 26
2023-08-30 12:18:40.542 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:40.542 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:40.682 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 0 - unrecognised urb status -1
2023-08-30 12:18:40.682 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 0
2023-08-30 12:18:40.682 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:40.682 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:41.402 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 22 - unrecognised urb status -1
2023-08-30 12:18:41.402 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 22
2023-08-30 12:18:41.402 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:41.402 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:41.782 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 29 - unrecognised urb status -1
2023-08-30 12:18:41.782 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 29
2023-08-30 12:18:41.782 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:41.782 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:42.886 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 26 - unrecognised urb status -1
2023-08-30 12:18:42.886 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 26
2023-08-30 12:18:42.886 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:42.886 11290-12876 libuvc             D  [stream.c:756/_uvc_process_payload] bad packet: error bit set
2023-08-30 12:18:43.186 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 0 - unrecognised urb status -1
2023-08-30 12:18:43.186 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 0
[... cut ...]
2023-08-30 12:18:45.358 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
2023-08-30 12:18:46.806 11290-12876 libusb             W  libusb: warning [handle_iso_completion] packet 4 - unrecognised urb status -1
2023-08-30 12:18:46.806 11290-12876 libuvc             D  [stream.c:835/_uvc_stream_callback] (isochronous transfer); transfer->packet_id: 4
2023-08-30 12:18:46.806 11290-12876 libuvc             D  [stream.c:837/_uvc_stream_callback] bad packet (isochronous transfer); status: 1
```

mcuee, 2023-08-31:
From the above log, it seems this is not a libusb issue but rather an Android issue.

blackox626, 2023-09-04:
@mecu    thanks 

The flickering of the video display on the Android 12 system is not related to the 'unrecognized urb status -1' error. 

The submit_iso_transfer works normally on Linux, but encounters problems on the Android system. Reducing the size of the urb and adjusting some urbs resolve the flickering issue on Android. 

The video display no longer flickers, but there are still 'unrecognized urb status -1' error logs appearing.

mcuee, 2023-09-04:
> The submit_iso_transfer works normally on Linux, but encounters problems on the Android system. Reducing the size of the urb and adjusting some urbs resolve the flickering issue on Android.
> 
> The video display no longer flickers, but there are still 'unrecognized urb status -1' error logs appearing.

Looks like an Android USB stack issue then, and not a libusb issue per se.

I will still keep this issue open though as it seems to be a common issue.

06110902002, 2023-10-07:
I faced the same problem in android 9.0

06110902002, 2023-10-07:
I faced the same problem in android 9.0
libusb 1.0.26

mcuee, 2023-10-08:
> I faced the same problem in android 9.0 libusb 1.0.26

That may well be Android 9.0 problem since it has an ancient Linux kernel. You have to get the support of your Android provider.

06110902002, 2023-10-08:
@mcuee  Thanks for your reply