libusb/libusb #1140: libusb_submit_transfer return LIBUSB_ERROR_NOT_FOUND

leonli8012, 2022-05-11:
In windows 11 x64 system, trying to connect a composite device( WinUSB and MSC), Mass Storage Device works fine, while I try to send data through WinUSB, my app exit with error at libusb_submit_transfer func with error code LIBUSB_ERROR_NOT_FOUND, I tried different interfaces and endpoints, can't solve this problem.

The device is well recognized in Device Manager, and while I try using VS Code with API WritePipe & ReadPipe, it works fine.

leonli8012, 2022-05-11:
forgot one thing, when the device is a single WinUSB device, everything works fine, sending and receiving are both OK.

mcuee, 2022-05-11:
Is this using IAD?
Try the pull request https://github.com/libusb/libusb/pull/965
* https://github.com/libusb/libusb/pull/965

leonli8012, 2022-05-27:
hi mcuee,
my device didn't generate IAD, and anther problem is I tried to modify the lib source file to support IAD but I fail to generate the lib file for windows, it's too complicated for me. I guess what I can do is to wait for official update release.

mcuee, 2022-05-27:
If your device is not IAD, #965 may not help you much. In the mean time, you can use the work-around by using WinUSB as the driver of the parent device (you can use Zadig to do that) -- you will lose the mass storage function though.

Please provide the debug log by setting LIBUSB_DEBUG=4 and then run your codes. Please also post the descriptors of your USB device. Thanks.

mcuee, 2022-06-18:
I will close this as the OP does not come back within a reasonable time.

closed 2022-06-18 by mcuee