libusb/libusb #1141: libusb_open LIBUSB_ERROR_ACCESS

RetroHQ, 2022-05-14:
I have compiled libusb for Windows using VS2019 as a static library and have a simple test app to initialise the library, get the devices and try and open a specific (FTDI) device. It always returns LIBUSB_ERROR_ACCESS (-3).

```
	libusb_context* usb_ctx;
	if (libusb_init(&usb_ctx) == 0)
	{
		libusb_device** devs;
		if (libusb_get_device_list(usb_ctx, &devs) >= 0)
		{
			libusb_device* dev;
			int i = 0;
			while ((dev = devs[i++]) != NULL)
			{
				libusb_device_descriptor desc;
				if (libusb_get_device_descriptor(dev, &desc) == 0)
				{
					if (desc.idVendor == 0x0403 && desc.idProduct == 0x6014)
					{
						libusb_device_handle* usb_dev;
						int ret = libusb_open(dev, &usb_dev);

						printf("%d\n", ret);
					}
				}
			}
		}
	}

```

I have also compiled the code under MinGW64 with the same results.

The drivers are setup correctly with "libusb0 (v1.2.6.0)" using Zadig and the device functions correctly when using OpenOCD under MinGW64 which is using libusb.

It must be something I am doing as the device functions in OpenOCD built in the same environment and the code appears to be the same as is used to initialise the device under OpenOCD.

Any thoughts?

mcuee, 2022-05-14:
Which version of libusb are you using? You should use libusb 1.0.26 and try agin. 1.0.24 is known to have problem. Your Openocd binary may be using a different version of libusb.

RetroHQ, 2022-05-14:
I am using a source built version of 1.0.26 with VS2019 and MinGW is using the same paths as OpenOCD for libusb, although I'm not entirely sure what version it is. I've just tried switching to WinUSB driver, and that does open correctly bother under VS and MinGW. The same for "libusbK (v3.1.0.0)". Although some other tools I rely on using with this device also need either libusb0 or libusbK, so at least I can use libusbK.

Eurgh. Even more confusingly after installing libusbK and confirming this worked then switching back to libusb0 to confirm it was still broken, it now also works. This is so bizarre as this was failing with two different devices (232 and 2232) and I only swapped one driver over and back, now both work.

No idea what's going on there, but one way or another I should have a workaround. Thanks for the input! :)

closed 2022-05-14 by RetroHQ

mcuee, 2022-05-14:
I see. Looks like you are missing libusbk.dll initially. You need to have libusbk.dll in order to use either libusbk.sys or libusb0.sys driver.