libusb/libusb #1142: libusb_bulk_transfer is blocked

aichimangguo, 2022-05-20:
libusb_bulk_transfer fucntions be invoked multiple times. It is locked. This time invoking libusb_clear_halt function can again to write.
Why this is happening?

mcuee, 2022-05-20:
If clear halt is required, most likely the device FW has some problems.

Please provide more info about libusb version, OS version, host USB chipset info, device info, to see how to debug further.

mcuee, 2022-05-20:
Please also provide the debug log by setting LIBUSB_DEBUG=4 and run your application.

diabolo38, 2022-05-23:
right some f/W and device stall their ep to signal some error or request for some attention on pc s/w 
in what case you get and bulk transfer with error and likely non 0 transfer size. 
After that pc sw  will check/clear the error in device  , clr halt on the ep as  needed and continue with data streaming to/fm ep.