libusb/libusb #1144: distinct libusb_device* returned for the same USB device

mildsunrise, 2022-05-24:
We seem to be experiencing an issue on Windows, where `libusb_get_device_list` constantly returns a new `libusb_device*` that seems to refer to the same physical USB device as another `libusb_device*` that was previously returned (and hasn't been unreffed) in the same context.

Is this a libusb bug, or is it expected behavior?
In other words, can we rely on each USB device having at most 1 reffed `libusb_device*` in a libusb context, or should we perform additional checks (and if so, which ones)?
The documentation doesn't seem to explicitly say, so I assumed it was a guarantee based on the reffing aspect of the API.

Refs: node-usb/node-usb#516 node-usb/node-usb#517

mcuee, 2022-05-24:
Take note hotplug is not yet supported under Windows.

Maybe you want to post your debug log with env variable LIBUSB_DEBUG=4.

mcuee, 2022-05-24:
If you have multiple devices with the same VID/PID, the best is to differentiate them with serial number if the device has serial number.

mildsunrise, 2022-05-24:
We're aware hotplug isn't supported under Windows; this is a hackish 'polyfill' that simply calls `libusb_get_device_list` periodically and compares it with the previously returned list. Right now we're assuming that libusb will never return distinct `libusb_device` referring to the same USB device. Based on your response I understand this isn't a guarantee, right?

Regarding the checks you suggested, wouldn't it be better to use libusb_get_bus_number + libusb_get_device_address?

mcuee, 2022-06-01:
Close this for now. Please reopen if you encounter further issues.

mildsunrise, 2022-06-01:
Hey, sorry I left this unattended. Upon further investigation, it was a bug on our side; libusb DOES seem to respect the "one device = one libusb_device" rule.

however I'm still worried that this not a strong guarantee according to what you said previously.
if it's not a guarantee, consider updating the documentation to warn users. something like:

> **Warning:** There is no strong guarantee that two `libusb_device*` will refer to distinct physical devices. To check that this is the case, you need to compare [insert whatever attributes are deemed appropriate here].

mildsunrise, 2022-06-01:
no, my question doesn't involve hotplugging.
assume I connect a physical device `A`, and then libusb returns a libusb_device `x` that refers to it.
if I don't disconnect `A`, and I don't unref `x`, can libusb ever return a distinct libusb_device `y` (within the same context) that also refers to `A`?

if the answer is that it can't happen, then you can close this issue :) but if this is allowed to happen, a note in the documentation would be nice

mcuee, 2022-06-01:
Then it can not happen. I will close this issue.