libusb/libusb #1146: Install libusb 1.0.24 on Mac (intel)

kosblag, 2022-05-26:
Could anyone please help to install libusb 1.0.24 on Mac?
Thank You!

mcuee, 2022-05-27:
Why do you need the older version?

You can always build from the source codes. Xcode project is provided.
https://github.com/libusb/libusb/releases/tag/v1.0.24

If you use Homebrew, there are hacks to install older version as well.
Ref: https://itnext.io/how-to-install-an-older-brew-package-add141e58d32

closed 2022-05-27 by mcuee

kosblag, 2022-05-27:
Installation of libusb from a GitHub commit URL is unsupported.

I use one repository, which works only with older version of libusb.

mcuee, 2022-05-27:
Not so sure what you mean here. If you do not want to use git repo, you can always use the release versio which you can download from gitbub or Sourceforge. And then you can build libusb from source.

Download of release tar ball.
https://github.com/libusb/libusb/releases
https://sourceforge.net/projects/libusb/files/libusb-1.0/

mcuee, 2022-05-27:
BTW, there are a few issues with libusb-1.0.24 and libusb-1.0.25 for macOS. It is really recommended to update to libusb-1.0.26.
* https://github.com/libusb/libusb/issues/777
* https://github.com/libusb/libusb/issues/1047