libusb/libusb #1147: [Android ]libusb v1.0.26 submit_iso_transfer return ENOMEM caused by event poll() stuck

droneboost, 2022-06-02:
Send PCM audio data from some Android phone to DAC device with usb iso transfer.
Appears to be short of kernel memory, but observed USB event handling thread stuck in poll() from certain time point and never returned. Here is the log output: https://gist.github.com/droneboost/9af7e59c40056d86961ff7f3046fcb10
...........................................................................................................................................................................................................................................
  [20.215666] [00005f8a] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
D/libusb: [20.215457] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.215830] [00005f8a] libusb: debug [usbi_wait_for_events] poll() returned 1
    [20.216075] [00005f8a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=576
D/libusb: [20.216113] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8a8360
D/libusb: [20.216123] [00005f8a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
D/libusb: [20.216231] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.216333] [00005f8a] libusb: debug [handle_iso_completion] all URBs in transfer reaped --> complete!
    [20.216653] [00005f8a] libusb: debug [usbi_handle_transfer_completion] transfer 0x70aa8aa020 has callback 0x70a72db6f4
    [20.216695] [00005f8a] libusb: debug [libusb_free_transfer] transfer 0x70aa8aa020
    [20.217036] [00005f8a] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
D/libusb: [20.217055] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8a84a0
D/libusb: [20.217090] [00005f8a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    **[20.217344] [00005f8a] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms  <= Never return from **here****
D/libusb: [20.217216] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.217654] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8a9760
    [20.217707] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.218282] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8a98a0
    [20.218349] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.218485] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab1a0
    [20.218519] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.219128] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab2e0
D/libusb: [20.219194] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.219327] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab420
    [20.219359] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.219950] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab560
    [20.220007] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.220126] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab6a0
    [20.220157] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.222058] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x70aa8ab7e0
D/libusb: [20.222227] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
...........................................................................................................................................................................................................................................
...........................................................................................................................................................................................................................................
 [20.587941] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dc20
    [20.587957] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.588532] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dd60
    [20.588578] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
    [20.588629] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dea0
    [20.588645] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
D/libusb: [20.589265] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.589340] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.589567] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.589637] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.589821] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.589883] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.589982] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.590012] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.590146] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
D/libusb: [20.590189] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.590289] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.590319] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.590452] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.590486] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.590565] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.590604] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.590753] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.590793] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.590914] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.590964] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
D/libusb: [20.591180] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.591258] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.591417] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.591467] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.591639] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.591698] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.591841] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.591887] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [20.592023] [00005fb8] libusb: debug [libusb_submit_transfer] transfer 0x708619dfe0
    [20.592057] [00005fb8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 576
E/libusb: [20.592208] [00005fb8] libusb: error [submit_iso_transfer] submiturb failed, errno=12
D/libusb: [20.592228] [00005fb8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
...........................................................................................................................................................................................................................................

mcuee, 2022-06-05:
Could you test with a normal Linux machine to see if you have the same problem or not?

Android + libusb isoc do not seem to work well. I tend to think this is a Anrdoid kernel issue.

Similar issues:
* 1020
* 1137

droneboost, 2022-06-07:
@mcuee , 

>Android phones which have no issues? Which are the phones which have issues?

FYI. Here below are the phones which I tested:

[ISO transfer OK phones]:
Huawei P9,                     Android 8.0.0,                 Linux Kernel: 4.4.23+,    CPU: Hisilicon Kirin 955
Huawei P40 Pro,             HarmonyOS 2.0.0.236,   Linux Kernel: 4.14.116,   CPU: Hisilicon Kirin 990 5G
HuaWei Mate30 Pro,      Harmony OS 2.0.0,         Linux Kernel: 4.14.116,   CPU: HuaWei Kirin 990. 
Xperia XQ-BC72(1 III),     Android 12,                   Linux Kernel: 5.4.134,     CPU: Qualcomm Snapdragon 888.

[ISO transfer NG phones]:
vivo X60 Pro,  Android 11,  Linux Kernel: 5.4.129,    CPU: MTK Dimensity 720
oppo K7x,       Android 11,  Linux Kernel: 4.14.186,  CPU: Samsung Exynos 1080

mcuee, 2022-06-07:
@droneboost Thanks a lot for the detailed report. 
I think this points more to the issues of the Android kernel than libusb. One way is to use Linux kernel debug info if that is possible. Or you may have to use USB HW Analyzer to see what happens.

droneboost, 2022-06-07:
> Or you may have to use USB HW Analyzer to see what happens.

OK, I got a USB Analyzer on my hand, I'll try it and feedback later.

droneboost, 2022-06-07:
@mcuee, I checked with my USB packet analyzer, the kernel do stop to send ISO packet to DAC devices when issue occurred, but one strange things is that on the same phone, other APP use libusb v1.0.22 has no problem to send audio data to DAC, no ENOMEM observed. Is there any bigger change in the event handling between v1.0.22 and v1.0.24?

mcuee, 2022-06-07:
The main change log is here.
https://github.com/libusb/libusb/blob/master/ChangeLog

There is one big change in this commit.
https://github.com/libusb/libusb/commit/d67eb5beaa44c17c09f089a522ff483f4978a978

Can you try your app with the same phone but using libusb-1.0.22 to see it works or not? And you can try to narrow down the issue to see if the above commit makes a difference.

droneboost, 2022-06-15:
Slow down the transfer speed will avoid the enomem, but can't catch up the audio sample rate.

closed 2023-05-18 by droneboost

mcuee, 2023-05-18:
> Slow down the transfer speed will avoid the enomem, but can't catch up the audio sample rate.

Thanks for the updates. Have you tried different Android phones to see if you see different transfer speed limit depending on the Android phone?