libusb/libusb #1150: MacOS 12 Monterey can not working.

Alvazz, 2022-06-10:
Hi all, I wan use on MacOS 12 Monterey, but can not working. I got some error -3 "access denied".

How can I fix it? Thanks.

mcuee, 2022-06-10:
What is the device? Most likely the kernel driver has claimed exclusive access to the device.

For example, if your device is a UVC device like a USB camera and then you will face this issue. You can not use libusb (or libuvc) in this case.

Alvazz, 2022-06-10:
Yes, device is a UVC device. If libuvc (libusb) is not suitable, which library should be used?

mcuee, 2022-06-10:
> Yes, device is a UVC device. If libuvc (libusb) is not suitable, which library should be used?

Sorry I do not know. You can check out and ask here.
* https://github.com/libusb/libusb/issues/1014

mcuee, 2022-06-10:
Ref: https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access
> Please take note that command line apps cannot use provisioning profiles and therefore cannot hold this entitlement so you have to run as root. As for the GUI application, as of now APple support is saying that `com.apple.vm.device-access` entitlement is not the right entitlement. So basically there is no solution as of now. This means libusb (and libraries like libuvc which uses libusb) may not be the right library to use if you hit this issue (Issue #972 and Issue #1014).

mcuee, 2022-06-10:
I will close this issue as it is out of control for libusb/libuvc.

closed 2022-06-10 by mcuee

Alvazz, 2022-06-15:
OK, thanks!