libusb/libusb #1153: [macOS 12.4+] failiure to set configuration and claim interface for devices

omerporze, 2022-06-16:
darwin_claim_interface seems to fail on macOS 12.4+ on some devices. 
I tried connection iOS devices and I get a `LIBUSB_ERROR_NOT_FOUND` as a result of that function.
On macOS version 12.3.1 the function works without issues.

For more info:
Attached here is output from the command `sudo usbmuxd -v -f` when connecting an iOS device to the computer.
macOS 12.3.1 (working as intended):
`[17:15:23.476][3] usbmuxd v1.1.2 starting up
[17:15:23.477][4] Creating socket
[17:15:23.477][4] Listening on /var/run/usbmuxd
[17:15:23.477][4] Initializing USB
[17:15:23.477][3] Using libusb 1.0.25
[17:15:23.482][4] Registering for libusb hotplug events
[17:15:23.482][4] Found new device with v/p 05ac:12a8 at 3-37
[17:15:23.482][2] Device 3-37: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[17:15:23.482][4] Detaching kernel driver for device 3-37, interface 1
[17:15:23.486][4] Setting configuration for device 3-37, from 5 to 4
[17:15:23.555][4] Found interface 1 with endpoints 04/85 for device 3-37
[17:15:23.562][4] Using wMaxPacketSize=512 for device 3-37
[17:15:23.562][4] USB Speed is 480 MBit/s for device 3-37
[17:15:23.562][4] 1 device detected
[17:15:23.562][3] Initialization complete
[17:15:23.562][4] Got lang ID 1033 for device 3-37 `
macOS 12.4 (failing to work):
`[17:58:39.477][3] usbmuxd v1.1.2 starting up
[17:58:39.477][4] Creating socket
[17:58:39.478][4] Listening on /var/run/usbmuxd
[17:58:39.479][2] chmod(/var/db/lockdown, 02775) failed: Operation not permitted
[17:58:39.479][4] Initializing USB
[17:58:39.479][3] Using libusb 1.0.26
[17:58:39.495][4] Registering for libusb hotplug events
[17:58:39.495][4] Found new device with v/p 05ac:12a8 at 2-52
[17:58:39.495][2] Device 2-52: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[17:58:39.495][4] Detaching kernel driver for device 2-52, interface 1
[17:58:39.496][4] Detaching kernel driver for device 2-52, interface 2
[17:58:39.496][4] Detaching kernel driver for device 2-52, interface 3
[17:58:39.497][4] Detaching kernel driver for device 2-52, interface 4
[17:58:39.497][4] Detaching kernel driver for device 2-52, interface 5
[17:58:39.497][4] Setting configuration for device 2-52, from 5 to 4
[17:58:39.568][4] Found interface 1 with endpoints 04/85 for device 2-52
[17:58:39.570][2] Could not claim interface 1 for device 2-52: error code: -5 LIBUSB_ERROR_NOT_FOUND
[17:58:39.570][4] 0 devices detected
[17:58:39.570][3] Initialization complete`
It appears that the failure stems from libusb not being able to claim the interface for some reason.
The attempt to claim the interface is done using: `libusb_claim_interface`
which repeatedly fails when calling `darwin_get_interface` or `darwin_set_configuration`.
I would greatly appreciate a fix and will supply more logs/info if needed.
Thanks

mcuee, 2022-06-16:
What are the devices in question? Apple has probably changed more stuff and the kernel now has exclusive access to the devices so that you can not use libusb. One example is UVC device.

As of now there are no solutions from libusb side. You can try run with sudo to see if that helps.

mcuee, 2022-06-17:
@omerporze  I will suggest you create an issue with usbmuxd side and see if this is a common problem or not.

If Apple changes the kernel side to claim the device exclusively, then libusb can not be used as of now (just like the situation for UVC camera) as mentioned in #1014.

mcuee, 2022-06-18:
@omerporze 
I just tested our iPhone X and it seems to work under macOS 12.4 (mac Mini M1).

```

mcuee@mcuees-Mac-mini usbmuxd % sudo usbmuxd -v -f
[09:56:19.488][3] usbmuxd v1.1.2 starting up
[09:56:19.489][4] Creating socket
[09:56:19.489][4] Listening on /var/run/usbmuxd
[09:56:19.489][4] Initializing USB
[09:56:19.489][3] Using libusb 1.0.26
[09:56:19.494][4] Registering for libusb hotplug events
[09:56:19.494][4] 0 devices detected
[09:56:19.494][3] Initialization complete
[09:56:22.958][4] Found new device with v/p 05ac:12a8 at 2-21
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
[09:56:22.959][4] Setting configuration for device 2-21, from 0 to 4
[09:56:22.959][2] Could not set configuration 4 for device 2-21: LIBUSB_ERROR_NO_DEVICE
[09:56:23.216][4] Found new device with v/p 05ac:12a8 at 2-22
[09:56:23.216][2] Device 2-22: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[09:56:23.216][4] Setting configuration for device 2-22, from 0 to 4
[09:56:23.219][4] Found interface 1 with endpoints 04/85 for device 2-22
[09:56:23.236][4] Using wMaxPacketSize=512 for device 2-22
[09:56:23.236][4] USB Speed is 480 MBit/s for device 2-22
[09:56:23.236][4] Got lang ID 1033 for device 2-22
[09:56:23.236][4] Got serial '222dxxxxxxxxxxx' for device 2-22
[09:56:23.236][3] Connecting to new device on location 0x20016 as ID 1
[09:56:23.236][3] Connected to v2.0 device 1 on location 0x20016 with serial number 222d2xxxxxx
^C
[09:56:26.392][3] Caught signal 2, exiting
[09:56:26.393][4] Event processing interrupted
[09:56:26.393][3] usbmuxd shutting down
[09:56:26.494][3] Removed device 1 on location 0x20016
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
[09:56:26.500][3] Shutdown complete
```

mcuee, 2022-06-18:
Same for an iPad 2019.

```
mcuee@mcuees-Mac-mini usbmuxd % sudo usbmuxd -v -f
[10:13:47.134][3] usbmuxd v1.1.2 starting up
[10:13:47.135][4] Creating socket
[10:13:47.135][4] Listening on /var/run/usbmuxd
[10:13:47.135][4] Initializing USB
[10:13:47.135][3] Using libusb 1.0.26
[10:13:47.141][4] Registering for libusb hotplug events
[10:13:47.141][4] Found new device with v/p 05ac:12ab at 2-15
[10:13:47.141][2] Device 2-15: Ignoring possibly bad configuration 5, choosing configuration 4 instead.
[10:13:47.141][4] Detaching kernel driver for device 2-15, interface 1
[10:13:47.143][4] Setting configuration for device 2-15, from 5 to 4
[10:13:47.156][4] Found interface 1 with endpoints 04/85 for device 2-15
[10:13:47.158][4] Using wMaxPacketSize=512 for device 2-15
[10:13:47.158][4] USB Speed is 480 MBit/s for device 2-15
[10:13:47.158][4] 1 device detected
[10:13:47.158][3] Initialization complete
[10:13:47.158][4] Got lang ID 1033 for device 2-15
[10:13:47.158][4] Got serial 'aa97282xxxxxx' for device 2-15
[10:13:47.158][3] Connecting to new device on location 0x2000f as ID 1
[10:13:47.158][3] Connected to v2.0 device 1 on location 0x2000f with serial number aa97282xxxxxx
^C
[10:13:56.188][3] Caught signal 2, exiting
[10:13:56.188][4] Event processing interrupted
[10:13:56.188][3] usbmuxd shutting down
[10:13:56.289][3] Removed device 1 on location 0x2000f
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 1 pipe 2
[10:13:56.295][3] Shutdown complete
```

mcuee, 2022-06-18:
I also tested a very old iPad Mini 1st gen and it seems to work as well.

mcuee, 2022-07-12:
> Macbook Pro Intel i7, will be able to also try M1 MBP in a few days.

That will be great. We have seen differences in macOS 12.4 behaviors between Intel Mac vs M1 Mac.
Ref: https://github.com/libusb/libusb/issues/1156

omerporze, 2022-07-24:
@mcuee I tested on M1 and it did not work as well. Same as in Intel. What is the diff that you saw in Intel vs M1?

mcuee, 2022-07-25:
> I tested on M1 and it did not work as well. Same as in Intel. What is the diff that you saw in Intel vs M1?

That is strange. There are some applications which are holding the device exclusively in your machine. You need to find out that application.

I do not have Intel machine. But there were problems with Intel side but not on M1 -- please refer to this issue. macOS 12.5 fixed the issue.
* https://github.com/libusb/libusb/issues/1156

omerporze, 2022-07-25:
Well I just tried updating to 12.5 and it fixed the issue. Looks like the problem was with macOS. Thanks for all the help!

closed 2022-07-25 by omerporze

mcuee, 2022-07-25:
Thanks a lot for the updates. Glad that updating macOS version to 12.5 fixed the issues for you.