libusb/libusb #1154: libusb sample hotplugtest.c not worked in android libusb v1.0.24

Nithu7, 2022-06-17:
Hi Team,

I have tried to run libusb sample **hotplugtest.c**  on android platform (OS v12) with my usb camera device and it is NOT working.
I checked the AOSP libusb version and it is **v1.0.24**

I replaced this with libusb version **v.1.0.21** & **v1.0.22** and it is working fine and I am not albe to see notification during device connect/disconnect on same android platform. I replaced libusb version **v1.0.23** and this also NOT working.

so v1.0.21 & 1.0.22 is working and v1.0.23 & v1.0.24 NOT working.
But In linux environment (ubntu 18.04) I did not see this kind of issue with **v1.0.24** as well.

So I Suspected this is something related with android + libusb version > v1.0.22
`
I just want to quickly check is this known issue,  is there any fix or patch is there?
Could you suggest?

Regards
Muthu

mcuee, 2022-06-17:
Can you try the following on top of 1.0.26 release? We do not support AOSP version -- you have to ask Google.
* #874

Nithu7, 2022-06-17:
Hi mcuee,

Sure I will try it,

Between AOSP libusb version I meant below. (v1.0.24 for android-12)
https://cs.android.com/android/platform/superproject/+/android-12.1.0_r8:external/libusb/

Nithu7, 2022-06-17:
Hi mcuee,

Updating information:

**v1.0.22 from https://github.com/libusb/libusb/releases**   **[OK]**
1. Download & Compile v1.0.22 from the link https://github.com/libusb/libusb/releases on my android environment.
2. Then take hotplugtest.c  from below path then generate executable by linking with v1.0.22  // Native compilation not JNI
    https://github.com/libusb/libusb/tree/master/examples
3. Confirmed hotplugtest.c Send event properly during device disconnect/reconnect.

**v1.0.23 from https://github.com/libusb/libusb/releases**  **[ FAIL]**
1. Download & Compile v1.0.23 from the link https://github.com/libusb/libusb/releases on my android environment.
2. Then take hotplugtest.c  from below path then generate executable by linking with v1.0.23  // Native compilation not JNI
    https://github.com/libusb/libusb/tree/master/examples
3. Confirmed hotplugtest.c NOT sending event during device disconnect/reconnect.

We compared  v1.0.22 & v1.0.23 libusb source and randomly applied following patch, after that v1.0.23 seems to be working for hotplugtest.c in android.

```
--- libusb-1.0.23/libusb/os/linux_usbfs.c	2019-08-29 02:26:18.000000000 +0530
+++ libusb-1.0.23-updated/libusb/os/linux_usbfs.c	2022-06-17 15:19:35.000000000 +0530
@@ -344,7 +344,7 @@
  * simply assume /dev/bus/usb rather then making libusb_init fail.
  * Make the same assumption for Android where SELinux policies might block us
  * from reading /dev on newer devices. */
-#if defined(USE_UDEV) || defined(__ANDROID__)
+#if defined(USE_UDEV)
 	if (ret == NULL)
 		ret = "/dev/bus/usb";
 #endif
@@ -534,10 +534,8 @@
 {
 #if defined(USE_UDEV)
 	return linux_udev_start_event_monitor();
-#elif !defined(__ANDROID__)
-	return linux_netlink_start_event_monitor();
 #else
-	return LIBUSB_SUCCESS;
+	return linux_netlink_start_event_monitor();
 #endif
 }
 
@@ -545,22 +543,20 @@
 {
 #if defined(USE_UDEV)
 	return linux_udev_stop_event_monitor();
-#elif !defined(__ANDROID__)
-	return linux_netlink_stop_event_monitor();
 #else
-	return LIBUSB_SUCCESS;
+	return linux_netlink_stop_event_monitor();
 #endif
 }
 
 static int linux_scan_devices(struct libusb_context *ctx)
 {
-	int ret = 0;
+	int ret;
 
 	usbi_mutex_static_lock(&linux_hotplug_lock);
 
 #if defined(USE_UDEV)
 	ret = linux_udev_scan_devices(ctx);
-#elif !defined(__ANDROID__)
+#else
 	ret = linux_default_scan_devices(ctx);
 #endif
 
@@ -573,7 +569,7 @@
 {
 #if defined(USE_UDEV)
 	linux_udev_hotplug_poll();
-#elif !defined(__ANDROID__)
+#else
 	linux_netlink_hotplug_poll();
 #endif
 }

```

Could you check and share your command about this? Can we get proper patch to fix it?

Regards
Muthu

mcuee, 2022-06-17:
@Nithu7 You should try 1.0.26 and git head to check if there are still issues or not. If you still have issues with the git head, then you can create a pull request with your patch. No point looking at 1.0.23 already.

Nithu7, 2022-06-17:
Hi @mcuee 

Got it, Previosly I forget to mention about latest version status, the issue still reproduced with v1.0.26 as well.

v1.0.21 & v1.022 - issue not come
v1.0.24, v1.0.25, v1.0.26  - issue has come.

mcuee, 2022-06-17:
> Hi @mcuee
> 
> Got it, Previosly I forget to mention about latest version status, the issue still reproduced with v1.0.26 as well.
> 
> v1.0.21 & v1.022 - issue not come v1.0.24, v1.0.25, v1.0.26 - issue has come.

Have you tried your patch against git head and see if it fixed the issue? git head is very much the same as 1.0.26 so I expected it has issues as well. Thanks.

Nithu7, 2022-06-21:
Hi @mcuee 

sorry I don't have much time to apply the patch.
The issue is still with libusb git head and cause the issue, the following patch will fix the issue.

libusb/os/linux_usbfs.h	2022-06-20 17:51:45.000000000 +0530
@@ -174,10 +174,8 @@
 {
 #if defined(HAVE_LIBUDEV)
 	return linux_udev_start_event_monitor();
-#elif !defined(__ANDROID__)
-	return linux_netlink_start_event_monitor();
 #else
-	return LIBUSB_SUCCESS;
+	return linux_netlink_start_event_monitor();
 #endif
 }
 
@@ -185,7 +183,7 @@
 {
 #if defined(HAVE_LIBUDEV)
 	linux_udev_stop_event_monitor();
-#elif !defined(__ANDROID__)
+#else
 	linux_netlink_stop_event_monitor();
 #endif
 }
@@ -194,7 +192,7 @@
 {
 #if defined(HAVE_LIBUDEV)
 	linux_udev_hotplug_poll();
-#elif !defined(__ANDROID__)
+#else
 	linux_netlink_hotplug_poll();
 #endif
 }

And It seems like following update cause the issue.
https://github.com/libusb/libusb/commit/2f3bc98b0d0f4766496df53c855685a5f0e5e7cf

I think the above update is still required, because android native module (like us) still want to use libusb instead of platform APIs (supported only JAVA layer).

Could you check it.?

mcuee, 2022-06-22:
@Nithu7 I will suggest that you create a pull requests so that others can review. Thanks. I do not know much about Android myself.

And indeed that commits may have some issues based on the comments.
* https://github.com/libusb/libusb/commit/2f3bc98b0d0f4766496df53c855685a5f0e5e7cf

Nithu7, 2022-07-31:
Hi @mcuee created pull request:
https://github.com/libusb/libusb/pull/1168