libusb/libusb #1156: libusb_get_port_numbers() is broken on MacOS 12.4 (21F79)

mvp, 2022-06-20:
Starting from MacOS 12.4 (21F79), `libusb_get_port_numbers()` is broken.

For example, on my Mac, building libusb from master branch and running `examples/listdevs` gives on MacOS 11.6.6:

```
0bda:5411 (bus 20, device 21) path: 1.3.4
0bda:0411 (bus 20, device 20) path: 5.3.4
0bda:0411 (bus 20, device 19) path: 5.3.3
0bda:5411 (bus 20, device 18) path: 1.3.3
0bda:0411 (bus 20, device 17) path: 5.4
0bda:5411 (bus 20, device 16) path: 1.4
0bda:0411 (bus 20, device 15) path: 5.3
0bda:5411 (bus 20, device 14) path: 1.3
0bda:0411 (bus 20, device 13) path: 5
0bda:5411 (bus 20, device 12) path: 1
``` 
Note that path is longer than 1 (contains dots) on older Mac, which is correct behavior.

But running the same libusb build against the same physical hub on MacOS 12.4 (21F79) gives this:
```
0bda:5411 (bus 20, device 13) path: 4
0bda:5411 (bus 20, device 12) path: 3
0bda:5411 (bus 20, device 11) path: 4
0bda:5411 (bus 20, device 10) path: 3
0bda:0411 (bus 0, device 10) path: 4
0bda:0411 (bus 0, device 9) path: 3
0bda:0411 (bus 0, device 8) path: 4
0bda:0411 (bus 0, device 7) path: 3
0bda:5411 (bus 20, device 9) path: 3
0bda:0411 (bus 0, device 6) path: 1
05ac:1460 (bus 20, device 4) path: 2
1050:0407 (bus 20, device 2) path: 2
05ac:100f (bus 20, device 1) path: 1
05ac:8104 (bus 128, device 6) path: 8
05ac:100e (bus 1, device 1) path: 1
```

So it is broken on MacOS 12.4 - path is always just length 1 - incorrect!

This seems to be recurrence of issue #693 (aka #707) which was resolved 2 years ago.
I do not see any suspicious changes to `darwin_usb.c` in last 2 years, so this must be change in MacOS 12.4 USB stack behavoir that needs to be fixed in libusb.

This issue breaks any software that depends on `libusb_get_port_numbers()`, in particular my utility [uhubctl](https://github.com/mvp/uhubctl).

cc: @hjelmn, @mcuee, @IgorAnohin.

mcuee, 2022-06-26:
Hmm, just tried on my Mac Mini M1 (macOS 12.4) and here is the output.

```
mcuee@mcuees-Mac-mini libusb % ./examples/listdevs 
1915:1025 (bus 2, device 10) path: 2.1.2.5.6
046d:c52b (bus 2, device 9) path: 2.1.2.7
1a40:0201 (bus 2, device 8) path: 2.1.2.5
1a40:0201 (bus 2, device 7) path: 2.1.2
0b95:1790 (bus 2, device 6) path: 6.3
0bda:5411 (bus 2, device 5) path: 2.1
05e3:0749 (bus 2, device 4) path: 6.1.1
0bda:0411 (bus 2, device 3) path: 6.1
0bda:0411 (bus 2, device 2) path: 6
0bda:5411 (bus 2, device 1) path: 2

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl
Current status for hub 2-2.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [1a40:0201 USB 2.0 Hub [MTT], USB 2.00, 7 ports, ganged]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-6.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-6 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect [0b95:1790 ASIX Elec. Corp. AX88179 000000000001E9]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

```

mcuee, 2022-06-26:
IOregExplorer output is also fine.

[image]

MPLew-is, 2022-06-26:
Have you tried an x86 Mac? That's where I'm experiencing this issue.

mcuee, 2022-06-26:
> Have you tried an x86 Mac? That's where I'm experiencing this issue.

I do not have an Intel Mac myself. Let's wait for more reports if the issue is really only for Intel Macs.

mvp, 2022-06-26:
I also have this issue on x86 MacBook Pro (circa 2020).

mcuee, 2022-06-27:
@mvp 
If IOreg Explorer IOUSB plane shows issues, I guess you can create a bug report to Apple.

Last time Apple was able to address the issues reported.

mvp, 2022-06-27:
@mcuee, I would have done it if knew how - never reported OS bugs before...

mcuee, 2022-06-27:
> @I would have done it if knew how - never reported OS bugs before...

@mvp No problem.
@hjelmn Maybe you can help to brief the process to @mvp, as I remember you did this before.

mvp, 2022-07-02:
@hjelmn , who can help us here? Do you know how to get ahold of Apple developers?
I do believe this is MacOS bug, since expected USB devices do not show up on IOUSB plane (x86 only, works fine on M1).

mvp, 2022-07-06:
Used Apple Feedback Assistant to file this issue to Apple FB10563007. Lets see what will come out of it.

seanm, 2022-07-08:
Could you share the report text?  Then others can file duplicates to increase priority.

mvp, 2022-07-11:
This is copy of my report to Apple (there is no response so far):

## Issue id: FB10563007
## Basic Information:
### Please provide a descriptive title for your feedback:
MacOS 12.4 (21F79) and later does not show all USB hubs on IOUSB device plane
### Which area are you seeing an issue with?
IOKit
### What type of issue are you reporting?
Incorrect/Unexpected Behavior
## Desciption:
### Please describe the issue and what steps we can take to reproduce it:
I am a developer for open source software uhubctl https://github.com/mvp/uhubctl.
This utility depends on open source library libusb which among other platforms supports MacOS.
Starting from MacOS 12.4 12.4 (21F79) I observe that libusb is not able to properly enumerate USB devices on x86 Mac (it works fine on M1 Macs).
This issue was filed with libusb https://github.com/libusb/libusb/issues/1156, (that link mostly contains steps to reproduce).
However, libusb developers suggest that this issue is with MacOS 12.4 itself - all appropriate USB devices should show on IOUSB plane properly, and on x86 Mac they do not after updating to MacOS 12.4.

Steps to reproduce:

1. Attach any USB3 hub, 10 ports or more to the Mac  - I used 16 port Rosonway USB 3.1 hub (this hub is constructed as 5 4-port USB hubs daisy chained together to provide 16 usable USB ports).
2. Download Apple utility IO Registry Explorer https://download.developer.apple.com/Developer_Tools/Additional_Tools_for_Xcode_13.4/Additional_Tools_for_Xcode_13.4.dmg
3. Using IO Registry Explorer on MacOS 11.6.6 (pre 12.4)  I can correctly observe all 5+5 4-port hubs that comprise Rosonway 16 port hub: https://user-images.githubusercontent.com/17501/174703987-592c3fa7-df72-4e07-b8b6-8538bf00702d.png
4. However, on MacOS 12.4+, having attached the same Rosonway hub, I can see only 2 hubs (1 USB2 + 1 USB3) listed on IOUSB plane https://user-images.githubusercontent.com/17501/174703530-f0684048-c492-4782-a561-0e98e377dc28.png

Expected result:
   All daisy chained hubs show up in IO Registry Explorer in IOUSB plane in proper tree (that works fine on MacOS 11.x or earlier, and it also works fine on MacOS 12.4 M1)

Actual result:
   On MacOS 12.4 x86, only one hub is present in IOUSB plane. This breaks device enumeration and also breaks libsub library and any software that is using it. Note that proper hub tree is visible on IOService plane https://user-images.githubusercontent.com/17501/174706834-a79a1879-70f7-4753-b5ee-6eb887a32f7d.png, but it is not used by USB stack enumeration and is not really helpful.

For more detailed info see https://github.com/libusb/libusb/issues/1156

mvp, 2022-07-19:
2 weeks later, I have not received any response from Apple for FB10563007. @seanm, @mcuee, @hjelmn: can someone submit perhaps duplicate report to Apple? We need to get their devs attention somehow :-(

MPLew-is, 2022-07-20:
[This seems to be fixed on macOS 12.5](https://github.com/mvp/uhubctl/issues/419#issuecomment-1190813421).

mvp, 2022-07-20:
@MPLew-is, are you checking this on x86 Mac? Asking because it was always working on M1 (ARM) Macs

MPLew-is, 2022-07-20:
Yep, see the linked issue in `uhubctl` - I'm using the exact same computer (x86 2020 MacBook Air) and hubs as a few weeks ago, and it's all back up and running again with the latest update.

mvp, 2022-07-20:
I can indeed confirm that it works on MacOS 12.5 on x86 Mac:
```
$ sw_vers
ProductName:	macOS
ProductVersion:	12.5
BuildVersion:	21G72
```
 
[image]

listdevs also works correctly (shows device path longer than 1): 
```examples/listdevs | grep 0bda
0bda:5411 (bus 20, device 8) path: 3.3.4
0bda:5411 (bus 20, device 7) path: 3.3.3
0bda:5411 (bus 20, device 6) path: 3.4
0bda:5411 (bus 20, device 4) path: 3.3
0bda:5411 (bus 20, device 3) path: 3
0bda:0411 (bus 0, device 5) path: 1.3.4
0bda:0411 (bus 0, device 4) path: 1.3.3
0bda:0411 (bus 0, device 3) path: 1.4
0bda:0411 (bus 0, device 2) path: 1.3
0bda:0411 (bus 0, device 1) path: 1
```

mvp, 2022-07-20:
Fixed in MacOS 12.5. No action necessary on libusb side, except for perhaps mentioning this in README that MacOS 12.4 has this bug that affects libusb.

closed 2022-07-20 by mvp

mcuee, 2022-07-21:
> Fixed in MacOS 12.5. No action necessary on libusb side, except for perhaps mentioning this in README that MacOS 12.4 has this bug that affects libusb.

This is great. Thanks for the updates.

seanm, 2022-08-10:
>2 weeks later, I have not received any response from Apple for FB10563007.

Almost fell of my chair laughing here, thanks!  :)  You're lucky to get a reply from Apple in 2 years, never mind 2 weeks!  They read them, but they rarely communicate back.