libusb/libusb #1157: How to open a device after libusb_handle_events

zundap1970, 2022-06-23:
Hi folks,

I use the "hotplugtest.c" example to detect when a USB is connected or disconnected.
libusb_handle_events API is blocking and return is case of connected\disconnected the device, now :

1.  How can discover which type of event ?
2. In case of device connection how to know which file under /dev/ should have to open? 
Thanks

mcuee, 2022-09-05:
The question does not seem to be clear. Please state what you want to do. 
What do you mean by "libusb_handle_events API is blocking and return is case of connected\disconnected the device"?

mcuee, 2022-09-05:
It seems to me libusb is not the right tool for you for your purpose, rather libudev may be the one.

closed 2022-10-22 by mcuee