libusb/libusb #1161: Why does libusb lose packets when reading data on the Linux platform.

zhousongs, 2022-06-27:
Hello, I am using libusb1.0.26; Integrated into androidstudio, and then read the data returned by the device on android7.0 device; The device reports an average of 8 packets of 256 bytes of data per millisecond.

However, during the actual test, it is found that there will be a packet loss rate of 0.001 to 0.005 (each packet of data reported by the equipment has a packet number); How to solve this problem;

The following methods are called when I read：

 int  nRet =libusb_interrupt_transfer(pHandle, 0x81, (unsigned char *)sBuf, sizeof(sBuf), &nActualBytes, 0);

mcuee, 2022-06-27:
Try using async transfer to see if that help.
https://libusb.sourceforge.io/api-1.0/libusb_io.html

mcuee, 2022-06-27:
The other question, do you have the same issue for a normal Linux machine? libusb on Android may not be as good as libusb under Linux.

zhousongs, 2022-06-27:
hello .  By the way, what other conditions will affect libusb1.0.26 to read data;

zhousongs, 2022-06-27:
For example, the CPU cannot exceed

zhousongs, 2022-06-27:
For example, the number of CPUs exceeded will affect this;

zhousongs, 2022-06-27:
For example, how much CPU size will affect this?

mcuee, 2022-06-27:
Sorry but you need to answer the question first. It could be your Android phone is the problem, or it could be the device. You can try to generate the debug log by running your application with LIBUSB_DEBUG=4.

zhousongs, 2022-06-27:
Hardware devices do not lose packets on Windows platform; On the Android phone, I changed to a different phone, and the final results were similar; There are a few packet losses

zhousongs, 2022-06-27:
Do you mean that the size of CPU does not affect the ability of libusb1.0.26 to read data?

diabolo38, 2022-06-27:
unless the end point is iso whaterver is the os number of cpu  you should not loose a single packet
 or somehow the os or the usb layer of the device f/w m driver is doing something bad by nature usb  bulk is 100% reliable
the androids/w stack is usualy  slow so as already suggested use assync api and queue enough (and big) request to overcome it , that"s all you can do.

mcuee, 2022-09-05:
Close this for now. Please reopen when you have the relevant info to show there is a potential issue with libusb.

kuro68k, 2023-08-07:
I have this same issue. Raspbian running on a Pi 4. I am using async I/O and double buffering, calling `libusb_handle_events_timeout_completed()` regularly.

The packet loss happens under high system load. Doing some experiments, it seems that if you don't call `libusb_handle_events_timeout_completed()` often enough, data is lost. For reasons I have yet to determine, calling it inside some loops causes the application to hang.

The test firmware I have implements a USB CDC (virtual serial) port running at Full Speed (12Mbps). On Windows there is no problem just reading about 62k of data per second, sent in one big chunk, via the COM port using libserial. On Linux the virtual TTY doesn't buffer very much and data is often lost. I tried using libusb to read the bulk endpoint directly, and it is more reliable but still suffers from lost packets.

I'm looking at ways to resolve it. It may be more of a Linux issue than a libusb issue, as on Windows it seems to be more reliable. That could be down to the Windows machine being a lot faster though.

tormodvolden, 2023-08-12:
@kuro68k Will you be able to share your firmware and host code? I am interested in reproducing this on an RPi 4.

mcuee, 2023-08-12:
@kuro68k 

I can reopen the issue if you provide more information.

kuro68k, 2023-08-12:
Unfortunately the firmware is proprietary, but I might be able to produce a demo firmware for you.

If you have an STM32 Nucleo board it's just a case of using CubeMX to create a USB CDC device and sending 62k of data every second. I can build it if you tell me what board you have.

I think I've found a reliable way to reproduce the issue too. Just build a kernel and it usually fails within 5 minutes.

mcuee, 2023-08-14:
@kuro68k 

I have reopend the issue. Please post your codes and debug log. Thanks.