libusb/libusb #1163: Windows: libusb_get_device_list deadlock with 8Bitdo Pro2 controller

Megamouse, 2022-07-07:
When I attach an 8Bitdo Pro2 BT Controller via USB and activate the XInput mode (it can do 4 different modes) then I am stuck in libusb_get_device_list.
I don't have this issue with any other controller. The controller works fine over Bluetooth.

I'm on commit 4239bc3 - libusb 1.0.26

here's a backtrace:
[image]

Megamouse, 2022-07-08:
I'm on windows 11.

I hope this is the correct log
<details>
  <summary>Log with controller attached</summary>

  ```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001265] [0000d0ac] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.001303] [0000d0ac] libusb: debug [usbi_add_event_source] add HANDLE 00000000000009E8 events 0
[ 0.001331] [0000d0ac] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001354] [0000d0ac] libusb: debug [usbi_add_event_source] add HANDLE 00000000000009E4 events 0
[ 0.001394] [0000d0ac] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001419] [0000d0ac] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003035] [0000d0ac] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.003107] [0000d0ac] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.003882] [0000d0ac] libusb: info [windows_init] UsbDk backend is not available
[ 0.003945] [0000d0ac] libusb: warning [libusb_init] installing new context as implicit default
[ 0.004211] [0000d0ac] libusb: debug [libusb_get_device_list]
[ 0.004298] [0000c3fc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.016240] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [A]
[ 0.016341] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [B]
[ 0.016428] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [C]
[ 0.016511] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [D]
[ 0.016858] [0000d0ac] libusb: debug [get_api_type] driver(s): mt7612US
[ 0.016935] [0000d0ac] libusb: debug [get_api_type] upper filter driver(s): xboxgip
[ 0.017009] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [16]
[ 0.017178] [0000d0ac] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.017240] [0000d0ac] libusb: debug [get_api_type] lower filter driver(s): bcbtums
[ 0.017563] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [36]
[ 0.017986] [0000d0ac] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.018103] [0000d0ac] libusb: debug [get_api_type] matched driver name against HID API
[ 0.018337] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [69]
[ 0.018636] [0000d0ac] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.018835] [0000d0ac] libusb: debug [get_api_type] upper filter driver(s): dc3d
[ 0.019074] [0000d0ac] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.019317] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.019635] [0000d0ac] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019827] [0000d0ac] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.020051] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [6B]
[ 0.020371] [0000d0ac] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.020562] [0000d0ac] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.020795] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [6C]
[ 0.021101] [0000d0ac] libusb: debug [get_api_type] driver(s): xusb22
[ 0.021311] [0000d0ac] libusb: debug [winusb_get_device_list] allocating new device for session [6D]
[ 0.021685] [0000d0ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_10DE&DEV_1ADA&SUBSYS_21723842&REV_A1\4&30760922&0&0208' bus number 1
[ 0.021827] [0000d0ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_06ED&SUBSYS_06ED1849&REV_00\3&11583659&0&A0' bus number 2
[ 0.022622] [0000d0ac] libusb: debug [winusb_get_device_list] found existing device for session [36]
[ 0.022701] [0000d0ac] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.022901] [0000d0ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 218 bytes)
[ 0.023171] [0000d0ac] libusb: debug [init_device] (bus: 2, addr: 8, depth: 2, port: 2): 'USB\VID_050D&PID_065A\5CF37072019B'
[ 0.023454] [0000d0ac] libusb: debug [winusb_get_device_list] found existing device for session [6B]
[ 0.023691] [0000d0ac] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.023904] [0000d0ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 2469 bytes)
[ 0.024152] [0000d0ac] libusb: debug [init_device] (bus: 2, addr: 7, depth: 2, port: 1): 'USB\VID_046D&PID_0825\8172D510'
[ 0.024528] [0000d0ac] libusb: debug [winusb_get_device_list] found existing device for session [6C]
  ```
</details>

<details>
  <summary>Log without controller attached</summary>
  
  ```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000219] [0000ca34] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000245] [0000ca34] libusb: debug [usbi_add_event_source] add HANDLE 00000000000009F0 events 0
[ 0.000267] [0000ca34] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000286] [0000ca34] libusb: debug [usbi_add_event_source] add HANDLE 00000000000009F4 events 0
[... 267 lines cut ...]
[16.700070] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEA8F0
[... cut ...]
[16.705789] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEBB00
[16.706153] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEB780
[16.706510] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEBE80
[16.706872] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEB080
[16.707237] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEC200
[16.707696] [0000ca34] libusb: debug [libusb_free_transfer] transfer 0000000026AEBCC0
  ```
  </details>

mcuee, 2022-07-09:
Does usbview or usb deview tree viewer see your device in this mode?

USBView (FTDI version)
https://ftdichip.com/utilities/

USB Device Tree Viewer:
https://www.uwe-sieber.de/usbtreeview_e.html

mcuee, 2022-07-09:
@Megamouse 
Please try the following pull request as well to see if it helps.
* https://github.com/libusb/libusb/pull/1127

Megamouse, 2022-07-09:
> Does usbview or usb deview tree viewer see your device in this mode?
> 
> USBView (FTDI version) https://ftdichip.com/utilities/
> 
> USB Device Tree Viewer: https://www.uwe-sieber.de/usbtreeview_e.html

They both stop responding as long es the controller is plugged in, and work again as soon as i unplug it

mcuee, 2022-07-09:
In that case, the device FW is the problem.

closed 2022-07-09 by mcuee

Megamouse, 2022-07-09:
At first glance the PR seems to work after it freezes a couple of seconds.
So at least the infinite deadlock is fixed.

I can't tell if the other devices still work as intended though, since I never really use the libusb stuff in our application.

I've attached the logs to the other comment.

Megamouse, 2022-07-09:
> In that case, the device FW is the problem. Report the issue to the vendor.

I mean, you should still be able to handle shitty drivers in your software...

Megamouse, 2022-07-09:
Anyway, I reported it to 8BitDo as well.

mcuee, 2022-07-09:
> > In that case, the device FW is the problem. Report the issue to the vendor.
> 
> I mean, you should still be able to handle shitty drivers in your software...

Yes we will try if there is a solution which does not affect normal devices. 

I have set this to the the duplicate of #1136.

mcuee, 2022-07-09:
@Megamouse 
I will suggest you post your comments in #1127.
* #1127