libusb/libusb #1165: USB hub reference is not decremented when a handle to a connected device is closed.

Oldanko, 2022-07-14:
I have written a program to output the number of references to connected USB devices to see how the number changes after different USB events. As a result, I've observed that when I `libusb_open` a device that is connected to the USB hub, the number of references to that device increases by one, as well as the number of references to the USB hub. However, when I `libusb_close` the device, the number of references to the device decreases by one, but the number of references to the USB hub stays the same.

In order to write this program, as I couldn't find a function to get the number of references in libusb, I've added a function with the following definition:
`
void LIBUSB_CALL libusb_refcnt_device(libusb_device *dev, long* refcnt)
{
    *refcnt = usbi_atomic_load(&dev->refcnt);
}
`

Following is the C++ code of the program:
`
#include <iostream>
#include <vector>
#include "libusb/libusb.h"

void print_ref_cnts(libusb_context *context) {
    libusb_device **devs;
    auto devs_size = libusb_get_device_list(context, &devs);

    for(ssize_t i = 0; i < devs_size; i++) {
        auto p_device = devs[i];
        libusb_device_descriptor descriptor{};
        libusb_get_device_descriptor(p_device, &descriptor);

        long refs;
        libusb_refcnt_device(p_device, &refs);

        std::cout << std::hex
                  << " 0x" << descriptor.idVendor
                  << " 0x" << descriptor.idProduct
                  << std::dec
                  << " " << refs << std::endl;
    }

    libusb_free_device_list(devs, true);
}

auto open_devices(libusb_context *context, unsigned short vendor_id, unsigned short product_id) {
    libusb_device **devs;
    auto devs_size = libusb_get_device_list(context, &devs);

    std::vector<libusb_device_handle*> handles;

    for(ssize_t i = 0; i < devs_size; i++) {
        libusb_device_descriptor descriptor{};
        libusb_get_device_descriptor(devs[i], &descriptor);
        if(descriptor.idVendor == vendor_id && descriptor.idProduct == product_id) {
            libusb_device_handle* dh;
            libusb_open(devs[i], &dh);
            handles.push_back(dh);
        }
    }

    libusb_free_device_list(devs, true);
    return handles;
}

int main() {
    libusb_context *context;
    libusb_init(&context);

    std::string s;
    while(std::getline(std::cin, s)) {
        std::cout << "Before opening\n";
        print_ref_cnts(context);
        auto handles = open_devices(context, 0x9da, 0x9090);
        std::cout << "After opening\n";
        print_ref_cnts(context);
        for(auto dh : handles)
            libusb_close(dh);
    }

    libusb_exit(context);
}
`

The program performs the following actions in a loop:
1. Read a line of std::cin input
2. Outpust vendor id, product id, and the number of references of every item in the list of connected devices.
3. Attempts to open devices that match given vendor id and product id.
4. Outpust vendor id, product id, and the number of references of every item in the list of connected devices again.
5. Closes any opened devices in step 3

While the program waits for the std::cin input, I performed the following actions, each having a corresponding entry in the resulting console output:
1. Disconnected the USB hub and a mouse with the vendor id and the product id which are being searched for.
2. Connect the USB hub
3. Connect the mouse to the hub
4. No changes

The program produced the following output:
`
Nothing connected
Before opening
 0x8086 0x9d2f 4
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
After opening
 0x8086 0x9d2f 4
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
Hub Connected
Before opening
 0x8086 0x9d2f 5
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 2
 0x639 0x7210 1
After opening
 0x8086 0x9d2f 5
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 2
 0x639 0x7210 1
Mouse connected
Before opening
 0x8086 0x9d2f 5
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 3
 0x639 0x7210 1
 0x9da 0x9090 1
After opening
 0x8086 0x9d2f 6
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 4
 0x639 0x7210 1
 0x9da 0x9090 2
No changes
Before opening
 0x8086 0x9d2f 7
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 4
 0x639 0x7210 1
 0x9da 0x9090 1
After opening
 0x8086 0x9d2f 9
 0xbda 0xb00b 1
 0x9da 0xfa10 1
 0x4ca 0x706e 1
 0x1a40 0x101 5
 0x639 0x7210 1
 0x9da 0x9090 2
`
From the logs we can conclude, that the hub VID=0x1a40 and PID=0x101, mouse VID=0x9da PID=0x9090. Before the mouse is connected (Hub Connected), the hub's reference count is 2 (presumably itself and the 0x639 0x7210?). After the mouse is connected but before creating a handle to it (Mouse connected - Before opening), the hub's reference count is incremented, and the mouse's reference count is set to 1. After opening the mouse (Mouse connected - After opening), both the hub's and the mouse's reference counts are incremented. However, after the mouse's device handle is closed (No changes - Before opening), the mouse's reference count is decremented, but the hub's reference count stays the same. You can also see an increasing reference count of the 0x8086 0x9d2f.

I hope I haven't made any mistakes with this. This looks like incorrect behavior to me, but I'm not certain about it.

mcuee, 2022-07-16:
Which OS are you testing this?

mcuee, 2023-03-27:
@Oldanko 

Any updates on this?

tormodvolden, 2023-03-30:
No, I don't think such a function should be added. An application shouldn't need to query these values. Hopefully the function is only needed here for debugging this issue in particular. Otherwise if this would be a more widespread problem we should provide necessary info in the debug log.

mcuee, 2023-03-31:
> No, I don't think such a function should be added. An application shouldn't need to query these values. Hopefully the function is only needed here for debugging this issue in particular. Otherwise if this would be a more widespread problem we should provide necessary info in the debug log.

I agree with you that we do not need a new function here.

On the other hand, I think it is good to add some debug info for reference count. I am trying to debug the following issue but it seems to be difficult without the debug log.
* #988

mcuee, 2023-03-31:
With the above patch, here is the debug output under Windows. From the run log it seems to me the reference count handling is working. Many of the clean-ups are carried out just before `libusb_exit`.

`05E3:0612` is an external USB Type C Hub (SuperSpeed).
`0781:5591` is a Sandisk USB drive connected to the Type C Hub.
`8086:A0ED` and `8086:9A13` are the two root hubs.
`05E3:0610` is an internal full speed USB hub.
`046d:c528` is Logitech USB receiver
`8087:0026` is Intel USB Bluetooth controller
`17CA:0575` is Egistec Touch Fingerprint sensor

```
MINGW64 /c/work/libusb/libusb_test/libusb_issue1165
$ ./examples/testlibusb.exe
Dev (bus 2, device 2): 05E3 - 0612 speed: 5G
Dev (bus 2, device 3): 0781 - 5591 speed: 5G
Dev (bus 1, device 2): 8087 - 0026 speed: 12M
Dev (bus 1, device 5): 04F2 - B6DD speed: 480M
Dev (bus 1, device 3): 1C7A - 0575 speed: 480M
Dev (bus 1, device 0): 8086 - A0ED speed: 5G
Dev (bus 1, device 7): 05E3 - 0610 speed: 480M
Dev (bus 1, device 1): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 0): 8086 - 9A13 speed: 5G

$ export LIBUSB_DEBUG=4

$ ./examples/testlibusb.exe
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001153] [00002648] libusb: debug [libusb_init_context] created default context
[ 0.001447] [00002648] libusb: debug [libusb_init_context] libusb v1.0.26.22791
[ 0.001766] [00002648] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b0 events 0
[ 0.001901] [00002648] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.002019] [00002648] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000b4 events 0
[ 0.002151] [00002648] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.002283] [00002648] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003816] [00002648] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005348] [00002648] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007095] [00002648] libusb: info [windows_init] UsbDk backend is not available
[ 0.007290] [00002648] libusb: debug [libusb_get_device_list]
[ 0.007438] [000035e0] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.014352] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.014652] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.014773] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.014910] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [8]
[ 0.015369] [00002648] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.015533] [00002648] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.015639] [00002648] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.015754] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.015926] [00002648] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016047] [00002648] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.016149] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [3E]
[ 0.016421] [00002648] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016543] [00002648] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016659] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [70]
[ 0.016779] [00002648] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016888] [00002648] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016997] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [71]
[ 0.017142] [00002648] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.017265] [00002648] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.017517] [00002648] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.017613] [00002648] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.017662] [00002648] libusb: debug [libusb_unref_device] Device 1.0 reference count decreased to 1
[ 0.017714] [00002648] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.017752] [00002648] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.017792] [00002648] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 1
[ 0.018683] [00002648] libusb: debug [libusb_ref_device] Device 2.0 reference count increased to 2
[ 0.018866] [00002648] libusb: debug [libusb_ref_device] Device 0.0 reference count increased to 2
[ 0.018961] [00002648] libusb: debug [winusb_get_device_list] found existing device for session [5]
[... 332 lines cut ...]
[ 0.058001] [00002648] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 1
[ 0.058056] [00002648] libusb: debug [libusb_unref_device] Device 1.2 reference count decreased to 0
[ 0.058104] [00002648] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.058150] [00002648] libusb: debug [libusb_unref_device] Device 1.0 reference count decreased to 5
[... cut ...]
[ 0.058874] [00002648] libusb: debug [libusb_unref_device] Device 2.0 reference count decreased to 0
[ 0.058922] [00002648] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.058971] [00002648] libusb: debug [libusb_exit] destroying default context
[ 0.059026] [000035e0] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.059449] [00002648] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b4
[ 0.059511] [00002648] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000b0
```

mcuee, 2023-04-02:
More debug logs are here, for Linux, macOS and Windows. It seems to me that libusb has no issues. It is just the cleanup is at the end. So if you print the reference count in the middle, it may seem to have an issue.
* https://github.com/libusb/libusb/issues/988

@Oldanko 
Please post the debug log to see if there is a real issue or not. Thanks. You may want to try my patch as well to print the increase/decrease of the reference count in detail.

wbnov, 2023-05-25:
I can see this problem on Windows: I poll the device list frequently (all 100-500 ms) to detect new devices. When a device is opened and is closed after polling the device list in between then the reference count of the hubs is not decremented correctly and we get the error "device still referenced" on libusb_close.

`// LibusbTest.cpp : 

#include <iostream>
#include <stdio.h>
#include "libusb.h"

void* my_device = NULL;
libusb_context* libusbContext = nullptr;
libusb_device** devList = nullptr;

void LIBUSB_CALL libusb_log_callback( libusb_context* ctx, enum libusb_log_level level, const char* str )
{
    if (ctx!=NULL )
        return;
    std::cout << str;
}

int poll_devices( int* devCount)
{
    *devCount = libusb_get_device_list( libusbContext, &devList );
    if ( *devCount <= 0 )
    {
        libusb_exit( libusbContext );
        return -1;
    }

    void* help;
    int my_cnt = 0;
    unsigned char dummy_str[255];

    for ( int i = 0; i < *devCount; ++i )
    {
        libusb_device_descriptor desc;
        libusb_device_handle* handle = nullptr;
        libusb_get_device_descriptor( devList[i], &desc );

        if ( (my_device == NULL) && (i>0) )
        {
            // we have found our device and we open it now. for real usage we are looking for vendor and product.
            if ( libusb_open( devList[i], &handle ) != LIBUSB_SUCCESS )
                continue;
            my_device = handle;
            std::cout << "DEVICE_" << i << ":0x" << std::hex << handle << " V:" << std::hex << desc.idVendor << "\r\n";
        }
    }
    libusb_free_device_list( devList, 1 );
    return 0;
}

int main()
{
    if ( libusb_init( &libusbContext ) != LIBUSB_SUCCESS )
        return -1;

    libusb_set_log_cb( libusbContext, libusb_log_callback, LIBUSB_LOG_CB_CONTEXT );  // LIBUSB_LOG_CB_GLOBAL |
    libusb_set_option( libusbContext, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG );

    int devCount = 0;
    for ( int i = 0; i<10; i++ )
    {
        int error = poll_devices( &devCount );
        if ( error < 0 )
            return -1;
        Sleep( 100 );
    }
  
    if ( my_device )
    {
        std::cout << "close my device";
        libusb_close( (libusb_device_handle*)my_device);
    }

    libusb_exit( libusbContext );
}
`

mcuee, 2023-05-31:
@bohdan-tymkiv 

I can reproduce your issue (`device still referenced` warning messages`) with your test code under Windows 10 and latest git.

But if I carry out a minor change and it will be okay under Windows. I tend to think your test code is not right since `libusb_open` and `libusb_close` have to match.

```
$ diff -u libusbtest_1165.cpp libusbtest_1165mod.cpp
--- libusbtest_1165.cpp 2023-05-31 11:30:16.564270800 +0800
+++ libusbtest_1165mod.cpp      2023-05-31 13:56:09.879722500 +0800
@@ -41,6 +41,8 @@
                 continue;
             my_device = handle;
             std::cout << "DEVICE_" << i << ":0x" << std::hex << handle << " V:" << std::hex << desc.idVendor << "\r\n";
+                       std::cout << "close my device";
+                       libusb_close( (libusb_device_handle*)my_device);
         }
     }
     libusb_free_device_list( devList, 1 );
@@ -64,12 +66,6 @@
             return -1;
         Sleep( 100 );
     }
-
-    if ( my_device )
-    {
-        std::cout << "close my device";
-        libusb_close( (libusb_device_handle*)my_device);
-    }

     libusb_exit( libusbContext );
 }
```

mcuee, 2023-05-31:
@wbnov 

Please try my modification to see it works for you or not. Thanks.

wbnov, 2023-05-31:
The purpose of this program is to keep my devices open to work with them and close them after work is done. That is normally after days or month.
So I use frequently polling for devices with my vendor and product id with libusb_get_device_list. If a new device that I need is found it is opened (libusb_open) and is kept open until my program is closed. And that is outside the loop of detecting devices and hence after libusb_free_device_list(). So I cannot use that modification.

mcuee, 2023-05-31:
> The purpose of this program is to keep my devices open to work with them and close them after work is done. That is normally after days or month. So I use frequently polling for devices with my [...]

I see, I believe your issue will be sorted out if hotplig is supported under Windows. Unfortunately due to lack of Windows developers, hotplug will not be ready for Windows any time soon.
* https://github.com/libusb/libusb/issues/86

Sorry but I can not offer more help to you in this case. You may have to live with the limitations for now.

@tormodvolden 
Just wonder if you have some other suggestions.

wbnov, 2023-07-21:
@mcuee 
For now I have added some code to free all remaining devices on libusb_exit. So there are no remaining memory leaks.
I think this could help many other users because polling the device list and then using dedicated devices is commonly used.
Maybe you could add this repairing to the repository.

	usbi_hotplug_exit( _ctx );
	usbi_io_exit( _ctx );

	/* now free all remaining devices
	 *   because of a bug in libusb some hubs remain referenced  */
	for (int cnt = 0; cnt < 100; cnt++)
	{
		if (list_empty( &(ctx)->usb_devs ))
			break;
		for_each_device( _ctx, dev ) {
			usbi_warn( _ctx, "device %d.%d still referenced - free here",
				dev->bus_number, dev->device_address );
			if (usbi_backend.destroy_device)
				usbi_backend.destroy_device( dev );
			list_del( &dev->list );
			DEVICE_CTX( dev ) = NULL;
			free( dev );
			break;
		}
	}

mcuee, 2023-07-26:
> Maybe you could add this repairing to the repository.

@wbnov 

Please create a PR. Thanks.

mcuee, 2023-10-09:
@tormodvolden 

How do you like the ideas mentioned by @wbnov? Thanks.

minicarpet, 2025-02-26:
I reproduce this issue on my side, could we merge the solution ?

seanm, 2025-02-26:
> I reproduce this issue on my side, could we merge the solution ?

There is nothing to merge that I can see.... no one created any PR it seems...

minicarpet, 2025-02-26:
I saw that windows hotplug implementation were mentionned, my bad.
I guess a quick solution would to add a function to return referenced object so user can use `libusb_unref_device` to unref all still referenced device

minicarpet, 2025-08-25:
After some test, I can reproduce it even without opening a device but only by referencing one:

[test_warn.log](https://github.com/user-attachments/files/21978079/test_warn.log)
[test_warn_2.log](https://github.com/user-attachments/files/21978080/test_warn_2.log)

in first case I ref a device so when freeing the list, hub 1.0 is not destroyed and the problem appears if I call twice `libusb_get_device_list`

(see the testlibusb.c from you exemple to reproduce)