libusb/libusb #1166: Reminder: Return Windows r1.0.27 distributions [and later] to Traditional Packaging Structure as used in r1.0.25 and below.

TheOriginalFreddoFrog, 2022-07-16:
Hi,

Just a reminder - from earlier comms - re the distribution/packaging format for subsequent releases: LibUSB needs to revert back to the package distribution format of Version 1.0.25 and below. 

Version 1.0.26 deviated from past distribution structures - making it very difficult for distributors other software kits to pull and integrate releases hence improvements from your distribution site into our work.

Thanks

FF.

tormodvolden, 2022-08-03:
Maybe you can help making the (appveyor) CI build lay out the wanted structure also?

VK3SIR, 2022-08-12:
Speaking for people in our project - we would be more than happy to assist where we can and to throw this to the team behind us.

Please provide me with and/or point us to project documentation and this can be facilitated.

We are here to help not hinder or pick to pieces. There was considerable discussion the first time that we raised this; we on the team that works with the JTSDK (that heavily relies on your software) cannot use the v1.0.26 release nor can we write software that auto-updates our work (and that of other associated projects) unless we have consistency of nomenclature and file distribution.

As a reminder, the distribution structure for v 1.0.25 (consistent with many-many previous releases) is:

C:.
│   libusb-1.0.def
│   README.txt
│
├───examples
│   ├───bin32
│   │       dpfp.exe
│   │       dpfp_threaded.exe
│   │       fxload.exe
│   │       listdevs.exe
│   │       sam3u_benchmark.exe
│   │       stress.exe
│   │       testlibusb.exe
│   │       xusb.exe
│   │
│   ├───bin64
│   │       dpfp.exe
│   │       dpfp_threaded.exe
│   │       fxload.exe
│   │       listdevs.exe
│   │       sam3u_benchmark.exe
│   │       stress.exe
│   │       testlibusb.exe
│   │       xusb.exe
│   │
│   └───source
│           dpfp.c
│           ezusb.c
│           ezusb.h
│           fxload.c
│           listdevs.c
│           sam3u_benchmark.c
│           testlibusb.c
│           xusb.c
│
├───include
│   └───libusb-1.0
│           libusb.h
│
├───MinGW32
│   ├───dll
│   │       libusb-1.0.dll
│   │       libusb-1.0.dll.a
│   │
│   └───static
│           libusb-1.0.a
│           libusb-1.0.dll.a
│
├───MinGW64
│   ├───dll
│   │       libusb-1.0.dll
│   │       libusb-1.0.dll.a
│   │
│   └───static
│           libusb-1.0.a
│           libusb-1.0.dll.a
│
├───VS2013
│   ├───MS32
│   │   ├───Debug
│   │   │   ├───dll
│   │   │   │   │   libusb-1.0.dll
[... 909 lines cut ...]

Regarding "msys-usb-1.0.dll" in 1.0.26 ... This is what we primarily use. It needs to be distributed as libusb-1.0.dll. Why has the nomenclature here changed and/or become inconsistent?

For our purposes (and for what was legacy) This is the main structure that WE NEED TO BE RETURNED TO for our work:

├───include
│   └───libusb-1.0
│           libusb.h
│
│
...
├───MinGW64
│   ├───dll
│   │       libusb-1.0.dll
│   │       libusb-1.0.dll.a
│   │
│   └───static
│           libusb-1.0.a
│           libusb-1.0.dll.a

If you look at https://hamlib-sdk.sourceforge.io you will see what we produce. There are 30-40,000 users of software produced under our kits at any one time (see https://pskreporter.info and search for WSJT-X and JTDX)

If you post proposed sources and/or links to our sources to our support forum (referred to in those links above) before release then we can easily make this happen. 

We support work that people with Nobel prizes produce (i.e. WSJT-X) - so having the expertise should be no issue if it is your wish for any of us to assist.

tikonen, 2023-11-18:
I understand that the binary distribution is not officially supported and more of a courtesy. It might be good to remark this in README of the binary distribution package to set the expectations. Most people just follow the download link and are left wondering what is going on, like happened to me. Or then just flat out dont provide binaries?

mcuee, 2023-11-20:
> I understand that the binary distribution is not officially supported and more of a courtesy. It might be good to remark this in README of the binary distribution package to set the expectations. [...]

We will still provide Windows binaries.

I have added a note here:
https://github.com/libusb/libusb/releases/tag/v1.0.26
Note: libusb-1.0.26-binaries.7z is provided as is and may not work for your toolchain. In that case, please build from the source using your toolchain.

tormodvolden, 2023-12-10:
I have uploaded a repacked libusb-1.0.26.7z file, and also for the 1.0.27 RC today, a libusb-1.0.27-rc1.7z.

closed 2023-12-10 by tormodvolden

hamlibdk, 2023-12-11:
More than happy with the release structure - returned to a Version 1.0.25 style structure :-) Well done, and thanks :-)

tormodvolden, 2023-12-11:
Well, it is more like 1.0.24, thus without the Debug builds, so there is no explicit Release folder. Note also 1.0.23 was a bit different, without the "VSxxxx" top folder.

rom1v, 2024-02-29:
Just for information, the structure of 1.0.26 was very convenient for me (it had exactly the expected structure for meson and had a pkgconfig file so everything worked automatically). With 1.0.27, I will have to write a script to recreate the structure of 1.0.26.

Refs #1468

rom1v, 2024-03-01:
For illustration, here is the script I adapted to recreate the expected directory structure and the pkgconfig file from the 1.0.27 release: https://github.com/Genymobile/scrcpy/commit/upgrade_libusb_1_0_27/

tormodvolden, 2024-03-01:
We are shipping binaries only for people who are unable to build themselves. If you already have a build setup, you should just get the source release tarball and build it. "make install" will install the pkgconfig files as well.

TheOriginalFreddoFrog, 2024-03-09:
Romain,

Thanks for the reply. Yet when they changed the structure they caused considerable issues for many of us that also support legacy versions of libusb.

We should just be able to automatically pull whatever configured, compiled version from a repo; we should not have to recompile it (which we have to do now). It should just drop-in – work.

This we know can be a difficult ask.

That was the point of the request.

Regards,

FF

TheOriginalFreddoFrog, 2024-03-09:
Tormold,

Consistent structuring, with a “drop-in” methodology for compatibility, is essential in this age.

Licensing is the main reason one tries, where possible, to pull from “official sources”. LibUSB at this time does not have a restrictive policy; yet one never knows when it can change and when laws of whatever nation can supersede release policies.

“I got this directly from The Internet automatically” is by far the safest approach across nearly every nation on this planet (where my end-points are utilised as I help maintain a SDK).

When one pulls from the “official release” one has consistency.  Distributed sources work at release time. Compilers and libraries change an evolve …

All your work and efforts are heavily appreciated. The point of the original post was to help those of us that have to maintain compatibility with old sources and legalities – current and potential. Consistency makes life simpler.

Regards,

FF.

tormodvolden, 2024-03-09:
> recompile it (which we have to do now).

BTW, why do you have to do that now?

tormodvolden, 2024-03-09:
@TheOriginalFreddoFrog This is so wrong at so many accounts that I don't know where to start.

libusb is released as a source tarball. The official release is the source code. As said before, the binaries are only a service to _end users_ who are not able to build it themselves, but can upgrade via a drop-in binary replacement if their applications load and link libusb dynamically. Building something on Windows at least used to be difficult, that's why we have been offering Windows binaries.

If you are distributing libusb with other software you should build libusb with the same toolchain as the other software. Period.

libusb is distributed under a GPL license. It is not easier to redistribute the binaries than the source, legal-wise. Do you have examples of legislation where downloading sources and building them would be illegal whereas downloading binaries would not be?

I can see the point that a modern toolchain could fail to build an old version of the sources, due to stricter checks etc. I haven't seen with libusb though. I can build e.g. v1.0.0 from 2008 on my current system without issues. But why would you want to build with an old version of libusb in the first place, since we put a lot of effort in keeping the library backwards compatible?

VK3SIR, 2024-03-18:
Tormold,

You have lost the point which is drop-in-compatibility (from a central, working, consistent source).

The issue is forward compatibility for what may come with what already exists.

Consistency.

Please read-back over what we have sent in the past. We in what we provide have a complex if not knife-edge that we walk. Then that does not also factor in if someone takes you to task for violating their IP (which is why one should always draw from a consistent source).

No point heading any further on this discussion as the point has been made clearly 😊

Regards,

Steve I (The Original Freddo Frog) !