libusb/libusb #1171: how can I get a 64-bit arm libusb-1.0.so

Dangal2018, 2022-08-11:
Hi,

Dangal2018, 2022-08-11:
I serach my problems in Stackoverflow , and then I try to install libusb-1.0-0-dev:amd64 , and I find libusb-1.0.so.0.1.0 in my /usr/lib/x86_64-linux-gnu , but I execute the file command ,It shows

mcuee, 2022-08-12:
Please use libusb-devel mailing list for technical support. Thanks. gitup is for issue report.
https://github.com/libusb/libusb/wiki#Support

closed 2022-08-12 by mcuee

Dangal2018, 2022-08-16:
thanks. @mcuee 
and I solved ... 
just let the host=aarch64-linux-gnu,to be frankly,**It needs match the prefix of CC and C++ completely.**

r4raatz, 2023-05-20:
Hi @Dangal2018, Can you please direct me to the solution since I'm also in need of libusb-1.0.so for my arm64-v8a android project?? even a plain .so file is fine for me or help me to build one??