libusb/libusb #1172: libusb_open returned -12 on windows

seago66, 2022-08-15:
Dear all , 

I have tried winusb driver and libusb driver on a usb device , when I use libusb to open this device with api libusb_open_device_with_vid_pid , it reported the following error : 
[ 0.032302] [00003b78] libusb: debug [set_composite_interface] interface[0] = \\.\USB#VID_18D1&PID_4EE7&MI_00#6&16B8D2A9&0&0000#{F72FE0D4-CBCB-407D-8814-9ED673D0DD6B}
[ 0.032302] [00003b78] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.032302] [00003b78] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.032302] [00003b78] libusb: debug [windows_get_device_list] setting composite interface for [F3]:
[ 0.033300] [00003b78] libusb: debug [set_composite_interface] interface[1] = \\.\USB#VID_18D1&PID_4EE7&MI_01#6&16B8D2A9&0&0001#{F72FE0D4-CBCB-407D-8814-9ED673D0DD6B}
[ 0.033300] [00003b78] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.033300] [00003b78] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.033300] [00003b78] libusb: debug [libusb_get_device_descriptor]
[ 0.033300] [00003b78] libusb: debug [libusb_get_device_descriptor]
[ 0.033300] [00003b78] libusb: debug [libusb_get_device_descriptor]
[ 0.033300] [00003b78] libusb: debug [libusb_open] open 1.2
// reported error here
[ 0.033300] [00003b78] libusb: debug [unsupported_open] unsupported API call for 'open' (unrecognized device driver)
[ 0.034297] [00003b78] libusb: debug [libusb_open] open 1.2 returns -12
[ 0.034297] [00003b78] libusb: debug [libusb_unref_device] destroy device 1.4

How to deal with this issue ?

Thanks !

mcuee, 2022-08-15:
How did you install the device driver? Did you use Zadig? Did Windows Device Manager show that your device is using WinUSB or libusb-win32 driver (libusb0.sys)?

Ref: https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

Is your device using IAD? If yes, you may want to try the pull request here. It is a known limitation of the current Windows libusb backend. You may have to use WinUSB as the driver of the whole device (replace Windows USB Composite Device driver with WinUSB) if you do not want to use the pull request.
https://github.com/libusb/libusb/pull/965

seago66, 2022-08-15:
@mcuee 
I am using Zadig to intall the device driver , and  Windows Device Manager shows ok for both devices. I tried using WinUsb as the driver of the whole devices , the "unsupported API call for 'open'" error disappeared , but it returned NULL for the open api .
Anyway,  thanks for your reply , it helped me a lot , I will study further to find out why libusb_open_device_with_vid_pid  returns NULL .

Thanks

mcuee, 2022-08-16:
Close this for now. Please reopen if you find issues with libusb Windows backend. Thanks.

closed 2022-08-16 by mcuee