libusb/libusb #1173: WinUSB 10.x support on Windows 11...

CartBlanche, 2022-08-16:
Does LibUSB support WinUSB 10.0.x driver on Windows 11?

mcuee, 2022-08-17:
No issues with libusb Windows backend on Windows 11.

mcuee, 2022-08-17:
Do you have any specific issues? If yes, please state the issue. Thanks.

mcuee, 2022-08-19:
Close for now. Please reopen if you encounter issues.

closed 2022-08-19 by mcuee

mcuee, 2023-05-24:
> xusb.exe can not open this device, any idea?

Your device is already using WinUSB and there is no need to use Zadig to replace the driver.
What is the output of xusb?

zoumi, 2023-05-24:
Here is the debug log:
```
D:\tools\examples>xusb -d 0123:5755
Using libusb v1.0.26.11717

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000205] [00004808] libusb: debug [libusb_init] created default context
[ 0.000326] [00004808] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000422] [00004808] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000476] [00004808] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000527] [00004808] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000596] [00004808] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000645] [00004808] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010273] [00004808] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.018443] [00004808] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.023776] [00004808] libusb: info [windows_init] UsbDk backend is not available
Opening device 0123:5755...
[ 0.024452] [00004808] libusb: debug [libusb_get_device_list]
[ 0.026879] [00002288] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.037704] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [68]
[ 0.037915] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [79]
[ 0.038254] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [7A]
[ 0.038546] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [7B]
[ 0.039120] [00004808] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.039272] [00004808] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.039492] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [13]
[ 0.040088] [00004808] libusb: debug [get_api_type] driver(s): RTSUER
[ 0.040188] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.040728] [00004808] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.040827] [00004808] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.041046] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [65]
[ 0.041558] [00004808] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.041660] [00004808] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.041882] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [66]
[ 0.042394] [00004808] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.042494] [00004808] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.042711] [00004808] libusb: debug [winusb_get_device_list] allocating new device for session [67]
[ 0.043138] [00004808] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1022&DEV_1639&SUBSYS_109F1D05&REV_00\4&2C93F132&0&0341' bus number 1
[ 0.043310] [00004808] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'ROOT\USB\0000' bus number 2
[ 0.043537] [00004808] libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ROOT\USB\0000'
[ 0.043827] [00004808] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1022&DEV_1639&SUBSYS_109F1D05&REV_00\4&2C93F132&0&0441' bus number 3
[ 0.044931] [00004808] libusb: debug [winusb_get_device_list] found existing device for session [68]
[ 0.045088] [00004808] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.045246] [00004808] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.045496] [00004808] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 4): 'USB\VID_05E3&PID_0608\6&38788AD6&0&4'
[ 0.046124] [00004808] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\5&22FFD888&1&0' reports 6 ports
[ 0.046183] [00004808] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\5&22FFD888&1&0'
[ 0.046519] [00004808] libusb: debug [winusb_get_device_list] found existing device for session [67]
[ 0.046743] [00004808] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.046907] [00004808] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 98 bytes)
[ 0.047157] [00004808] libusb: debug [init_device] (bus: 3, addr: 1, depth: 1, port: 2): 'USB\VID_0123&PID_5755\205832735734'
[ 0.047502] [00004808] libusb: debug [winusb_get_device_list] found existing device for session [66]
[ 0.047724] [00004808] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047927] [00004808] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 751 bytes)
[ 0.048171] [00004808] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 3): 'USB\VID_04F2&PID_B65A\0001'
[ 0.048520] [00004808] libusb: debug [winusb_get_device_list] extra GUID: {B987F121-43D2-43AC-95BF-CEB14212DB6F}
[ 0.048659] [00004808] libusb: debug [winusb_get_device_list] found existing device for session [13]
[ 0.049143] [00004808] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049202] [00004808] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[... cut ...]
[ 0.063185] [00004808] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.063422] [00004808] libusb: debug [libusb_unref_device] destroy device 1.0
  Failed.
[ 0.063909] [00004808] libusb: debug [libusb_exit] destroying default context
[ 0.064171] [00002288] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.065507] [00004808] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.065590] [00004808] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
```

mcuee, 2023-05-24:
@zoumi 

Your issue is different from the orginal report. Please raise a new issue and tell us more about this device. Is this device developed by your or your company? It may be that the FW is faulty. To check that, please post the output of the device descriptor details (USBview output or USBTreeView output under Windows, or `lsusb -vvv` output under Linux. Please also mention the Windows version and USB chipset.

USBView
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview

USBTreeView
https://www.uwe-sieber.de/usbtreeview_e.html

zoumi, 2023-05-24:
Thank you! It's solved now.It's seems that my device has two interfaces. But I had set only one WCID. After removing another interface everything works. 
Thank you again!
Here is the very useful link:
https://github.com/pbatard/libwdi/wiki/WCID-Devices

mcuee, 2023-05-24:
Glad that you sorted out the issue and thanks for reporting back the issue.