libusb/libusb #1178: libusb_interrupt_transfer() Blocks Until All `length` Bytes Received

ewhac, 2022-08-24:
The documentation for `libusb_interrupt_transfer()` states the following:

> For interrupt reads, the `length` field indicates the maximum length of data you are expecting to receive. If less data arrives than expected, this function will return that data, so be sure to check the `transferred` output parameter.

In my mind, the implication of this is that, the moment *any* data arrives before the timeout expires, `libusb_interrupt_transfer()` will return success and fill in `transferred` with the quantity of bytes actually received.

This is not what happens.  Consider the following pseudo-code:

```c
	/*  sizeof (struct dev_input_report) < sizeof (readbuf)  */
	struct dev_input_report *rep;
	int retval, retsiz;
	uint8_t readbuf[256];

	retval = libusb_interrupt_transfer (g_handle, LIBUSB_ENDPOINT_IN | DEV_ENDPOINT,
	                                    readbuf, sizeof (readbuf),
	                                    &retsiz, 0);
	if (!retval  &&  retsiz == sizeof (struct dev_input_report)) {
		rep = (struct dev_input_report *) readbuf;
		handle_report (rep);
	}
```

Based on the docs, the expectation is that `libusb_interrupt_transfer()` will block forever (`timeout` = 0) until something arrives.  The first `transferred` bytes of `readbuf` will be filled with the report, and the function returns.

What actually happens is that `libusb_interrupt_transfer()` blocks until `length` bytes have been collected, which may require multiple reports -- not what you want when dealing with an interactive input device.

So either I'm misunderstanding something, or it would appear the docs are incorrect.

tormodvolden, 2022-08-29:
Please provide debug logs.

ewhac, 2022-09-01:
Is output from `LIBUSB_DEBUG=4` sufficient?

tormodvolden, 2022-09-01:
Yes, that is good, max output.

ewhac, 2022-09-02:
Test program.  This program attempts to locate a Logitech G13 "Gameboard" and get events from it.  Compile with `gcc libusbtest.c $(pkg-config --cflags --libs libusb-1.0)`

```C
#include <fcntl.h>
#include <linux/input.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <err.h>

#include <libusb.h>

#define	USB_VENDOR_ID_LOGITECH		0x046d
#define	USB_DEVICE_ID_LOGITECH_G13	(0xc21c)

#define	G13_ENDPOINT_KEY		1
#define	G13_ENDPOINT_LCD		2
#define	G13_REPORT_SIZE_KEY		8
#define	G13_REPORT_SIZE_LCD		0x3c0

struct g13_input_report {
	uint8_t	report_id;
	uint8_t	joy_x, joy_y;
	uint8_t	keybits[5];
};

libusb_context			*g_usbctx;
struct libusb_device_handle	*g_handle;

int
connect_g13 (void)
{
	libusb_device	**usbdevs;
	ssize_t		ndevs;
	int		retval;

	retval = libusb_init (&g_usbctx);
	if (retval < 0) {
		warn ("libusb init failed");
		return -1;
	}
	ndevs = libusb_get_device_list (g_usbctx, &usbdevs);
	if (ndevs < 0) {
		warn ("libusb_get_device_list failed");
		return -1;
	}

	for (int i = 0;  i < ndevs;  ++i) {
		struct libusb_device_descriptor desc;

		retval = libusb_get_device_descriptor (usbdevs[i], &desc);
		if (retval < 0) {
			warn ("failed to get descriptor %d", i);
			return -1;
		}
		printf ("Checking index %d:\n", i);
		printf ("   idVendor  = 0x%04hx\n", desc.idVendor);
		printf ("   idProduct = 0x%04hx\n", desc.idProduct);
		printf ("   bcdDevice = 0x%04hx\n", desc.bcdDevice);
		if (    desc.idVendor == USB_VENDOR_ID_LOGITECH
		    &&  desc.idProduct == USB_DEVICE_ID_LOGITECH_G13)
		{
			retval = libusb_open (usbdevs[i], &g_handle);
			if (retval < 0) {
				warn ("usb device open failed");
				return -1;
			}

			/*
[... 332 lines cut ...]
[ 0.013782] [000c280c] libusb: debug [libusb_get_device_descriptor]  
Checking index 21:
   idVendor  = 0x1d6b
   idProduct = 0x0002
   bcdDevice = 0x0518
[ 0.013801] [000c280c] libusb: debug [libusb_submit_transfer] transfer 0x563aaa6ce420
[ 0.013809] [000c280c] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 256
[ 0.013829] [000c280c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.013835] [000c280c] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.013843] [000c280c] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ ==== Press/release key on G13 16 times here ==== ]
[ 7.643531] [000c280c] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 7.643565] [000c280c] libusb: debug [reap_for_handle] urb type=1 status=0 transferred=256
[ 7.643575] [000c280c] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 7.643584] [000c280c] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[ 7.643590] [000c280c] libusb: debug [usbi_handle_transfer_completion] transfer 0x563aaa6ce420 has callback 0x7fa53bda89d0
[ 7.643597] [000c280c] libusb: debug [sync_transfer_cb] actual_length=256
[ 7.643613] [000c280c] libusb: debug [libusb_free_transfer] transfer 0x563aaa6ce420
nbytes: 256, keybits = [ 00 00 82 00 80 ]
[ 7.643636] [000c280c] libusb: debug [libusb_submit_transfer] transfer 0x563aaa6e1850
[ 7.643644] [000c280c] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 256
[ 7.643665] [000c280c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.643674] [000c280c] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
^C
```

tormodvolden, 2022-09-03:
If you are working with a HID device, maybe you should consider using HIDAPI instead:
https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices
I don't know if that would solve this issue though.

ewhac, 2022-09-04:
The HID descriptor reported by the G13 is essentially meaningless, so it has to be parsed "by hand:"

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x00,        // Usage (0x00)
0xA1, 0x01,        // Collection (Application)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x09, 0x01,        //   Usage (0x01)
0x85, 0x01,        //   Report ID (1)
0x95, 0x07,        //   Report Count (7)
0x75, 0x08,        //   Report Size (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x03,        //   Report ID (3)
0x09, 0x02,        //   Usage (0x02)
0x96, 0xDF, 0x03,  //   Report Count (991)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x07,        //   Report ID (7)
0x09, 0x03,        //   Usage (0x03)
0x95, 0x04,        //   Report Count (4)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x04,        //   Report ID (4)
0x09, 0x04,        //   Usage (0x04)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x05,        //   Report ID (5)
0x09, 0x05,        //   Usage (0x05)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x06,        //   Report ID (6)
0x09, 0x06,        //   Usage (0x06)
0x96, 0x01, 0x01,  //   Report Count (257)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 61 bytes
```