libusb/libusb #1179: Segfault with libusbk driver

smaract-svonohr, 2022-08-26:
We've had issues previously with libusbk and libusb v1.0.24, but these have been fixed in #943. However with libusb v1.0.26 we still observe crashes, just much more rarely. The crash is observed in this line: https://github.com/libusb/libusb/blob/ff8fe9397d4c61f072d209703f854c9f62ed17e3/libusb/os/windows_common.c#L485-L486 with the message:
> Exception thrown: read access violation.
> dev_handle->**dev** was nullptr.

Disabling debugging completely by undefining `ENABLE_LOGGING` seems to fix the issue, so it's actually printing this debug message which causes the crash. I'm not very familar with the libusb code but I belive there is a race condition. Since during this debug message there are no locks held, the device was probably freed by some other thread.

I've attached a debug log leading up to the crash. Not sure if it's helpfull, but you can see just before the log stops this message is printed a lot of times. I can't provide an exact guide how to trigger this bug. It just happens with one specific PC and our software using libusb. If you need any additional information let me know.

[libusb.log](https://github.com/libusb/libusb/files/9432104/libusb.log)

mcuee, 2022-08-26:
> It just happens with one specific PC and our software using libusb. If you need any additional information let me know.

What is the OS and USB chipset in question for this PC? Does the issue happen with WinUSB driver? If it does not happen with WinUSB driver, you may consider switching to WinUSB if that is possible.

smaract-svonohr, 2022-08-26:
The OS is: Windows 10 Enterprise 2016 LTSB: NT10.0 Build 14393.1358 Version 1607 SP0
The USB chipset is the onboard USB controller provided by the Intel Skylake Chipset. There is also an additional Renesas Electronics USB 3.0 Host Controller, but the device in question is connected to the Intel Controller.

The device using the libusbk driver is actually not our device, but from a third party (MATRIX VISION GmbH mvBlueFOX3 Device), so I'm not sure if other drivers work. Our device uses the WinUSB driver. During device enumeration our software may try to communitcate with the libusbk device, which seems enough to trigger the bug.

mcuee, 2022-08-26:
@smaract-svonohr 
You can always use Zadig to switch the driver to WinUSB and try to see if the results are different or not. Thanks.

Zadig: https://zadig.akeo.ie/

smaract-svonohr, 2022-08-26:
Yes, I might try that later. There are probably other workarounds, I'm fine with disabling logging for now. Nevertheless, I think there is a bug in libusb and it should be fixed.

mcuee, 2022-08-26:
@smaract-svonohr 
The reason to test for WinUSB is to see if the issue is only specific to libusbk or it is also with WinUSB. If it is because of logging, then it may have the same issue with WinUSB as well. Thanks.

tormodvolden, 2022-08-29:
I think this debug print should be before the usbi_mutex_unlock(), because afterwards the device may have been closed. What do you think @dmitry-zakablukov ?

smaract-svonohr, 2022-08-30:
Is it even guaranteed, that the `completion_key` corresponds to a `libusb_device_handle`? Or that the device is still open, once the `GetQueuedCompletionStatus` returns? Afaik closing a device does not wait for transfers to finish. The easiest fix is probably to remove the bus_number and device_address from the log message.

tormodvolden, 2022-08-30:
True, and I think your suggestion is good. (I had looked at how the loop was going through open devices, but this debug print is of course in the !found case so that was not relevant.)

smaract-svonohr, 2022-08-30:
I just took a look again at the log I posted above and found this interesting part:
```
[19.102320] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E410 for handle 00000173403347D0 (device 2.1)
[19.102323] [000020dc] libusb: debug [libusb_close]  
[19.102324] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E410 for handle 00000173403347D0 (device 2.1)
[19.102326] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E410 for handle 00000173403347D0 (device 2.1)
[19.102328] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E410 for handle 00000173403347D0 (device 2.1)
[19.102330] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E4F0 for handle 00000173403347D0 (device 2.1)
[19.102331] [000020dc] libusb: debug [libusb_open] open 2.6
[19.102332] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E4F0 for handle 00000173403347D0 (device 2.1)
[19.102334] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E4F0 for handle 00000173403347D0 (device 2.6)
[19.102336] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E4F0 for handle 00000173403347D0 (device 2.6)
[19.102337] [00001b9c] libusb: debug [windows_iocp_thread] ignoring overlapped 0000000EF091E4F0 for handle 00000173403347D0 (device 2.6)
```
The handle always stays the same, but after closing and opening a different device the device_address suddenly changes. The memory was probably reused from the previous closed device. This shows that the bus_number and device_address is not accurate, unless we actually match a transfer to the overlapped field. But then, there won't be any log message of course.

dmitry-zakablukov, 2022-08-30:
> I think this debug print should be before the usbi_mutex_unlock(), because afterwards the device may have been closed. What do you think @dmitry-zakablukov ?

Yes, this might help, but I would suggest to store `dev_handle->dev->bus_number` and `dev_handle->dev->device_address` in variables under the lock rather than logging under the lock.

smaract-svonohr, 2022-08-30:
As stated, I think even under the lock the information may not be reliable. But at least it won't crash. If you think the additional information adds value to the debug message, I'm also fine with this solution.

dmitry-zakablukov, 2022-08-30:
> As stated, I think even under the lock the information may not be reliable.

Then I agree to remove the bus_number and device_address from the log message.

smaract-svonohr, 2022-09-01:
Thanks for the fix. The issue is gone with these changes.

commit 687bf21a (2022-09-03): windows: Avoid invalid dereference in debug print

The device structure is not always valid at this point.

Fixes #1179
Closes #1186
closed 2022-09-01 by smaract-svonohr