libusb/libusb #1180: USB Control transfer retrieving interface descriptor

symdeb, 2022-08-26:
Just a question. Does the library support control transfer to retrieve interface descriptors ? I have a case for a project where a call not seem to reach the SETUP stage code for an embedded controller firmware:

Unbunt 22.04.01 
installed  with sudo apt get install libusb-1.0-0-dev  
gcc  C program to create a control transfer.

The device is just plugged in. Linux did set the address (0x05) and configure (0x09) and retrieved the string descriptors successfully and install the device. 

Then using libusb the following call using libusb is issued:
int ret = libusb_control_transfer(handle,0x81,0x06,0x2601,1,data,5,0);
Libusb returns an error -1: Input/Output Error

Using this change to a custom device descriptor request does reach the firmware:
int ret = libusb_control_transfer(handle,0x80,0x06,0x2601,1,data,5,0);

Trying to figure out how to further nail down if this is related to libusb or the firmware.
The firmware code is quite complex and have not yet found out where this gets stuck

Further on, used wireshark to check if usb transfers occur.
When using 0x80 the descriptor request shows up (even though it invalid) 
Using 0x81 there is nothing shown in wireshark at all.

Really curious now if this might be an issue in the libus of I am doingseomthing wrong. 
According to the USB spec, interface descriptors can be retrieved in the configured state.

[image]

tormodvolden, 2022-08-29:
Yes, normally you can use libusb to craft your own control requests. If you suspect a bug in libusb, please attach a dump from `lsusb -v -d VID:PID` and a log from running your program with the environment variable `LIBUSB_DEBUG=4`

Typically control requests go to the default endpoint 0 (which all devices must have), thus 0x80 (IN) or 0x00 (OUT).

BTW, there is also libusb_get_descriptor()

mcuee, 2022-09-05:
@symdeb 
Closing this for now as it does not seem to be an issue with libusb. Please reopen if you find the issue. Thanks.

closed 2022-09-05 by mcuee

symdeb, 2022-10-04:
Looked at get_descriptor. 
The desired request to make is:   libusb_control_transfer(handle,0x81,0x06,0x2601,1,data,5,0);
How to do accomplish this with get_descriptor ?
1.  The bmRequest type is  0x81. Though get_descriptor always uses 0x80
2. The Index required is 1 (interface 1), but get_descriptor puts desc_index in the lower byte of wValue, thus wIndex is always 0
3. wValue should be 0x26 
Example: 
libusb_get_descriptor(handle, 0x26, 2, data, 5  );
The device receives:
req->bmRequest:80 req->bRequest:06 req->wValue:2602 , req->wIndex:0000, req->wLength:0005