libusb/libusb #1184: Error claming Interface on Ubuntu 22.04 LTS

greenscreenflicker, 2022-08-30:
I copied the following example.
https://raw.githubusercontent.com/tytouf/libusb-cdc-example/master/cdc_example.c

And i get the error: 
Error claiming interface: LIBUSB_ERROR_NOT_FOUND

Please advice how I can get this fixed.

Issue seems similar to #1153.

tormodvolden, 2022-08-31:
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2022-08-31:
You have to set up the right permission and please do not open multiple github issues on the same question. I will close this one and you can ask further questions in https://github.com/libusb/libusb/issues/1185.

mcuee, 2022-08-31:
@greenscreenflicker 

Please read the FAQ here as well.
https://github.com/libusb/libusb/wiki/FAQ#running-libusb

closed 2022-08-31 by mcuee