libusb/libusb #1185: Not enough user privilages to open usb.

greenscreenflicker, 2022-08-30:
I copied the following code:
https://raw.githubusercontent.com/tytouf/libusb-cdc-example/master/cdc_example.c

I cannot execute it as a default user (Error Error finding USB device vid 1a86 pid 7523), but I can execute it as root.

I'm already in the right groups.

> michael : michael adm tty dialout cdrom sudo dip plugdev lpadmin lxd sambashare

and it's in the group dialout.

tormodvolden, 2022-08-31:
Please check if libusb_detach_kernel_driver() is successful, maybe it fails unless you are root. Your normal user, being in the right groups, have full access to the device node. However, unbinding the kernel driver from the device might require elevated privileges.

In general, we can only help if you provide full debug logs.

mcuee, 2022-08-31:
@greenscreenflicker 

Please checkout the FAQ here as well.
https://github.com/libusb/libusb/wiki/FAQ#running-libusb

greenscreenflicker, 2022-08-31:
@tormodvolden @mcuee 
I run the following function and code. 
> void usb_connect_to_cdc_with_pid_vid(uint16_t vid,uint16_t pid){
https://gitlab.com/greenscreenflicker/mcu_tracer2_pc/-/blob/usb/usb/usb_connection.c

Lets start as a normal user:
I get the following output.

michael@aorus:~/git/mcu_tracer2_pc$ ./mcu_tracer 
lib usb init successfull
Error finding USB device vid 1a86 pid 7523

So there seems the issue that I cannot connect to the device.  
libusb_detach_kernel_driver() is not even called.

Lets run it in root.
>lib usb init successfull
>round 0
>libusb_kernel_driver_active: Not active
>round 1
>libusb_kernel_driver_active: Not active
>Error claiming interface: LIBUSB_ERROR_NOT_FOUND

So it seems like the interface did find the thing now. (But some microseconds before it recognized it...)

tormodvolden, 2022-09-01:
Note there is also libusb_set_auto_detach_kernel_driver() that you can try, instead of coding your own logic.

greenscreenflicker, 2022-09-01:
yes, i'll try that. but i guess that won't be the issue. It's copied.
https://raw.githubusercontent.com/tytouf/libusb-cdc-example/master/cdc_example.c

Is there any working example of lib usb with cdc?

tormodvolden, 2022-09-01:
libusb is intended for custom USB class protocols. CDC is a standard protocol, well supported by all operating systems, which take care of everything and present the user with e.g. /dev/ttyACM0 or COM5. Why do you want to access a CDC device with libusb?

Conversely, if you make a USB device and want to use it with a custom driver, do not declare it as a standard (e.g. CDC) class.

mcuee, 2022-09-05:
Close this for now. This is not a libusb issue after all.

closed 2022-09-05 by mcuee