libusb/libusb #1190: Bridged network adapter not working?

hdfsyu, 2022-09-02:
When I try to change the network adapter type to bridged, it fails to find the IP address (sometimes it can find the IP but not the DNS)
I know i'm using the right network adapter (i checked using `ifconfig -a`) this happens on both windows and linux (and i'm trying to bridge a Wi-Fi adapter)

mcuee, 2022-09-02:
Why do you think this has anything to do with libusb? What exactly are you trying to do?

hdfsyu, 2022-09-02:
@mcuee I am trying to connect to the internet using a bridged adapter so I can use online multiplayer, but it doesn't detect the ip address, sometimes it does but the DNS cannot be found

mcuee, 2022-09-02:
Close this issue as it has nothing to do with libusb.

closed 2022-09-02 by mcuee

hdfsyu, 2022-09-02:
oh sorry, i accidentally went into the wrong repository? i was supposed to put this in another repo... sorry about that