libusb/libusb #1192: To support MSYS2 clang/clang64 toolchain environment

mcuee, 2022-09-04:
As of now, libusb build will fail with MSYS2 clang and clang64 toolchain.
* https://github.com/libusb/libusb/pull/1152#issuecomment-1236231482

The following quick fix will address the issue but it is better to fix it properly.
``` sed -e "s/-mwin32//g" -e "s/-Wl,--add-stdcall-alias//g" -i configure.ac```

tormodvolden, 2022-09-04:
MinGW ld --help says:
`  --add-stdcall-alias                Export symbols with and without @nn`
How is the export on clang64 without this option? Is another option needed there?

tormodvolden, 2022-09-04:
The `-mwin32` was added in commit da5df37c without explicit commenting on it.

I wonder if it is only needed on Cygwin, as e.g. done here: https://git.furworks.de/opensourcemirror/mpv/commit/58bf7fd1f31f185f8ab89102e8173901ce0f22b5

mcuee, 2022-09-04:
Interestingly CMake does not seem to have this issue. I will check what compiler flag CMake uses for MSY2 clang and clang64.

mcuee, 2022-09-04:
Probably it will be good to support more MSYS2 environments (including CI).
https://www.msys2.org/docs/environments/

	Name	Prefix	Toolchain	Architecture	C Library	C++ Library
msys	MSYS	/usr	gcc	x86_64	cygwin	libstdc++
mingw64	MINGW64	/mingw64	gcc	x86_64	msvcrt	libstdc++
ucrt64	UCRT64	/ucrt64	gcc	x86_64	ucrt	libstdc++
clang64	CLANG64	/clang64	llvm	x86_64	ucrt	libc++
mingw32	MINGW32	/mingw32	gcc	i686	msvcrt	libstdc++
clang32	CLANG32	/clang32	llvm	i686	ucrt	libc++
clangarm64	CLANGARM64	/clangarm64	llvm	aarch64	ucrt	libc++

mcuee, 2022-09-04:
> The `-mwin32` was added in commit [da5df37](https://github.com/libusb/libusb/commit/da5df37c4d1a2841c6f9cdc626386d096f5f312d) without explicitly commenting on it.
> 
> It looks like it is for having _WIN32 defined: https://www.appsloveworld.com/cplus/100/1297/glew-apientry-warnings-and-possibly-related-errors-on-msys2
> 
> From a comment removed, it looks like _WIN32 is not defined by Cygwin and MSYS so those need it. In another project it is only added for Cygwin (whereas `-mwindows` is added for MinGW) : https://github.com/mpv-player/mpv/blob/master/waftools/detections/compiler.py#L72
> 
> We have a few places probing for `_WIN32` || `__CYGWIN__` but I don't know why PLATFORM_WINDOWS is not used instead. The mentioned commit left it this way.

I am not so sure if this is a legacy left from @pbatard's old multilib toolchain.
 Ref: https://github.com/libusb/libusb/blob/master/.private/bm.sh

tormodvolden, 2022-09-04:
Is -mwin32 needed on MSYS/gcc? If so, I can add probing for it in configure.ac. Otherwise we may delete it. FWIW it is not needed when crossbuilding on Linux using MinGW.

mcuee, 2022-09-04:
> Is -mwin32 really needed on MSYS/gcc? If so, I can add probing for it in configure.ac. Otherwise we may delete it. FWIW it is not needed when crossbuilding on Linux using MinGW.

It is not needed for MSYS/gcc.

tormodvolden, 2022-09-04:
> It is not needed for normal MSYS2 mingw32/mingw64. I believe it is only needed for cygwin or the multilib gcc.

OK, then I'd suggest removing -mwin32 for mingw/msys and only keep it for cygwin.

> And I believe `-Wl,--add-stdcall-alias` is only needed for clang32 and not clang64. But I am not familiar with clang.

So it is supported by clang32 but not recognized by clang64?

mcuee, 2022-09-04:
> > And I believe `-Wl,--add-stdcall-alias` is only needed for clang32 and not clang64. But I am not familiar with clang.
> 
> So it is supported by clang32 but not recognized by clang64?

It is supported by both.

tormodvolden, 2022-09-04:
So that part in the "quick fix" in OP was not needed?

mcuee, 2022-09-04:
> So that part in the "quick fix" in OP was not needed?

Sorry the "quick fix" is needed for clang32/clang64. Without it the build will fail.
```
lld: error: unknown argument: --add-stdcall-alias

```

mcuee, 2022-09-04:
@tormodvolden 
BTW, you may want to add CI build for MSYS2 clang32/clang64 (not so sure about clangarm64). I have created them in my repo based on msys2.yml.
Ref: https://github.com/mcuee/libusb/tree/master/.github/workflows

tormodvolden, 2022-09-04:
From llvm.org
> LLD is a drop-in replacement for the GNU linkers that accepts the same command line arguments and linker scripts as GNU.

Hmm...

Can you tweak your clang build to use GNU ld instead of LLD?

tormodvolden, 2022-09-04:
PR #1193 seems to build well on clang32/64 CI. Not sure if it is using GNU or LLVM linker.

neheb, 2022-09-05:
msys2 has a libusb package for clangarm64. So it builds but who knows if it actually works.

IIRC, _WIN32 is defined in true windows platforms and WIN32 is defined by fake ones (Cygwin).

mcuee, 2022-09-05:
> msys2 has a libusb package for clangarm64. So it builds but who knows if it actually works.

Hopefully someone with Windows on ARM64 can test it out. 

Apparently libusb works well on Windows on ARM64 (at least the MSVC build) but unfortunately libwdi/Zadig does not work there yet. So only device with existing WinUSB driver will work.
* https://github.com/pbatard/libwdi/issues/155
* https://github.com/libusb/libusb/issues/1046

mcuee, 2022-09-05:
@tormodvolden 

History about the interchangeability of MinGW and MS DLLs, quite a long discussion back in 2010 and commit https://github.com/libusb/libusb/commit/30b42bc6544e2c6a8e1809cb6819f670dc206904 as a result.
https://marc.info/?l=libusb-devel&m=128097247705942&w=2

driver1998, 2022-09-12:
Hi, I have some Windows ARM64 devices (a Snapdragon laptop, and a Raspberry Pi 4), if you want me to test something, please let me know.

driver1998, 2022-09-12:
On my Snapdragon 8cx laptop:

The package from MSYS2 does not include samples, so I built myself with the patches from the [PKGBUILD](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-libusb/PKGBUILD).

This is with a Acer USB-C dock (3x USB 3.0, 1x HDMI), a USB card reader, and a RTL8152 Ethernet dongle pluged in.

```
driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./testlibusb.exe
Dev (bus 1, device 1): 214B - 7250 speed: 480M
Dev (bus 1, device 2): 25A4 - 9311 speed: 12M
  Manufacturer:              USB C
  Product:                   USB C Video Adaptor
Dev (bus 2, device 0): 1D6B - 0004 speed: 5G
Dev (bus 1, device 0): 1D6B - 0004 speed: 5G
Dev (bus 1, device 4): 067B - 2731 speed: 480M
Dev (bus 1, device 3): 0BDA - 8152 speed: 480M
```
```
driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./listdevs.exe
214b:7250 (bus 1, device 1) path: 1
25a4:9311 (bus 1, device 2) path: 1.1
1d6b:0004 (bus 2, device 0)
1d6b:0004 (bus 1, device 0)
067b:2731 (bus 1, device 4) path: 1.2
0bda:8152 (bus 1, device 3) path: 1.4
```
```
driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./xusb -k
Using libusb v1.0.26.11724

Opening device 0204:6025...
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2'
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'URS\VEN_QCOM&DEV_0497&SUBSYS_CLS08180&HOST\3&252BAFA3&0&0'
libusb: info [cache_config_descriptors] could not access configuration descriptor 1 (dummy) for 'USB\VID_0BDA&PID_8152\00E04C360225': [31] 连到系统上的设备没有发挥作用。
  Failed.
```

With the RTL8152 dongle unplugged:

```
driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./xusb -k
Using libusb v1.0.26.11724

Opening device 0204:6025...
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2'
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'URS\VEN_QCOM&DEV_0497&SUBSYS_CLS08180&HOST\3&252BAFA3&0&0'
  Failed.
```

```
driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./sam3u_benchmark.exe
Error finding USB device

driver1998@PRIMUS-QC CLANGARM64 /e/libusb-1.0.26/examples
$ ./dpfp.exe
Could not find/open device
```

mcuee, 2022-09-13:
> On my Snapdragon 8cx laptop:
>
> The package from MSYS2 does not include samples, so I built myself with the patches from the [...]

Thanks a lot for the help.

Since Zadig/libwdi do not work yet under Windows on ARM64, it will be good to have a device with WinUSB driver so that you can test using `xusb` example.

Ref: 
* https://github.com/pbatard/libwdi/issues/155

mcuee, 2022-09-13:
@driver1998 

The other thing you may want to help check is that whether clang32/clang64 port is working or not on Windows on ARM64.

Ref:
* https://github.com/libusb/libusb/pull/1193

commit ec4c3c43 (2023-01-28): windows: Only link with --add-stdcall-alias if supported by linker

E.g. the LLVM LLD doesn't support this option.

Closes #1192
References #944
closed 2023-01-28 by tormodvolden