libusb/libusb #1196: gdb cannot break in main (Cannot access memory) of libusb program?

sdbbs, 2022-09-09:
Posted this originally on https://stackoverflow.com/questions/73664317/gdb-cannot-break-in-main-cannot-access-memory-of-libusb-program, but thought it might be better if it is posted here as well:

Consider this libusb program, which I compile under MINGW64 (part of MSYS2, which is updated as of today) on Windows 10 (also with latest updates as of today):

`libusb-test.c`
```c

// build under MINGW64 on Windows with (assuming mingw64/mingw-w64-x86_64-libusb 1.0.26-1 installed):
// gcc -Wall -g -I/mingw64/include/libusb-1.0 libusb-test.c -o libusb-test.exe -lusb-1.0

#include <inttypes.h> //PRIu64
#include <stdio.h>
#include <string.h>
#include <assert.h>
#include "libusb.h"

int main(int argc, char *argv[]) {

  libusb_device **devs;
  ssize_t cnt;
  int r=0, i;
  struct libusb_device_descriptor desc;

  r = libusb_init(NULL);
  if (r < 0) {
    printf("error: Cannot libusb_init, exiting\r\n");
    return r;
  }

  cnt = libusb_get_device_list(NULL, &devs);
  if (cnt < 0) {
    printf("error: Cannot libusb_get_device_list (Failed to enumerate USB devices), exiting\r\n");
    libusb_exit(NULL);
    return 1;
  }

  for (i = 0; devs[i]; i++) { // or: for (libusb_device **dev = devs; *dev; dev++)
    libusb_device *dev = devs[i];
    libusb_device_handle *handle = NULL;
    printf("Trying device %d: %p\r\n", i, dev);
    int ret = libusb_get_device_descriptor(dev, &desc);
    if (ret) {
      printf("  Failed to read device %d descriptor (%d)\r\n", i, ret);
    } else {
      ret = libusb_open(dev, &handle);
      if (ret) {
        printf("    Failed to open device %d (%d)\r\n", i, ret);
      } else {
        printf(  "    device %d open ( handle %p )\r\n", i, handle);
      }
      if (handle) {
        libusb_close(handle);
        handle = NULL;
      }
    }
  } // end for

  libusb_free_device_list(devs, 1);

  libusb_exit(NULL);
  printf("Program finished; exiting.");
  return r;
}
```

The program, after compiling, actually seems to run fine:

```none
$ ./libusb-test.exe
Trying device 0: 000001eeb9321890
    Failed to open device 0 (-5)
Trying device 1: 000001eeb9320c30
    device 1 open ( handle 000001eeb93242e0 )
...
Trying device 12: 000001eeb9322640
    device 12 open ( handle 000001eeb93242e0 )
Trying device 13: 000001eeb7a7bc50
    Failed to open device 13 (-12)
Program finished; exiting.
```

... however, if I try to debug with `gdb` by breaking into `main` - it fails with "Cannot insert breakpoint":

```none
$ gdb --args ./libusb-test.exe
GNU gdb (GDB) 12.1
...
Reading symbols from ./libusb-test.exe...
(gdb) b main
Breakpoint 1 at 0x140001593: file libusb-test.c, line 11.
(gdb) r
Starting program: C:\msys64\tmp\libusb-test.exe
[New Thread 20144.0x24c0]
[New Thread 20144.0x436c]
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x140001584

Command aborted.
(gdb)
```

Why does this happen - and how can I get `gdb` to break into this program?

mcuee, 2022-09-12:
Sorry but I do not think this has anything to do with libusb.

mcuee, 2022-09-12:
Close this issue based on the Stackoverflow info.
>indeed, it turns out, also the simple "Hello world", without libusb, will not break at main in gdb (same problem "Cannot access memory at address".

closed 2022-09-12 by mcuee