libusb/libusb #1198: MSVC projects use /GL and /LTCG in Release

RazielXYZ, 2022-09-21:
The provided msvc projects (for VS 2019, at least) are using /GL and /LTCG (Whole program optimization and link-time code generation) in release configurations, currently.

This is a bit of an issue for the static libs specifically, as they can not be linked into projects that do not use LTCG, nor can they be linked into projects that are being built with any slightly different cl compiler revision, requiring libusb to be re-built every time.
/GL and /LTCG are also very difficult to override/remove in script-based or automated builds, without modifying the vcxproj files directly.

I would suggest either removing /GL and /LTCG from the relevant projects, or providing separate projects with LTCG for the pipelines that may want to continue using it.

tormodvolden, 2022-11-17:
Is this about latest master? Because the whole msvc tree was reworked in commit aa633da. I cannot find any LTCG in the files.

RazielXYZ, 2022-11-20:
You're right, this was about v1.0.26 (and earlier, presumably). The reworked projects seem to be configured without /GL, so should be fine!

closed 2022-11-20 by RazielXYZ