libusb/libusb #1205: Documentation about caveats of submitting transfers

darkdragon-001, 2022-10-13:
It took my team a month including comparison to kernel CDC-ACM implementation via signal analyzer to find out about the following behavior:
- `libusb_submit_transfer()` only schedules transfers executed in the next frame
- If one wants libusb to continue receiving data even after a transfer has been finished in the current frame, multiple transfers need to be submitted
- It is possible to submit multiple transfers on the same endpoint in the same interface

It would be really helpful if some of this behavior is explained in the documentation for `libusb_submit_transfer()`.

martinling, 2022-10-20:
The fact that it's possible, and indeed advisable, to queue up multiple transfers is something that the documentation could certainly make clearer. Note that there's also some lack of clarity about what is and isn't guaranteed in terms of ordering and concurrency when multiple transfers are queued up on an endpoint - see #1077.

Your other two points though, about the relationship of transfer submission to frames on the wire, are completely dependent on the backend, OS, driver and hardware.

It's not surprising that a call to `libusb_submit_transfer` doesn't get a transfer onto the wire until the next frame, since that call will require libusb to submit an URB to the OS, which has to pass the request to the host controller, and there's delays at each step. The driver and/or host controller hardware probably need to plan a frame in advance, in order to make sure all devices gets polled as needed and allowing for reserved bandwidth for isochronous transfers. So it makes sense that you might never see anything until the frame after you made the submit call.

But there's no fundamental reason that a `libusb_submit_transfer` call *couldn't* result in the host starting the transfer within the current frame, if everything in the stack were able to react quickly enough. And there's no guarantee that the transfer will start in the _next_ frame, as opposed to some later one.

So while these may be useful observations, they're not things you can rely on, and the libusb documentation shouldn't be trying to specify anything about when a transfer will actually hit the wire, because that's almost completely out of libusb's control. It's just a frontend to each operating system's USB APIs, not a low level interface to a host controller.

It's also not guaranteed that transfers will continue back-to-back even if you do queue up multiple transfers. The WinUSB backend will not do this currently. I have just written up PR #1208 to address that, but it requires a special option which restricts the transfer and buffer sizes you can use.

darkdragon-001, 2022-10-20:
@martinling Thanks a lot for the elaborate explanations!

So I guess some hint that submitting multiple transfers on the same endpoint might make sense in order to continuously receive data as soon as it is available (maybe also adding the hint that unknown delays of the driver / host controller to schedule transfers need to be taken into account).