libusb/libusb #1206: Access violation with WinUSB when calling libusb_cancel_transfer

mxk, 2022-10-13:
When a device handle is closed, any outstanding transfers have their [`dev_handle` field set to `NULL`][set-null]. If `libusb_cancel_transfer` is then called on such transfers, `winusb_cancel_transfer` tries to [access `transfer->dev_handle->dev`][access], causing a crash.

I think the solution is either for `libusb_cancel_transfer` to make sure that `dev_handle` is still set, or for `do_close` to set `USBI_TRANSFER_CANCELLING` flag.

Also, will callbacks still be invoked for transfers that had their device handle closed?

[set-null]: https://github.com/libusb/libusb/blob/f3619c4078a921c3d6bf3238ff7a8dc70d60e40e/libusb/core.c#L1423
[access]: https://github.com/libusb/libusb/blob/f3619c4078a921c3d6bf3238ff7a8dc70d60e40e/libusb/os/windows_winusb.c#L2112

mcuee, 2022-10-17:
Are you using synchronous API? If yes, this is the same as #703.

mxk, 2022-10-17:
I don't think so. The synchronous API doesn't support cancellation at all, right?

I'm submitting a transfer and using a dedicated event thread to process callbacks. For my use case, it's fine for cancellation to be asynchronous, but `libusb_cancel_transfer` shouldn't crash, and there needs to be a guarantee that the callback function will be called eventually, even if the underlying device handle is closed while the transfer is in flight.

I worked around this issue for now by ensuring that the device handle outlives any pending transfers.

mcuee, 2022-10-20:
@mxk 

I think @dickens's comment in #297 is valid for async transfer as well.

* https://github.com/libusb/libusb/issues/297#issuecomment-675762505
> Closing a device while there are outstanding transfers is 100% an application error. Handling this gracefully has been discussed in #540, #610 and #703.