libusb/libusb #1209: ld: symbol(s) not found for architecture x86_64

lime-n, 2022-10-19:
I get the following error following example code from the modules documentation:
```
Undefined symbols for architecture x86_64:
  "_libusb_get_device_list", referenced from:
      _main in exercise_1-fb9727.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Here is the script I have been using:
```
#include <libusb-1.0/libusb.h>
#include <stdio.h>
#include <stdlib.h>

int main(){
libusb_device **list;
libusb_device *found = NULL;
ssize_t cnt = libusb_get_device_list(NULL, &list);
ssize_t i = 0;
int err = 0;
if (cnt < 0)
    perror("Some Error");
    exit(1);
 
for (i = 0; i < cnt; i++) {
    libusb_device *device = list[i];
    if (device) {
        found = device;
        break;}}
 
if (found) {
    libusb_device_handle *handle;
    err = libusb_open(found, &handle);
    if (err)
        perror("Another Error");
        exit(1);
        }
 
libusb_free_device_list(list, 1);

return 0;
}
```

mcuee, 2022-10-20:
You may need to provide more context of the question. If you use macOS, I will recommend you to install libusb using Homebrew.

lime-n, 2022-10-20:
Hi @mcuee , I initially installed `libusb` with `homebrew`, then I made sure it linked with the `include`path with `brew unlink libusb && brew link libusb` and moved the `libusb` to the `include` path.

mcuee, 2022-10-21:
And is your mac using ARM processor (M1/M2) or x86_64 processor? You need to install homebrew according to the processor architecture used.

lime-n, 2022-10-21:
Hi @mcuee I am using the x86_64 processor and have no issues with homebrew installations on other libraries, nor with libusb. It seems to install successfully

mcuee, 2022-10-22:
I do not think this is a libusb issue. You may want to ask in libusb-devel mailing list or other place for more help. The github is more for reporting libusb issue and not for technical support.

closed 2022-10-22 by mcuee

mcuee, 2022-10-22:
I just tested it myself and I have no issues building this test program. 
```
mcuee@mcuees-Mac-mini test % cat test1.c
#include <libusb-1.0/libusb.h>
#include <stdio.h>
#include <stdlib.h>

int main(){
libusb_device **list;
libusb_device *found = NULL;
ssize_t cnt = libusb_get_device_list(NULL, &list);
ssize_t i = 0;
int err = 0;
if (cnt < 0)
    perror("Some Error");
    exit(1);

for (i = 0; i < cnt; i++) {
    libusb_device *device = list[i];
    if (device) {
        found = device;
        break;}}

if (found) {
    libusb_device_handle *handle;
    err = libusb_open(found, &handle);
    if (err)
        perror("Another Error");
        exit(1);
        }

libusb_free_device_list(list, 1);

return 0;
}
mcuee@mcuees-Mac-mini test % gcc -o test1 test1.c -lusb-1.0

mcuee@mcuees-Mac-mini test % file test1
test1: Mach-O 64-bit executable x86_64
mcuee@mcuees-Mac-mini test % otool -L test1
test1:
	/src/staging/libusb/macos64/lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
```

```

mcuee, 2022-10-22:
It seems to me you are not using `-lusb-1.0` but `-lusb`.

```

mcuee@mcuees-Mac-mini test % gcc -o test1 test1.c -lusb
ld: warning: ignoring file /usr/local/lib/libusb.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_libusb_get_device_list", referenced from:
      _main in test1-ae484f.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

```

lime-n, 2022-10-25:
Brilliant! this works for me. 
Why would it be using `-lusb` rather than `-lusb-1.0` and is there a way to change the default option? Because it seems my default is `-lusb`

mcuee, 2022-10-25:
> Brilliant! this works for me. Why would it be using `-lusb` rather than `-lusb-1.0` and is there a way to change the default option? Because it seems my default is `-lusb`

Sorry I have no idea what you mean by that. `-lusb` is for libusb-0.1 or libusb-compat-0.1 and not libusb-1.0. You can try to learn to use pkg-config to give you the correct link option.

lime-n, 2022-10-25:
Essentially I run it with:
```
gcc test.c
```
by default this equates to
```
gcc -o test1 test1.c -lusb
```
I will check into the pkg-config option. Thank you.