libusb/libusb #1210: libusb_submit_transfer returns LIBUSB_ERROR_IO

Osurac529, 2022-10-23:
Hi,

I am trying to read audio data from a usb sound card that is connected to a powered usb hub that is connected to a raspberry pi 4 model B. At first I was doing this on android and followed all the android libusb specific things to do, such as, get the file descriptor and pass it through jni to native and all the other related things about that. So the android attempt was failing when calling libusb_submit_transfer returning LIBUSB_ERROR_IO. So I decided to try similar code on a raspberry pi and it fails in the same place. I have been running my program as root (sudo) on raspberry pi but not on android.

I am using libusb-1.0.26 and I built it with:
```
./configure --prefix=/home/pi/audio --disable-udev
make install
```

Here is the standalone code file I am using that should be able to build like this:
```
g++ --std=c++17 -L lib -lusb-1.0 -I include/libusb-1.0  -o main main.cpp
```

```
#include <iostream>
#include <stdexcept>
#include <sstream>
#include <vector>

#include "libusb.h"

template<typename Status>
Status ThrowIfError(const Status status)
{
    if(status < LIBUSB_SUCCESS)
    {
        std::ostringstream stream;
        stream << "status: " << status << " " << libusb_error_name(status);
        throw std::runtime_error(stream.str());
    }

    return status;
}

struct AudioInputInfo
{
    AudioInputInfo(libusb_device* device,
                   const libusb_device_descriptor& device_descriptor,
                   uint8_t interface_number,
                   uint8_t alternate_setting,
                   uint8_t endpoint_address)
        : device(libusb_ref_device(device))
        , interface_number(interface_number)
        , alternate_setting(alternate_setting)
        , endpoint_address(endpoint_address)
    {
        libusb_device_handle* device_handle = nullptr;
        ThrowIfError(libusb_open(device, &device_handle));

        unsigned char string[256];
        int string_length = 0;
        
        ThrowIfError(string_length = libusb_get_string_descriptor_ascii(device_handle, 
                                                        device_descriptor.iManufacturer, 
                                                        string, 
                                                        sizeof(string)));
        manufacturer.assign(reinterpret_cast<char*>(string), string_length);
        
        ThrowIfError(string_length = libusb_get_string_descriptor_ascii(device_handle, 
                                                        device_descriptor.iProduct, 
                                                        string, 
                                                        sizeof(string)));
        product.assign(reinterpret_cast<char*>(string), string_length);
        
        ThrowIfError(string_length = libusb_get_string_descriptor_ascii(device_handle,
                                                        device_descriptor.iSerialNumber, 
                                                        string, 
                                                        sizeof(string)));
[... 646 lines cut ...]
endpoint address: 131
[ 0.025455] [000005c8] libusb: debug [libusb_open] open 1.5
[ 0.025500] [000005c8] libusb: debug [usbi_add_pollfd] add fd 9 events 4
[ 0.025515] [000005c8] libusb: debug [libusb_get_device_descriptor] 
[ 0.025527] [000005c8] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.025545] [000005c8] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.025560] [000005c8] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.025577] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 0
[ 0.025592] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 1
[ 0.025606] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 1
[ 0.025619] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 2
[ 0.025632] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 2
[ 0.025645] [000005c8] libusb: debug [libusb_kernel_driver_active] interface 3
[ 0.025661] [000005c8] libusb: debug [libusb_claim_interface] interface 2
[ 0.025812] [000005c8] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.025829] [000005c8] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.025847] [000005c8] libusb: debug [libusb_alloc_transfer] transfer 0x845b40
submit transfer before...
[ 0.025878] [000005c8] libusb: debug [libusb_submit_transfer] transfer 0x845b40
[ 0.025891] [000005c8] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 2080
[ 0.025907] [000005c8] libusb: error [submit_iso_transfer] submiturb failed error -1 errno=2
[ 0.025920] [000005c8] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
terminate called after throwing an instance of 'std::runtime_error'
  what():  status: -1 LIBUSB_ERROR_IO
```

mcuee, 2022-10-26:
Can you test this on a normal x86_64 Linux machine? isoc transfer is tricky.

Osurac529, 2022-10-27:
I ran the same code on ubuntu and get the same result, here is the debug log output from libusb. I'm guessing I am doing something wrong, any hints?

Built:
```
./configure --prefix=/home/mike/usbaudio/libusb --disable-udev
make install
g++ --std=c++17 -I libusb/include/libusb-1.0 -o usbaudio.o -c usbaudio.cpp 
g++ -L libusb/lib -lusb-1.0 -o usbaudio usbaudio.o -lusb-1.0
```

Ran as sudo because otherwise get permission denied:
```
sudo ./usbaudio
```
uname -a:
```
Linux sidewinder 5.15.0-46-generic #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

log output:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.005955] [0001e352] libusb: debug [libusb_get_device_list] 
[ 0.005978] [0001e352] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.005982] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.005985] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.005989] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.005992] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.005995] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.005998] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.006001] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.006003] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.006006] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.006009] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.006011] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.006014] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.006017] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.006019] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.006022] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.006025] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.006028] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.006031] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.006034] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.006041] [0001e352] libusb: debug [libusb_open] open 1.16
[ 0.006053] [0001e352] libusb: debug [usbi_add_pollfd] add fd 9 events 4
[ 0.006057] [0001e352] libusb: debug [libusb_alloc_transfer] transfer 0x55954ae780c0
[ 0.006060] [0001e352] libusb: debug [libusb_submit_transfer] transfer 0x55954ae780c0
[ 0.006063] [0001e352] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006078] [0001e352] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006082] [0001e352] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.006086] [0001e352] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.006321] [0001e352] libusb: debug [handle_events] poll() returned 1
[ 0.006331] [0001e352] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.006334] [0001e352] libusb: debug [handle_control_completion] handling completion status 0
[ 0.006338] [0001e352] libusb: debug [disarm_timerfd] 
[ 0.006341] [0001e352] libusb: debug [usbi_handle_transfer_completion] transfer 0x55954ae780c0 has callback 0x7f841eba5fc0
[ 0.006344] [0001e352] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.006349] [0001e352] libusb: debug [libusb_free_transfer] transfer 0x55954ae780c0
[ 0.006353] [0001e352] libusb: debug [libusb_alloc_transfer] transfer 0x55954ae81c90
[ 0.006355] [0001e352] libusb: debug [libusb_submit_transfer] transfer 0x55954ae81c90
[ 0.006358] [0001e352] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006368] [0001e352] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006371] [0001e352] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.006617] [0001e352] libusb: debug [handle_events] poll() returned 1
[ 0.006626] [0001e352] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=18
[ 0.006629] [0001e352] libusb: debug [handle_control_completion] handling completion status 0
[ 0.006633] [0001e352] libusb: debug [disarm_timerfd] 
[ 0.006636] [0001e352] libusb: debug [usbi_handle_transfer_completion] transfer 0x55954ae81c90 has callback 0x7f841eba5fc0
[ 0.006639] [0001e352] libusb: debug [sync_transfer_cb] actual_length=18
[... 160 lines cut ...]
endpoint address: 131
[ 0.009646] [0001e352] libusb: debug [libusb_open] open 1.16
[ 0.009657] [0001e352] libusb: debug [usbi_add_pollfd] add fd 9 events 4
[ 0.009660] [0001e352] libusb: debug [libusb_get_device_descriptor] 
[ 0.009663] [0001e352] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.009667] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.009670] [0001e352] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.009673] [0001e352] libusb: debug [libusb_kernel_driver_active] interface 0
[ 0.009677] [0001e352] libusb: debug [libusb_kernel_driver_active] interface 1
[ 0.009681] [0001e352] libusb: debug [libusb_kernel_driver_active] interface 1
[ 0.009684] [0001e352] libusb: debug [libusb_kernel_driver_active] interface 2
[ 0.009687] [0001e352] libusb: debug [libusb_kernel_driver_active] interface 2
[... cut ...]
submit transfer before...
[ 0.009740] [0001e352] libusb: debug [libusb_submit_transfer] transfer 0x55954ae864f0
[ 0.009743] [0001e352] libusb: debug [submit_iso_transfer] need 1 urbs for new transfer with length 2080
[ 0.009747] [0001e352] libusb: error [submit_iso_transfer] submiturb failed error -1 errno=2
[ 0.009750] [0001e352] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
terminate called after throwing an instance of 'std::runtime_error'
  what():  status: -1 LIBUSB_ERROR_IO
```

mcuee, 2023-05-18:
@Osurac529 
I think this is not a libusb issue. Rather you have to know the protocol in order to use libusb for your device.

I will close this issue now. Please re-open if you have new info.

closed 2023-05-18 by mcuee