libusb/libusb #1211: Describe the difference between libusb0 / libusb1 / OpenUSB in the FAQ

cher-nov, 2022-10-31:
This is a request to describe just a very obscure topic. Thanks anyway.

mcuee, 2022-11-01:
Is the following okay?
* https://github.com/libusb/libusb/wiki/FAQ#libusb-win32-libusbk-and-libusb-project-what-about-openusb

cher-nov, 2022-11-01:
@mcuee Nice, thank you.

I also saw a few things here that seem to me to be mistakes, but for which it's too pointless to create new issues:

1. It would be nice to normalize **libusbk** / **libusbK** spellings in the text (now there are **libusbk.sys**, **libusbK.sys**, **libusbk.dll** and **libusbK.dll** spellings, which is kinda confusing and difficult to distinguish when reading).

2. The sentence `If libusbk.dll is present, it will use libusbK.dll` seems very unclear and ambiguous to me.

3. Is there an erroneous repetition here?
> If libusbK.dll is not present, then it will directly talk to device **using WinUSB using WinUSB API**.

mcuee, 2022-11-08:
Updated based on your feedbacks.
1) Updated
2) Updated and hopefully it is better now
3) Kept, WinUSB mainly means the kernel driver (winusb.sys) here. WinUSB API means winusb.dll.

cher-nov, 2022-11-08:
@mcuee Thank you for your effort!

Not to be nitpicking here, but case-sensitive CTRL+F on "libusbk" reveals the 7 remaining occurrences of this in the text.

Also you didn't answer on this:

> It would also be good to specify the version of the API that OpenUSB forked from.

mcuee, 2022-11-08:
Updated. Please check again. I did not change on instance of `libusbk` to `libusbK` as is the correct URL (https://libusbk.sourceforge.net/UsbK3/index.html.