libusb/libusb #1213: libusb_init() Logging - Option to Set Logging Callback Right After context Created

AtlasSM, 2022-10-31:
From core.c, if ```USE_SYSTEM_LOGGING_FACILITY``` is not defined we get to this part in ```static void log_str```:

```
	/* Global log handler */
	if (log_handler)
		log_handler(NULL, level, str);
	else
		fputs(str, stderr);
```
while ``log_handler`` isn't configured (Doing it after calling this function), it will go to ``fputs``.
For services that only have the ability to log via callback functions, ``libusb_init()`` logs are not available.
Adding a context creation function can provide a way of getting the logs after context creation while calling ``libusb_set_log_cb`` before ``libusb_init()``

mcuee, 2022-11-01:
Would you please create a PR for this issue? Thanks.

tormodvolden, 2022-11-13:
Do I understand correctly that you suggest having a separate context creation function and a context initialization (including enumeration) function? I think that was also considered in the discussion that instead lead to PR #1026. But this PR won't help for the logging issue. However, can't you simply use libusb_set_log_cb() with NULL ctx (= default context) before libusb_init() and it will then take effect for the created context?

hjelmn, 2023-01-10:
```libusb_init_context``` is probably the way to go here. We can update it to ensure that the callback function can be set as an option.

hjelmn, 2023-01-10:
I would argue that ```libusb_set_log_cb``` should be deprecated and replaced with``` libusb_set_option```. There are two modes and they can easily be defined as:

```c
libusb_set_log_cb(NULL, log_cb, LIBUSB_LOG_CB_GLOBAL)
```
becomes
```c
libusb_set_option(NULL, LIBUSB_SET_LOG_CB, log_cb);
```
And
```c
libusb_set_log_cb(ctx, log_cb, LIBUSB_LOG_CB_CONTEXT)
```
becomes
```c
libusb_set_option(ctx, LIBUSB_SET_LOG_CB, log_cb);
```

hjelmn, 2023-01-10:
I have this prototyped already. Can open a PR for it separately from ```libusb_init_context```.

mcuee, 2023-02-01:
@hjelmn 

Please help to open the PR if you are ready. Thanks.

hjelmn, 2023-02-10:
Will get this up soon. I want to get it into 1.0.27. I am working on unit testing now since we have no unit testing of libusb_set_option.

mcuee, 2023-03-24:
@hjelmn 

Just wondering if you have any updates for the PR. Thanks.

mcuee, 2023-03-30:
@hjelmn 
Since you mentioned that your github notification was not working, ping again. Thanks.

hjelmn, 2023-04-01:
Code is done but I need to finish up some testing. Will get that done tomorrow and open the PR.

mcuee, 2023-04-12:
@hjelmn 

PR #1265 has been merged. So this can be closed, right?

closed 2026-03-12 by mcuee