libusb/libusb #1215: libusb_init: Should not log warning for normal usage

photron, 2022-11-02:
Since commit 6b29aeb9ca22ed5ea9768020895698bd452c2a85 libusb_init logs a warning "installing new context as implicit default" when calling for a non-NULL libusb_context argument for the first time.

Why should this be a warning for perfectly good and normal usage? I think this should be lowered to debug level.

mcuee, 2022-11-02:
@tormodvolden and @benzea
The above seems to be a good suggestion. Please check as well. Thanks.

tormodvolden, 2022-11-13:
Yes, I also think this should be info/debug and not warning, and that it is meant as a "warning" for the programmer with regards to earlier behavior, but not for the user running the program to starting wondering if something is wrong.

benzea, 2022-11-17:
Yes, that should never have been a warning, as it will happen even in the case of normal API usage. The only important warning/error message is `API misuse! Using non-default context as implicit default.` message and it should be kept.

commit 1c99382a (2022-12-19): core: Do not warn about new context as implicit default

Only leave a debug message.

Fixes #1215
closed 2022-12-19 by tormodvolden