libusb/libusb #1220: Question: LibUSB don't work with device that has usbser.sys as driver

DanielMartensson, 2022-11-22:
I have an STM32 that has Communication Device Class (CDC) enabled and I cannot connect to my STM32 with LibUSB.

I'm using the correct vendor ID of the device is 0x0483 and product ID is 0x5740
[image]

The device name is:
[image]

Unit stack:
[image]

I can connect with Win32 C++ API but not with LibUSB. See working example how to reproduce the problem.
I think that the problem is that LibUSB cannot connect to USB ports of my Windows 11 computer, due to `usbser.sys` driver. Is that correct?

Is there anything I can do to make this work with LibUSB?

```cpp

#include <windows.h>
#include <stdio.h>
#include <stdio.h>
#include <libusb-1.0/libusb.h>

int main()
{
	// Try with Win32 API
	HANDLE hComm;
	hComm = CreateFileA("\\\\.\\COM8",                  //port name
		GENERIC_READ | GENERIC_WRITE,                   //Read/Write
		0,                                              // No Sharing
		NULL,                                           // No Security
		OPEN_EXISTING,                                  // Open existing port only
		0,                                              // Non Overlapped I/O
		NULL);                                          // Null for Comm Devices

	if (hComm == INVALID_HANDLE_VALUE)
		printf("Win32 API: Error in opening serial port\n");
	else
		printf("Win32 API: Opening serial port successful\n");

	CloseHandle(hComm);                                 // Closing the Serial Port

	/* Try with LibUSB */
	struct libusb_context* ctx;
	int rc = libusb_init(&ctx);
	if (rc < 0) {
		fprintf(stderr, "LibUSB: Error initializing libusb: %s\n", libusb_error_name(rc));
		exit(1);
	}else {
		printf("LibUSB: Init!\n");
	}

	/* Set debugging output to max level. */
	libusb_set_debug(ctx, 3);

	/* Look for a specific device and open it. */
	int VENDOR_ID = 0x0483;
	int PRODUCT_ID = 0x5740;
	libusb_device_handle* devh = libusb_open_device_with_vid_pid(ctx, VENDOR_ID, PRODUCT_ID);
	if (!devh) {
		fprintf(stderr, "LibUSB: Error finding USB device\n");
	}
	else {
		printf("LibUSB: Device connected with LibUSB\n");
	}

	return 0;
}
```

tormodvolden, 2022-11-22:
The usbser.sys driver "grabs" the USB device and translates it into a virtual serial port. From Win32 API you are from userspace dealing with a serial port, not with the underlying USB device.

In order to access the USB device with libusb, you can use "zadig" and unbind the usbser.sys driver and instead bind WinUSB.sys to it. Or you change the USB descriptors in the firmware so that it won't present itself as a virtual serial port, but something unknown that Windows then allows WinUSB.sys and libusb to use.

Please take a look at the Windows specific information on libusb.info (also in  the FAQ section).

DanielMartensson, 2022-11-22:
> 

Thank you for the quick reply.

Ok, two options then. "zadig" or change my USB descriptor in the firmware...of my USB unit? So all I need to do is to make my USB device introduce itself for my PC with...another descriptor e.g vendor ID and product ID? 

[image]

DanielMartensson, 2022-11-22:
> 

Well, I could change the PID and VID numbers so it match STLINK-V2 from the USB 
http://www.linux-usb.org/usb.ids

When I connect a STLink-V2 device, then it works with LibUSB.

DanielMartensson, 2022-11-22:
> 

I want to do it at the cleanest way as possible. Then I assume that it's the "zadig" software I need to use?

DanielMartensson, 2022-11-22:
Great!

I tried "zadig" and then replaced the `usbser.sys` with `WinUSB (v6.1.7600.16385)` and now when I run my C++ code above.

```cpp
Win32 API: Error in opening serial port
LibUSB: Init!
Vendor ID: 0x4f2, Product ID: 0xb6bf
Vendor ID: 0x8086, Product ID: 0x9a13
Vendor ID: 0x483, Product ID: 0x5740
Vendor ID: 0x8086, Product ID: 0xa0ed
Vendor ID: 0x6cb, Product ID: 0xdf
Vendor ID: 0x8087, Product ID: 0x26
LibUSB: Device connected with LibUSB
```

closed 2022-11-22 by DanielMartensson