libusb/libusb #1225: Return kernel driver name under Windows

mcuee, 2022-12-31:
Right now we only provide the driver name (WinUSB, libusb0.sys or libusbk.sys) in debug messages. It will be good to provide this function. For other OS, we can returen a generic name.

Linux -- return `usbfs`
macOS -- return `NIL`?
BSDs -- return `ugen`

mcuee, 2022-12-31:
Reference issue from avrdude project. 
https://github.com/avrdudes/avrdude/issues/1244#issuecomment-1366668250

hjelmn, 2023-01-10:
Maybe IOKit for macOS. Not likely going to change but that is the actual driver we are using.

sonatique, 2023-10-17:
> It will be good to provide this function. For other OS, we can returen a generic name.
Hello @mcuee : do you know if someone is working on this?

mcuee, 2023-10-17:
> Hello @mcuee : do you know if someone is working on this?

As far as I know, nobody is working on this.

fabiensanglard, 2023-12-08:
I volunteer to implement it if it is still needed (It looks like it is given the diversity of `windows_usb_api_backend`). I could implement it by adding a method pointer in `usbi_os_backend`.

```
const char* (*get_driver_name)() {
} ```

And expose that in libusb.h

```
const char* API_EXPORTED libusb_get_driver_name()
{
    return usbi_backend->get_driver_name();
}
```

And I would implement that method in all the backends. Would that work with libusb approvers or there a better to implement it?

mcuee, 2023-12-09:
Looks like a good plan. A PR will be welcome.

But before submitting a PR, let's hear from @hjelmn and @tormodvolden to see if they like the proposal or not.

fabiensanglard, 2023-12-11:
Ping @hjelmn / @tormodvolden  ?

sonatique, 2023-12-11:
@fabiensanglard : this looks good to me. I was wondering whether this information could be added to an existing structure returned by an function of the current API, but I didn't find a match, so I am OK with augmenting the API.

It would be even cooler to be able to retrieve the driver version alongside the driver name, but this is another completely different story and may not apply to all backend.

tormodvolden, 2023-12-11:
We are a bit busy with releasing 1.0.27 now. I will come back to this later.

fabiensanglard, 2023-12-26:
Sorry about the presubmit spam. Next time, I will add the issue reference once all tests are passing to avoid it.

I still need to figure out the signing but  https://github.com/libusb/libusb/pull/1404 is in reviewable state.

sonatique, 2023-12-26:
> I still need to figure out the signing

Hello, thank you. Don't worry about / waste your time with signing your commit, it is actually not a requirement here (as opposed to what is said by the "Merge is blocked". See for instance this answer from an admin: https://github.com/libusb/libusb/pull/1397#issuecomment-1866328324