libusb/libusb #1226: libusb0.sys problem with avrdude Micronucleus programmer

mcuee, 2022-12-31:
Reference: Input/output error when using libusb0.sys driver with libusb-1.0+libusb-compat-0.1
https://github.com/avrdudes/avrdude/issues/1244#issuecomment-1366668250

mcuee, 2022-12-31:
Debug messages will be provided soon.

mcuee, 2023-02-13:
Similar issue reported here for USBtinyISP.
* https://github.com/MCUdude/MiniCore/issues/253

mcuee, 2023-02-13:
These programmers are based on V-USB. They are mostly compliant to USB 1.1 specification as low-speed USB device. But they are not compliant in some aspects like handling of communication errors and electrical specifications.
https://www.obdev.at/products/vusb/index.html

tormodvolden, 2023-02-13:
On issues with libusb0.sys, it would be worth trying out PR #1018 just to see more exactly what the reported driver errors are. In any case, debug logs will be needed.

mcuee, 2023-02-16:
The issue goes away with libusb0.sys 1.2.7.3 version, therefore this is not a libusb Windows backend issue. I will close this one.
https://github.com/avrdudes/avrdude/issues/1244#issuecomment-1433026901

closed 2023-02-16 by mcuee

tormodvolden, 2023-02-16:
Great.

BTW, why is the avrdude projext insisting on using the libusb 0.1 API, to the point of maintaining their own avrdude-libusb? Or whatever the reason is, I just looked at https://github.com/avrdudes/avrdude/wiki/The-story-on-libusb-for-Windows . That page is maybe not up to date. And on https://github.com/avrdudes/libusb it says

> This library was completely rewritten to support the default Windows WinUSB driver, as well as the legacy libusb0.sys driver, while maintaining a libusb-0.1 compatible programming API.

Wouldn't that be covered by libusb + libusb-compat?

mcuee, 2023-02-16:
For the purpose of avrdude, libusb-0.1 API is okay. And it takes time to move the code base to libusb-1.0 API. So the project will still be using libusb-0.1 API mainly (some codes can use either libusb-1.0 API and fall back to libusb-0.1 API).
* https://github.com/avrdudes/avrdude/issues/906#issuecomment-1120433920

As for the reason of avrdude-libusb, I think part of the reasons was due to the previous problems of libusb Windows backend when using libusb0.sys. 

You can also read through the comments here.
* https://github.com/avrdudes/avrdude/issues/906
 
> Wouldn't that be covered by libusb + libusb-compat?

That is indeed what Arduino's distribution of avrdude is using. Previously they were using libusb-win32 for a long time.
* https://github.com/arduino/avrdude-packing/issues/15